letscontrolit/ESPEasy #1194: Frequent crashes to investigate (MQTT IMPORT PLUGIN?)

giig1967g, 2018-03-29:
It seems that receiving an MQTT IMPORT update frequently, the unit crashes.
For frequently I mean 2 or 3 times per second.
As we cannot control the speed of the broker when sending the MQTT updates, the issue is quite serious.

As a practical example, consider a remote unit for setting a thermostate temperature using two push buttons (up and down) that sends to this unit the temp setting. The user will push the buttons until the temperature is set at a speed of 3 or 4 pushes for second. The remote unit then will publish to MQTT the value 3 or 4 times per second, and the local unit will crash.

Adding a timer in order to add a pause before processing the changes, avoid the crashes (example below).

### Steps to reproduce
How can we trigger this problem?

_**See example and pictures at this post:**_
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5102#p26724

Create the following tasks: MQTT Import, Dummy and OLED (see picture)
Add the following SIMPLE rules:
```
On System#Boot do
  gpio,14,1 //z1
  taskvalueset,1,1,1 // r1=off
endon

on import1#set1 do
  event,zone1
endon

on zone1 do
  if [temp#t1]<[import1#set1]
    taskvalueset,1,1,0
    gpio,14,0 //turn zone1 ON
  else
    taskvalueset,1,1,1
    gpio,14,1 //turn zone1 OFF
  endif
endon
```

Then modify using MQTT the import1#set1 value (using a speeds like you were pressing a push button to increase the temperature setting of a thermostate, say 3 times per second)

After few pushes the unit will crash (see log at https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5102#p26724).

Working configuration:
If I modify the code as follows adding a timer, it doesn't crash anymore:

```
on import1#set1 do
  timerset,1,2
endon

on rules#timer=1 do
  event,zone1
endon
```

### System configuration
Hardware:
WEMOS D1 8266 or 8255 (lite)

Software or git version:
mega-20180326

TD-er, 2018-03-29:
Is it the same crash error as discussed yesterday with the panic _yield() in the core library?

giig1967g, 2018-03-29:
Yes it is.
I opened here only after excluding a syntax error and only finding a workaround, so that the cause can be investigated.

TD-er, 2018-03-29:
OK, just checking, since the actual error was not mentioned here.
Then it is probably a bug in PubSubClient calling a delay or yield from within an event handler.
These may only be called from setup() or loop().

giig1967g, 2018-03-29:
yes. sorry.

giig1967g, 2018-03-29:
Just an update. 
I did more testing:
If I send the same command to update the temperature using SendTo from the remote unit to this unit, the error does not occur.
I believe the error is inside MQTT import.

redskinhu, 2018-03-29:
Hello

I have something similar (esp07, mega-20180327 and older fw too):

IMPT : Connected to MQTT broker with Client ID=Livingroom_ESP07-Import
IMPT : [MQTT_Input_Livingroom_ESP07#PWM_14_onoff_state] subscribed to home/Livingroom_ESP07/PWM/14/oo_state
IMPT : [MQTT_Input_Livingroom_ESP07#PWM_14_onoff_command] subscribed to home/Livingroom_ESP07/PWM/14/oo_command
IMPT : [MQTT_Input_Livingroom_ESP07#PWM_14_brightness_state] subscribed to home/Livingroom_ESP07/PWM/14/br_state
IMPT : [MQTT_Input_Livingroom_ESP07#PWM_value] subscribed to home/Livingroom_ESP07/PWM/14
MQTT : Connection lost
MQTT : Connected to broker with client ID: ESPClient_EC:FA:BC:02:14:0E
Subscribed to: home/Livingroom_ESP07/#
IMPT : [MQTT_Input_Livingroom_ESP07#PWM_14_onoff_command] : 1.00
EVENT: MQTT_Input_Livingroom_ESP07#PWM_14_onoff_command=1.00
ACT  : publish home/Livingroom_ESP07/PWM/14/oo_state,1
ACT  : publish home/Livingroom_ESP07/PWM/14,150

Exception (3):
epc1=0x402152f2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40220872 depc=0x00000000

ctx: cont 
sp: 3fff50e0 end: 3fff5860 offset: 01a0

>>>stack>>>
3fff5280:  00000050 00000001 3fff5304 4020b2b7  
3fff5290:  00000089 3fff5304 00000000 40217048  
3fff52a0:  0000004d 00000001 00000003 40217081  
3fff52b0:  000108e6 3fff53c4 00000003 402221e3  
3fff52c0:  000108e6 3fff53c4 00000005 40222267  
3fff52d0:  000108e6 3fff53c4 00000005 402409b7  
3fff52e0:  00000000 00000000 00000000 00000000  
3fff52f0:  00000000 00000000 00000000 00000000  
3fff5300:  00000000 3fff8fc4 0000005f 0000005d  
3fff5310:  00000001 3fff5550 3fff3019 40242144  
3fff5320:  656d6f68 76694c2f 72676e69 5f6d6f6f  
3fff5330:  30505345 57502f37 34312f4d 5f6f6f2f  
3fff5340:  74617473 00000065 6d6f6800 694c2f65  
3fff5350:  676e6976 6d6f6f72 5053455f 502f3730  
3fff5360:  312f4d57 6f6f2f34 6d6f635f 646e616d  
3fff5370:  6f680000 4c2f656d 6e697669 6f6f7267  
3fff5380:  53455f6d 2f373050 2f4d5750 622f3431  
3fff5390:  74735f72 00657461 68000000 2f656d6f  
3fff53a0:  6976694c 6f72676e 455f6d6f 37305053  
3fff53b0:  4d57502f 0034312f 00000000 00000000  
3fff53c0:  00000000 3fff8eb4 0000003f 00000035  
3fff53d0:  3fff8e5c 0000004f 00000040 3fff8e24  
3fff53e0:  0000002f 00000027 3fff8dec 0000002f  
3fff53f0:  00000027 3fff8dac 0000000f 00000001  
3fff5400:  3fff8dc4 0000001f 00000017 3fff8d84  
3fff5410:  0000001f 00000010 3f800000 4020b2b7  
3fff5420:  3fff5349 00000001 3fff5584 4020b2e9  
3fff5430:  3fff55a0 3fff8177 3fff5584 00000024  
3fff5440:  3fff55a0 00000003 3fff5550 40216b9a  
3fff5450:  00000031 3fff7dd4 00000000 3fff5550  
3fff5460:  00000001 40292369 3fff560c 00000030  
3fff5470:  00000000 ffffffff 3fff560c 3fff5a08  
3fff5480:  00000010 3fff555c 00000000 4020f61d  
3fff5490:  40241ce0 3fff5740 3fff574c 00000014  
3fff54a0:  400a6a7f 00000000 00000000 3fff5a08  
3fff54b0:  00000010 3fff5540 00000000 4020f61d  
3fff54c0:  00000000 3fff5550 00000000 3fff5a08  
3fff54d0:  00000010 3fff5560 00000000 3fff560c  
3fff54e0:  3fff475c 00000000 3fff18b8 40259868  
3fff54f0:  402920cb 00000003 3fff5540 402140e8  
3fff5500:  00000000 3fff5560 3fff1974 40260d35  
3fff5510:  40209ea0 3ffe8cd3 3fff1974 3fff560c  
3fff5520:  3fff5a08 000001fd 000001fd 4020f2f8  
3fff5530:  3fff5a08 0000012d 0000012d 4020f2f8  
3fff5540:  00000000 3ffe8af4 00000001 4020f768  
3fff5550:  00000300 00000c00 00000000 40107600  
3fff5560:  00000000 00000000 00000000 00000000  
3fff5570:  00000000 3fff5700 3fff8d4c 0000002f  
3fff5580:  00000027 3fff6bcc 0000000f 00000001  
3fff5590:  3fff8a84 0000000f 00000000 00000000  
3fff55a0:  3fff7dac 0000000f 00000001 3fff8a4c  
3fff55b0:  0000002f 00000027 3fff8144 40207ea4  
3fff55c0:  3fff8144 00000003 3fff55e0 3fff5620  
3fff55d0:  3fff5620 000108dd 3fff8144 40254cc0  
3fff55e0:  0031002c 3fff811b 00000020 402085de  
3fff55f0:  656d6f68 76694c2f 72676e69 5f6d6f6f  
3fff5600:  30505345 57502f37 34312f4d 5f6f6f2f  
3fff5610:  6d6d6f63 00646e61 3fff8144 402084f0  
3fff5620:  0000ff01 00000014 3fff574c 4020b2e9  
3fff5630:  3fff1a88 3fff5790 3fff574c 4020b33a  
3fff5640:  3ffe9046 3fff80f4 00000002 00000003  
3fff5650:  3fff1a88 3fff5790 00000005 40241f39  
3fff5660:  0000001d 3fff80f0 00000001 40250d7c  
3fff5670:  00000bb2 3fff7444 00000001 3fff5790  
3fff5680:  00000000 000003d5 000003d5 4020f2f8  
3fff5690:  3fff5a08 000003d5 000003d5 00000003  
3fff56a0:  3fff475c 00000000 3fff18b8 40259868  
3fff56b0:  00000014 00001ea8 000003d5 402140e8  
3fff56c0:  00000010 3fff5750 3fff195c 40260d35  
3fff56d0:  40209ea0 00000005 3fff195c 4020b3f0  
3fff56e0:  00000010 00000001 3fff18b8 402177fa  
3fff56f0:  3fff5a08 00000369 00000369 4020f2f8  
3fff5700:  3fff1a88 00000005 3fff5730 4020f768  
3fff5710:  3fff5a08 000003d2 000003d2 4020f2f8  
3fff5720:  3fff1a88 00000005 3fff5750 4020f768  
3fff5730:  3fff5a08 000001fd 000001fd 4020f2f8  
3fff5740:  3fff7dc4 0000001f 00000017 3fff80f4  
3fff5750:  0000001f 00000010 00000000 4020b210  
3fff5760:  3ffe8af4 3fffdad0 40285af3 40217d8a  
3fff5770:  00000000 00000000 00000000 00000003  
3fff5780:  3fff1a88 00000005 3fff5790 402209ac <
3fff5790:  00000300 3fff0c00 00000000 3fff4801  
3fff57a0:  00000000 00000000 00000000 00000000  
3fff57b0:  00000000 00000003 3fff8bfc 0000000f  
3fff57c0:  00000000 3fff8c14 0000000f 00000000  
3fff57d0:  3fff7234 0000000f 00000000 00000000  
3fff57e0:  00000001 00000000 00000003 00000024  
3fff57f0:  3fff31f3 3fff4838 000000ff 4021b3b7  
3fff5800:  00000000 00000000 000108db 3fff4838  
3fff5810:  3fffdad0 3fff1a88 3fff1a48 40240a0b  
3fff5820:  3fffdad0 00000000 3ffe83b0 4024a665  
3fff5830:  00000000 00000000 00000001 4020c2d5  
3fff5840:  3fffdad0 00000000 3fff4832 4020c300  
3fff5850:  feefeffe feefeffe 3fff4840 4020f864  
<<<stack<<<

TD-er, 2018-04-14:
This is fixed, right?

redskinhu, 2018-04-14:
Hi

mega-20180414 installed. Looks good.

Thank You
T