letscontrolit/ESPEasy #1195: An MQTT message is published more than once, after deep sleep.

demestav, 2018-03-29:
### Steps to reproduce
How can we trigger this problem?
- Add a task with any delay. In my case this was a DHT11 sensor, and tested various delay values ranging from 60 to 120 seconds.
- Enable deep sleep mode. In my case this was set to 20 seconds awake and 60 seconds delay.
- Enable publishing to MQTT. In my case this was set to OpenHUB controller.
- Issue a subscriber to the MQTT broker to watch the messages.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
Yes.

### Expected behavior
Tell us what should happen?
The sensor values should be published to MQTT only once every time the module exits from deep sleep.

### Actual behavior
Tell us what happens instead?
The sensor values are published more than once to MQTT. Most of the time they are published twice, but sometimes more. A strange pattern that emerges is that not all values are published i.e.:

```
/testdht11/Temperature 23.00
/testdht11/Humidity 40.00
/testdht11/Temperature 23.00
```
Looking at the log, I can see that the sensor is indeed triggered twice (or sometimes more) every time the module exists from deep sleep.

### System configuration

Hardware:
NodeMCU development board

Software or git version:
Precompiled test package from 28/03/2018 i.e. `ESP_Easy_mega-20180328_test_ESP8266_4096.bin`

demestav, 2018-04-13:
I have pulled and tested the `normal` and `dev` version of https://github.com/letscontrolit/ESPEasy/commit/d083a58cd7738e0c3fe93d9a39d15db5f8e8beb8 

In both versions MQTT publishing **does not** work after deep sleep. When deepsleep is disabled then all data is normally published to MQTT .

Please let me know if you require any more details to pinpoint this issue.

demestav, 2018-04-13:
Yes sure,
The output is:
```
575882 : DHT  : Temperature: 24.00
575882 : DHT  : Humidity: 45.00
580881 : DHT  : Temperature: 24.00
580881 : DHT  : Humidity: 45.00
584070 : FILE : Saved config.dat
584160 : SLEEP: Entering deep sleep in 30 seconds.
614160 : EVENT: System#Sleep
614187 : SLEEP: Powering down to deepsleep...
��␀␀�␀␀␀��␀�␀���␀�␀␀␀�␀��␀�␀
```
Note that the last line is garbled text in my terminal too (i.e. not a problem with copy and paste).

At time `584070` is where I activated the deepsleep feature.

From the LED indicator, I understand that the module stays awake for about a second or less. It doesn't seem enough time to connect to WiFi then to MQTT broken and to publish.

demestav, 2018-04-13:
I just realized that the reset cable got disconnected so please disregard my previous comment. Sorry about that!

Here is the output of 2 cycles:
```
INIT : Booting version:
269 : INIT : Rebooted from deepsleep #10
271 : FS   : Mounting...
295 : FS   : Mount successful, used 75802 bytes of 957314
304 : CRC  : No program memory checksum found. Check output of crc2.py
308 : CRC  : Settings CRC           ...FAIL
336 : CRC  : SecuritySettings CRC   ...OK
441 : INIT : Free RAM:15784
442 : INIT : I2C
442 : INIT : SPI not enabled
458 : WIFI : Set WiFi to AP+STA
1345 : WIFI : AP Mode ssid will be test1_0 with address 192.168.4.1
1346 : WIFI : Connecting XXXXXX attempt #0
1358 : WD   : Uptime 0 ConnectFailures 0 FreeMem 11408
1397 : DHT  : Temperature: 25.00
1397 : DHT  : Humidity: 39.00
2391 : DHT  : Temperature: 25.00
2391 : DHT  : Humidity: 36.00
3391 : DHT  : Temperature: 25.00
3391 : DHT  : Humidity: 32.00
4480 : WIFI : Connected! AP: XXXXX Ch: 11 Duration: 3071 ms
4508 : DHT  : Temperature: 25.00
4508 : DHT  : Humidity: 31.00
5608 : DHT  : Temperature: 25.00
5608 : DHT  : Humidity: 30.00
6692 : EVENT: System#Sleep
6719 : SLEEP: Powering down to deepsleep...
{l␀l��|␀�d�|␃␄␄␌�␌d�␌b|��␃�

INIT : Booting version:
268 : INIT : Rebooted from deepsleep #11
269 : FS   : Mounting...
294 : FS   : Mount successful, used 75802 bytes of 957314
303 : CRC  : No program memory checksum found. Check output of crc2.py
306 : CRC  : Settings CRC           ...FAIL
335 : CRC  : SecuritySettings CRC   ...OK
439 : INIT : Free RAM:15912
439 : INIT : I2C
439 : INIT : SPI not enabled
455 : WIFI : Set WiFi to AP+STA
1342 : WIFI : AP Mode ssid will be test1_0 with address 192.168.4.1
1343 : WIFI : Connecting XXXXXX attempt #0
1354 : WD   : Uptime 0 ConnectFailures 0 FreeMem 11664
1392 : DHT  : Temperature: 25.00
1392 : DHT  : Humidity: 39.00
2388 : DHT  : Temperature: 25.00
2388 : DHT  : Humidity: 35.00
3387 : DHT  : Temperature: 25.00
3387 : DHT  : Humidity: 32.00
4486 : WIFI : Connected! AP: XXXXXX Ch: 11 Duration: 3079 ms
4516 : DHT  : Temperature: 25.00
4516 : DHT  : Temperature: 25.00
4517 : DHT  : Humidity: 31.00
5615 : DHT  : Temperature: 25.00
5615 : DHT  : Humidity: 30.00
6699 : EVENT: System#Sleep
6726 : SLEEP: Powering down to deepsleep...
```

TD-er, 2018-04-13:
Hmm very strange, why is it trying to set the wifi to AP+STA?

What version are you running?
This log output of the AP+STA is introduced in the last 2 days. Before that it was just in a pull request, not a pre compiled version of the nightly builds.

demestav, 2018-04-13:
I `git cloned` the  and then followed the instructions for building and uploading using PlatformIO.

From the `info` section of the module
```
Firmware
--
Build | 20101  - Mega (core 2_3_0)
GIT version |  
Plugins | 71 [Normal] [Testing] [Development]
Build Md5 | 4d44355f4d44355f4d44355f4d44355f
Md5 check | fail !
Build time | Apr 13 2018 10:52:26
```

TD-er, 2018-04-13:
Ah, that's today's build.
But you're using the 2.3.0 core library.
Why the dev build and not the "normal"?
Do you need something from the extra plugins not included in normal?

Could you try the normal build from today's build, or build one for yourself using the "normal_esp8266_" followed by the memory size you need?
Those builds use Core 2.4.0 which solves a number of issues related to the wifi code.
But it takes a bit more memory which is an issue with the Testing and Dev builds.

TD-er, 2018-04-13:
That's very strange it is trying to start AP + STA.
That means the node does not accept the set password + SSID config as valid. (or some other reason, but given the timings in the log, that's my best estimate)

demestav, 2018-04-13:
With deepsleep disabled the node is indeed in AP mode (I can see the ssid when searching WiFi) and at the same time, it is publishing to MQTT.

Isn't this the expected behavior?

TD-er, 2018-04-13:
Nope, AP should not be active.

demestav, 2018-04-13:
After some time the AP mode is disabled. From the log:
```
52164 : DHT  : Humidity: 16.00
57164 : DHT  : Temperature: 24.00
57164 : DHT  : Humidity: 16.00
61121 : WD   : Uptime 1 ConnectFailures 0 FreeMem 13488
62164 : DHT  : Temperature: 24.00
62164 : DHT  : Humidity: 16.00
65145 : WIFI : AP Mode Disabled for SSID: test1_0
65146 : WIFI : Set WiFi to STA
65161 : MQTT : Connection lost
65162 : WIFI : Disconnected! Reason: 'Assoc leave' Connected for 1 m 0 s
65162 : WIFI : Connecting XXXXXX attempt #0
66372 : WIFI : Connected! AP: XXXXXX Ch: 11 Duration: 1208 ms
67354 : DHT  : Temperature: 24.00
67355 : DHT  : Humidity: 16.00
70129 : WIFI : DHCP IP: 192.168.0.29 (test1-0) GW: 192.168.0.1 SN: 255.255.255.0   duration: 3757 ms
70484 : MQTT : Connected to broker with client ID: ESPClient_X
70486 : Subscribed to: /test1/#
72513 : DHT  : Temperature: 24.00
72513 : DHT  : Humidity: 16.00
```

TD-er, 2018-04-13:
That's correct, there is a 60 seconds timeout to let users connect to the AP.
As soon as someone connects, the timeout is stopped and when all are disconnected, the timeout is started again at 60 seconds.

demestav, 2018-04-13:
So regarding your comment https://github.com/letscontrolit/ESPEasy/issues/1195#issuecomment-381096904 ,why isn't ok that the WiFi is AP+STA? According to the log, it seems to connect just fine to the WiFi.

TD-er, 2018-04-13:
It should only be "STA" and not "AP+STA", since the AP should only be there when it cannot connect as STA (as a normal station to another accesspoint)
As said, as long as this "AP+STA" is there, there is something preventing it from  normal operations.
And this is specific to the deep sleep mode, so there is something wrong in the logic here, which may explain the issues you're experiencing with MQTT publish after deep sleep.

demestav, 2018-04-16:
I see. Should I open another issue for this?

demestav, 2018-04-16:
Indeed I think the AP+STA is gone now. Here is the new log:
```
INIT : Booting version:  (core 2_4_0)
204 : INIT : Rebooted from deepsleep #4
206 : FS   : Mounting...
231 : FS   : Mount successful, used 75802 bytes of 957314
241 : CRC  : No program memory checksum found. Check output of crc2.py
245 : CRC  : Settings CRC           ...FAIL
273 : CRC  : SecuritySettings CRC   ...OK
375 : INIT : Free RAM:19656
375 : INIT : I2C
376 : INIT : SPI not enabled
390 : INFO : Plugins: 47 [Normal] (core 2_4_0)
390 : WIFI : Connecting XXXXXX attempt #0
401 : WD   : Uptime 0 ConnectFailures 0 FreeMem 16528
437 : DHT  : Temperature: 24.00
437 : DHT  : Humidity: 17.00
1434 : DHT  : Temperature: 24.00
1434 : DHT  : Humidity: 17.00
2434 : DHT  : Temperature: 24.00
2434 : DHT  : Humidity: 17.00
3518 : WIFI : Connected! AP: XXXXXX Ch: 11 Duration: 3059 ms
3547 : DHT  : Temperature: 24.00
3547 : DHT  : Humidity: 17.00
4643 : DHT  : Temperature: 24.00
4643 : DHT  : Humidity: 17.00
5744 : DHT  : Temperature: 24.00
5744 : DHT  : Humidity: 17.00
6829 : EVENT: System#Sleep
6855 : SLEEP: Powering down to deepsleep...
```
Is it possible that there is not enough time to establish a connection with the MQTT broker? I noticed that even though in the settings the `Sleep awake time` is set to 20 sec, here the module goes to sleep after 7 seconds.

DittelHome, 2018-04-29:
Today i have tested with ESP_Easy_mega-20180428_normal_ESP8266_4096.bin.
The sensor values are published more than once to MQTT. I have set the awake time to 1 and the sensor delay to 60. I get two values. If i set the awake time to 5 seconds, i get 5 values. 
Problem still exits.

TD-er, 2018-04-29:
As soon as the wifi issues are resolved, this one is high on my todo list (MQTT related issues)

adcomrod, 2018-05-29:
Hi. Experiencing same issue. Tried first with ESP_Easy_mega-20180524_normal_ESP8266_4096 and ESP_Easy_mega-20180524_dev_ESP8266_4096. Went back to ESP_Easy_v2.0-20180322_normal_ESP8266_4096 and the issue was there also. As soon as the esp wakes up and connects to the broker, it start to publish the values over and over until the sleep message is displayed:
..
24320 : DHT : Humidity: 74.70
25320 : DHT : Temperature: 22.30
25320 : DHT : Humidity: 74.70
26319 : DHT : Temperature: 22.40
26319 : DHT : Humidity: 74.80
28917 : DHT : Temperature: 22.40
28917 : DHT : Humidity: 74.80
29916 : DHT : Temperature: 22.40
29917 : DHT : Humidity: 74.70
30396 : SLEEP: Entering deep sleep in 30 seconds.
As soos as the sleep is disabled the publish stops

uzi18, 2019-09-04:
@adcomrod any update ?

TD-er, 2019-10-27:
Is this still an issue?

TD-er, 2019-11-04:
This one is probably not an issue anymore, so I will close it.
Please let me know if it still is an issue.

closed 2019-11-04 by TD-er

RicardoCst, 2021-08-06:
Having the same issue with the AM2320 now. Its publishing the humidity and temperature messages 2 times.