letscontrolit/ESPEasy #1196: Add new formats #

giig1967g, 2018-03-29:
Hi,
I really appreciated the #949 enhancement.

I would like to suggest to add more formats:
#OC = instead of 1/0 display OPEN/CLOSE
#UD = instead of 1/0 display UP/DOWN
#OC! and #UD! for inverted logic.

And numeric conversions:
#D = to convert to integer (round up if >= 0.5 and down if < 0.5)
#Fx = to convert to a float with x decimals (F1, F2, etc.)
#Fx.y = to converto to float with "x" minimum digits and "y" fixed decimals. Example: if I have a value of 5.2, using #F3.2 will result in "005.20", like "000.00" format in Excel) for justify in lcd screens
#Sx.y =like # Fx.y but instead of filling with 0 will fill with spaces: (if I have a value of 5.2, using #S3.2 will result in "  5.20") 
#RU = Round UP: to convert to upper integer part (like CEILING function in excel)
#RD = Round Down: to convert to lower integer part (like FLOOR in excel)

What do you think?

Grovkillen, 2018-03-29:
Sounds like a nice addition. I added it to future milestone but @uzi18 might think it's a great idea and want it as well = sooner :smile:

TD-er, 2018-03-29:
What should be done when `#OC` gets an input other than 0 or 1 ?
Should we just do similar like C-style Boolean behavior 
```
if (input == 0) return "O";
else return "C";
```

Grovkillen, 2018-03-29:
You mean 0.1...0.9 for example?

giig1967g, 2018-03-29:
I'd say
```
if (input == 0) return "O";
else return "C";
```
It's responsibility of the user to use this format on PIN state only.

giig1967g, 2018-03-29:
Or:

```
if (input == 0) return "O";
elsif (input == 1) return "C";
else return "ERR";
```

TD-er, 2018-04-14:
I think this is fixed, correct?

giig1967g, 2018-04-15:
fixed. thanks.

closed 2018-04-15 by giig1967g