letscontrolit/ESPEasy #1199: Feature Request: Configure script

ToltingColtAcres, 2018-03-29:
This is a feature request for the devs, not an issue per se. Another open issue/feature enhancement request to create a minimal build got me thinking...

It would be extremely useful to those of us who build our own versions to have a "configure" script, similar to that used by many linux applications, which will prompt the user to enable/disable various optional plug-ins, allowing us to save memory/flash space.

If implemented, the configure script could create a config.h used by the various plug-ins to determine if they should be compiled in. e.g. you wrap the switch plug-in in an #ifdef _CONFIG_SWITCH block which, if the device isn't configured in, will produce no associated code (thus reduce the size of the binary) 

I'm not familiar enough with Atom and programming in this environment to know if its possible to do this, but, I would hope a dev with more knowledge than I could implement the idea.

As ESPEasy becomes more popular, and the list of devices grow, it will become exceedingly difficult to produce a binary image which does not consume all available flash space.

TD-er, 2018-03-29:
You can have a look at this (already merged) PR: https://github.com/letscontrolit/ESPEasy/pull/1182  and this not yet finished PR for the Sonoff POW: https://github.com/letscontrolit/ESPEasy/pull/1184
My plan is to make support for the Sonoff POW a sample project to show how to set board specific defaults and perhaps even some pre-configured devices, since you have to change quite a lot of the default settings to get the POW to work. (default pins for I2C are used by another chip on the POW)

Maybe that will inspire others to improve upon these standard boards to have them almost usable direct from flash.

ToltingColtAcres, 2018-03-29:
I like the concept of a product-specific build, although I imagine that can get quite complicated as the number of products onto which ESPEasy can be flashed grows substantially. But, for those of us who use individual products (e.g. I use Sonoff 4CH Pros and Basics) this is a great enhancement.

Would be nice to also have pre-configured (disabled) devices on creation of the initial flash config file as well to minimize setup time, just click on the devices tab and enable the switches/relays you intend to use.

TD-er, 2018-03-29:
I agree we have to make sure it stays manageable and maintainable.
And I will surely keep that in mind when preparing the Sonoff Pow example.
I guess we may get some directory with board definitions and just have to include a .h file from it to select a pre-config.

ToltingColtAcres, 2018-04-25:
I was curious if any progress has been made in this arena. I recently attempted to perform a 2-step OTA firmware upgrade on a 1MB Sonoff 4CH Pro and was unable to do so, I received an insufficient space error message when attempting to flash a newer image. It appears the new images built towards the middle/end of April are significantly larger than their February/March predecessors. I assume this has to do with the recent change to a newer framework.

I was able to update 2 devices manually by flashing from a PC, but one unit is located in an area where doing this is not easy.

The newer builds are significantly faster in many areas so I was hopeful to upgrade all my units.

TD-er, 2018-04-26:
The faster speed unfortunately does not imply good stability.
There are a lot of issues with the wifi and some other issues.