letscontrolit/ESPEasy #1200: Add support for LM35 environmental sensor?

mattlward, 2018-03-29:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Look for it in config, does not exist.
Does the problem persists after powering off and on? (just resetting isn't enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:
Wemos D1, LM35
Software or git version:

TD-er, 2018-03-29:
[This](http://www.ti.com/lit/ds/symlink/lm35.pdf) one?
That's just a temp-to-voltage converter.
The analog input of the ESP8266 is not that accurate to be useful I guess, so how do you want to connect it?

There is currently already support for the analog input, if I'm not mistaken, and you can add a formula to convert the plugin output.

mattlward, 2018-03-29:
That will do for me... I am not looking for extreme accuracy.  Just a simple fan controller.  

Sorry to bug you.

beicnet, 2018-03-29:
@TD-er The source is just a half line what is really simple and good, but it will not work with 3.3V need to go from 5V and also need a level shifter to analog input if I'm not mistaking.

Grovkillen, 2018-03-29:
Can a INA219 be used to measure the voltage level?

beicnet, 2018-03-29:
@Grovkillen Did you just misspost or?

Grovkillen, 2018-03-29:
I didn't read the specs but if the LM35 is analog and output values between 5V..0V then any voltage sensor will suffice. BUT an early :beer: might have got me mistaken.

beicnet, 2018-03-29:
@Grovkillen why should anyone use INA219 only for a little sensor? Complications, no?
So, if you wan to use LM35 you would need INA219 too, weird combo! Kappa

Grovkillen, 2018-03-29:
All right, you got me by the nose. :stuck_out_tongue_winking_eye:

beicnet, 2018-03-29:
@mattlward But, what you think about [TMP36](http://www.analog.com/media/en/technical-documentation/data-sheets/TMP35_36_37.pdf), it can work from 2.7V to 5.5V and no need for level shifter, but it has ±2°C accuracy, also they are pin compatible.

beicnet, 2018-03-29:
@Grovkillen No worries! :beer:  :wink:

TD-er, 2018-03-29:
@beicnet Ofcourse you may need some resistors to get it into range 0...1V
My reaction was more about no need to program for something analog.

beicnet, 2018-03-29:
LM35, TMP36 (Celsius), LM335 (Kelvin) pin compatible (all analog), but what I'm usig is SMT160-30 it's digital, best sensor for me ever! (mentioned sensors are all TO-92 casing).

beicnet, 2018-03-29:
and it's all easy (500 and 0.5 if you are pumping the sensor from 5v supply):
 `degreesC = (vout * 500) / 1023;` or `degreesC = (vout - 0.5) * 100.0;`

mattlward, 2018-03-30:
The same math can be applied and the output can be calculated based on the input resolution for 3.3v as well.  That is my plan, It is a direct fit without any level shifters and a little math.

beicnet, 2018-03-30:
@mattlward LM35 can't be used with 3.3v power supply!

Grovkillen, 2018-03-31:
Can we close this?

beicnet, 2018-03-31:
@Grovkillen Why do you wanna close it?

It can be done with TMP36 instead of LM35, same casing, same pinout, but the supply voltage is 2.7v to 5.5v

beicnet, 2018-03-31:
Yes, thats right, but there is one similar digital one too SMT160-30

mattlward, 2018-03-31:
Yes

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er