letscontrolit/ESPEasy #1204: platformio.ini [env:hard_SONOFF_BASIC] - No "System Information" as Device

Oxyandy, 2018-03-30:
platformio.ini
has a custom line for building Sonoff basic firmware
It's great ! 
Minimal - however it does not even include "System Info" as a device, if you could add that, ta
Easy fix ;)
```
[env:hard_SONOFF_BASIC]
upload_speed     = ${common.upload_speed}
framework        = ${common.framework}
platform         = ${common.platform}
lib_deps         = ${common.lib_deps}
lib_ldf_mode     = ${common.lib_ldf_mode}
board_flash_mode = ${Sonoff.board_flash_mode}
board            = ${Sonoff.board}
build_flags      = ${Sonoff.build_flags} -D PLUGIN_SET_SONOFF_BASIC
```

TD-er, 2018-03-30:
What do you mean with 'system info' ?
I just added these to make a start at building specific versions for popular boards.
The first board-build supported will be the Sonoff POW, since the plugin needed to read the power was using too much resources to be used in a more generic build and it demands some very specific settings.

Oxyandy, 2018-03-30:
They say a picture says a thousand words ;)
Closed ?
[image]
And... importantly
Thanks !

Oxyandy, 2018-03-30:
As is now
[image]

TD-er, 2018-03-30:
Just make a list of things missing, without having to solder components to these boards.

Oxyandy, 2018-03-30:
```
#ifdef PLUGIN_SET_ONLY_SWITCH
    #ifndef PLUGIN_SET_NONE
        #define PLUGIN_SET_NONE
    #endif
    #ifndef USES_P001
        #define USES_P001   // switch
    #endif
    #ifndef USES_P003
        #define USES_P003   // pulse
    #endif
    #ifndef USES_P037
        #define USES_P037   // MQTTImport
    #endif
#endif
```
So what line is added for "System info", I'm not familiar enough, (yet)
plus don't want to have to do it each time.
Then this is closed.

Oxyandy, 2018-03-30:
_P026_Sysinfo.ino
This

Oxyandy, 2018-03-30:
```
#ifdef PLUGIN_SET_ONLY_SWITCH
    #ifndef PLUGIN_SET_NONE
        #define PLUGIN_SET_NONE
    #endif
    #ifndef USES_P001
        #define USES_P001   // switch
    #endif
    #ifndef USES_P003
        #define USES_P003   // pulse
    #endif
    #ifndef USES_P026
        #define USES_P003   // System Info

    #ifndef USES_P037
        #define USES_P037   // MQTTImport
    #endif
#endif
```

Oxyandy, 2018-03-30:
@TD-er so if you could save that, so is not needed to be edited each time is great, 
nothing else I can think of
Cheers

susisstrolch, 2018-03-30:
Please, also add the Dummy device to the list of basic plugins.

TD-er, 2018-03-30:
Did you notice any positive speed improvements on the Sonoff Basic build?
For the Sonoff defines I selected the 2.4.0 core libraries. These builds will have a limited set of plugins, so memory is less of an issue and it allows us to test the 2.4.0 libs also.
The webserver has improved a lot in speed.

Oxyandy, 2018-03-30:
@susisstrolch - Hi, yes you would have a better idea than me being a minimal script writer
So now this
```
#ifdef PLUGIN_SET_ONLY_SWITCH
    #ifndef PLUGIN_SET_NONE
        #define PLUGIN_SET_NONE
    #endif
    #ifndef USES_P001
        #define USES_P001   // switch
    #endif
    #ifndef USES_P003
        #define USES_P003   // pulse
    #endif
    #ifndef USES_P026
        #define USES_P026   // System Info
    #endif
    #ifndef USES_P033
        #define USES_P033   // Dummy Device
    #endif
    #ifndef USES_P037
        #define USES_P037   // MQTTImport
    #endif
#endif
```

Oxyandy, 2018-03-30:
@TD-er - Speed Improvements ?
Ok I have compiled a firmware, feeling more awake today..
Will have a look..
I have been using 2.0 Dev 4, and very confident it works great, 
Dev 11 was a fail, not tried any in-between 
- but have built an excellent testing environment which allows me to swap / change firmwares with the least effort.. But I have only just got myself all prepared to this level..
My tool box ready, but not yet fully made use of it..
Starting this evening is all going to happen, I am dedicating many, many hours this evening
Cheers
Feedback on those tests will come...
Anytime you want testing and feedback just ask, I will make myself a resource, still preparing !

closed 2019-10-27 by TD-er