letscontrolit/ESPEasy #1206: WD trigger

s0170071, 2018-03-30:
i've got the WD triggered after boot with latest core and latest ESPEasy. 

ESPEasy #1146 merge does work, haven't tested the other versions so far.
I know that info is a bit thin so please do not act on it unless others have the same problem. I'll post if I know more.

s0170071, 2018-03-30:
it may very well be that this was a hick up in the configuration... 

What can I do to push that json based config a bit ? I am wasting quite a lot of time on such problems...

TD-er, 2018-03-30:
You mean what we discussed to use Cereal?

s0170071, 2018-03-30:
Exactly

TD-er, 2018-03-30:
Well pointing to a cheap supplier of time would be nice :)
On a more serious note, there are also other issues with stability like deepsleep + MQTT.
If you could test my last PR, that would be great.

TD-er, 2019-11-04:
Since we already have fixed numerous issues that did cause WD reboots, I will close this one.

closed 2019-11-04 by TD-er