letscontrolit/ESPEasy #1211: Plugin template file and hints on writing new plugins

sentinel073, 2018-03-30:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

This is a proposal:
@TD-er had the idea to have a plugin template file to help those who want to create a new plugin.
I have made an attempt to create a template file. I went through the bugs and wiki and created a list of hints I thought are good to know when developing a plugin. I have also added a few bugs which I think provide good information - I couldn't summarise it though so I have written down those cases.
More information has to be added in the template file.
For some reason I can't upload a file here. 
The file is available on the forum, on [this link](https://www.letscontrolit.com/forum/download/file.php?id=2285) (the forum link is [here](https://www.letscontrolit.com/forum/viewtopic.php?p=26876#p26876)).

Grovkillen, 2018-03-31:
Make this a PR I think.

sentinel073, 2018-03-31:
shall do I do that? How?

Grovkillen, 2018-04-08:
You create a fork, branch that with a informative name. Upload the files and then make a pull request.

sentinel073, 2018-04-17:
I made the Pull Request. Please review it as it probably needs more updates. When checking the past issues, I found the following which I think are useful too. I just couldn't figure out what to write:
https://github.com/letscontrolit/ESPEasy/issues/914
Problems with the _P053_PMSx003 Dust Plugin

https://github.com/letscontrolit/ESPEasy/issues/892
Make sure all plugins (devices) use the correct sensor type

https://github.com/letscontrolit/ESPEasy/issues/719
DS18b20 incorrect reading after cold start

https://github.com/letscontrolit/ESPEasy/pull/1004
[issue #892 ] Only use sensor type when really needed  #1004 

https://github.com/letscontrolit/ESPEasy/issues/856
illegal instruction  #856 

https://github.com/letscontrolit/ESPEasy/issues/674
Esp easy update temperature to domoticz generate a reboot #674

https://github.com/letscontrolit/ESPEasy/issues/515
rewrite of PLUGIN_WRITE handling as not working with multiple device of same kind #515

tonhuisman, 2022-05-07:
This seems top be solved, so can be closed.

closed 2022-05-07 by TD-er