letscontrolit/ESPEasy #1213: Email notification causes crash and reboot on ESP32

Budman1758, 2018-03-31:
Title pretty much says it all.  This has been true for several of the ESP32 releases.  This is the latest available. Consistent and repeatable.  The exact same setup works fine on ESP 8266.  Here is a slice of the serial log.....

UDP  : Send Sysinfo message
DS   : Temperature: 27.25 (28-ff-2-c5-83-16-3-51)
EVENT: dallas1#temp=81.05
EVENT: dallas1#temp=81.05 Processing time:5 milliSeconds
FILE : Saved /notification.dat <<<<<<<<<<<<<<<<<Right here I pressed the test button
FILE : Saved /config.dat
FILE : Saved /security.dat
EMAIL: Connecting to 192.168.1.13425<<<<<<<It's port 25 in the config.....

220 catmakanic.com ESMTP MDaemon 13.5.1; Fri, 30 Mar 2018 20:42:10 -0700
EHLO catmakanic.com
250-catmakanic.com Hello catmakanic.com, pleased to meet you
250-ETRN
250-AUTH LOGIN CRAM-MD5 PLAIN
250-8BITMIME
250 SIZE
AUTH LOGIN
334 VXNlcm5hbWU6
CORRUPT HEAP: Bad tail at 0x3ffe5d64. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c", line 199, function: multi_heap_free
abort() was called at PC 0x4010f807 on core 1

Backtrace: 0x400883f8:0x3ffd59c0 0x400884f7:0x3ffd59e0 0x4010f807:0x3ffd5a00 0x40088109:0x3ffd5a30 0x40084262:0x3ffd5a50 0x400847e5:0x3ffd5a70 0x4000bec7:0x3ffd5a90 0x401085b8:0x3ffd5ab0 0x401085e5:0x3ffd5af0 0x400ee5a9:0x3ffd5b10 0x400f1d0c:0x3ffd5b60 0x400f1f7a:0x3ffd5c00 0x400f1fdb:0x3ffd6030 0x400e58e5:0x3ffd6050 0x400e63e3:0x3ffd65a0 0x4016a0b5:0x3ffd65c0 0x401011f3:0x3ffd65e0 0x40101243:0x3ffd6600 0x40101279:0x3ffd6630 0x401014bd:0x3ffd6670 0x400f4f27:0x3ffd66b0 0x400feee8:0x3ffd66d0 0x401529cc:0x3ffd6700

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
▒U

INIT : Booting version: (custom)
INIT : Cold Boot
FS   : Mounting...
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK
INIT : Free RAM:168656
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
EVENT: System#Wake Processing time:4 milliSeconds
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy 32-0 with address 192.168.4.1
WIFI : Connecting xxxxxxxxxx attempt #0
OTA  : Arduino OTA enabled on port 8266
EVENT: System#Boot
EVENT: System#Boot Processing time:2 milliSeconds
WD   : Uptime 0 ConnectFailures 0 FreeMem 110844
DS   : Temperature: 27.13 (28-ff-2-c5-83-16-3-51)
EVENT: dallas1#temp=80.83
EVENT: dallas1#temp=80.83 Processing time:1 milliSeconds
WIFI : Connected! AP: xxxxxxxxxx (38:D5:47:BD:B7:F8) Ch: 0 Duration: 3060 ms
WIFI : DHCP IP: 192.168.1.115 (ESP-Easy-32-0) GW: 192.168.1.1 SN: 255.255.255.0   duration: 12 ms
EVENT: WiFi#Connected
EVENT: WiFi#Connected Processing time:6 milliSeconds
NTP  : NTP send to 2.pool.ntp.org (216.218.254.202)
NTP  : NTP replied: 30 mSec
Current Time Zone:  DST time start: 2018-03-11 02:00:00 offset: -420 minSTD time start: 2018-11-04 02:00:00 offset: -480 min
EVENT: Time#Initialized
EVENT: Time#Initialized Processing time:4 milliSeconds
EVENT: Clock#Time=Fri,20:42
EVENT: Clock#Time=Fri,20:42 Processing time:3 milliSeconds

TD-er, 2018-03-31:
Judging the log, it may be the base64 encoder is changed between ESP32 and ESP8266 library.

TD-er, 2018-03-31:
What is the length of the entered password?

Budman1758, 2018-03-31:
Password length is 7 lowercase letters. That connects to my server running SMTP daemon.

TD-er, 2018-04-01:
I've tested on the ESP32 and indeed confirmed.

tonhuisman, 2022-04-18:
This seems to have been fixed a long time ago, so can be closed.

closed 2022-04-18 by TD-er