letscontrolit/ESPEasy #1214: add a new section in the forum to show our real projects

giig1967g, 2018-03-31:
Hi,
I would like to suggest to create a new section in the forum that will allow us to upload photos and configuration of our real projects. 

It could be a repository of ideas in terms of possibilities of configuration, rules and devices and in terms of mechanical casing and design.

Is this the correct place to ask for this ?

mstjerna, 2018-04-01:
I think this is a great idé. Why not just create a new topic on the forum

Grovkillen, 2018-04-01:
None of us involved with the FW have the rights on the forum. Open a thread and send PM to the admins, on the forum.

Good idea though :)

closed 2018-04-01 by Grovkillen

papperone, 2018-04-02:
don't we have such session already on the wiki ? --> https://www.letscontrolit.com/wiki/index.php/UserProjects

Grovkillen, 2018-04-02:
I guess it's a bit easier to just upload to the forum...?