letscontrolit/ESPEasy #1215: [ESP32] Cannot login to accesspoint on new install

TD-er, 2018-04-01:
The ESP32 version does only work when the SSID and password are being set as default at compile time.
Not when ESPeasy changes to accesspoint mode to aid configuration.

TD-er, 2018-04-01:
It appears to be not specific to the ESP32. See this [forum topic](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=5139)

TD-er, 2018-04-02:
See PR #1220

pwassink, 2018-04-04:
Using Serial commands you will be able to get it online also,  

Just built an esp32 mega, found same problem, 
Input of the WifiSSID,  WifiKey, WifiConnect and Save .. did the trick

TD-er, 2018-04-04:
I know, but that's not how it should work.
And the ESP32 is way more forgiving with STA and AP mode compared to the ESP8266.
If you look at my PR, you'll see the ESP32 just works (perhaps not the save part from the setup page, haven't tested that last night)

pwassink, 2018-04-04:
It did not work in my case, that's why i ended here, looking for this issue.

Freshly built esp32 Mega, Virgin esp32-devboard, latest gitsources merged, build and upload ok, reboot, came up ok, but refused to serve the initial login page at 192.168.4.1 several times.

TD-er, 2018-04-04:
With this pull request also used? https://github.com/letscontrolit/ESPEasy/pull/1220

pwassink, 2018-04-04:
I am not sure, i am new to this git/platformio combination, did a merge about an hour ago.
highest i could find locally is #1218 so probably not ?

TD-er, 2018-04-04:
The pull request is not yet merged, that's why it is still a pull request ;)
But it is not yet stable enough to merge, so I mis-use the PR feature to let others test along with me.

pwassink, 2018-04-04:
Sounds logical :-) 
Well after some time I was able to find your pull-request #1220 / 1215, able to compare it in the web-git acces with my copy online there. The compare result told me i was able to merge that, wow. 
And there it stops. Was not able to get a local copy of that pull-request with my local git-gui either. Have to spend some more time reading i guess. 

Sorry for now, will study some more.

Grovkillen, 2018-04-04:
I simply download the source from @TD-er s GitHub page. Make sure you choose the particular branch and then download the zip (green button).

Grovkillen, 2018-04-04:
Direct link to branch: https://github.com/TD-er/ESPEasy/tree/bugfix/SoftAP_event_based_wifi

TD-er, 2018-04-04:
@pwassink I think a lot of people would like to have some crash-course in Git use.
I see the same questions over-and-over again and I remember my first (year of) experience with Git, that was quite a feeling of helplessness. Even though I've been programming computers since I was 7 years old and am now close to being 42.
Maybe we should look into that, since it will save all of us a lot of time and the people who want to start programming and contributing a lot of time and frustration.
Knowing the tools is very important. Even more important than knowing how to program :)

pwassink, 2018-04-04:
@ grokillen, That will work, downloaded that zipfile now, will try this evening !

@td-er, Yes could not agree more, This guy was used to create scripts, c-code, html and watsoever in plain Vi, so mouse's and gui's and "visual" ide's or languages are still not my favorite items, learning curve is quite steep, and it also took a while after the switch to platformio to collect enough motivation to even try that after mastering the esp-easy 1.* versions and the arduino-ide...
But stay tuned. will give this a try.. Thanks for patience.

Budman1758, 2018-04-05:
Just to confuse things more for everybody I just flashed a unit successfully, it restarted fine and presented itself as ESP_Easy_0 (pretty sure) in AP mode. I connected just fine to it. Entered my wifi creds. It rebooted and connected to my wifi on the first try and I was able to access the web pages as supposed to. Config proceeded just as it's supposed to.
Only 2 things...
#1. When it connected to my wifi the page that shows its new IP appeared but it was blank beneath the header.
#2. When the page showed for access points there was only 1 SSID listed and it was not mine which should have been at the top of the list. (based on signal strength)  I checked "other", filled in the SSID name and password and like I said it connected immediately.

Budman1758, 2018-04-05:
32

TD-er, 2018-04-07:
@Budman1758 On the ESP32 it already just worked and now it should work also on the ESP8266.

I've been looking like forever, but it appeared there is something very strange with Core 2.3.0
So please test with my latest updated PR for this.
https://github.com/letscontrolit/ESPEasy/pull/1220

Budman1758, 2018-04-08:
@TD-er I still haven't mastered the art of compiling the firmware. I would be more than happy to test anything you need if you could provide a bin file.  Sorry about that. ;-)

pwassink, 2018-04-17:
Yes, in the esp32 combination i had the fault it it worked now
.

closed 2018-04-17 by Grovkillen