letscontrolit/ESPEasy #1216: Long Standing Bugs, Part #1 - GPIO getting stuck & Random Rule Firing.

Oxyandy, 2018-04-02:
First of all I have spent many hours preparing this post, this was done to absolutely confirm what I am seeing is not just 'me' or 'my hardware', but really as a result of source changes.

The challenge:
Take a ride with me in my time machine and produce a v2.0.0-dev6 as it should have been !
A bug was introduced between v2.0.0-dev5 & v2.0.0-dev6 and it still exists today.
I have thought long & hard about the best way to handle it, if a Dev 6 can be created with the minimal of changes, then confirmed **fixed** by me... 
Only then can the current source be looked at see how/why it still exists.

The symptoms are:
1. When adding a publish command in the middle of a rule which watches a GPIO, entries appear which are completely unexpected.
2. The GPIO state gets stuck as '1' or shows as the wrong 'true' state on the Devices page.
3. Changes in the state are not correctly reported to my Dashboard on Domoticz. (Not matching Devices)
4. Once stuck, manually moving switch makes no change.
_### Steps to reproduce
How can we trigger this problem?_
Oh boy, a tough one I am afraid...
While I can confirm without ANY doubt, Dev 4 & 5 works (not tried older), 
Dev 6 Fails & the problem/s has been introduced.
I am hoping an experienced coder & an experienced coder with ESPeasy source 
can do a DIFF on the source changes between Dev5 & Dev6 and see if anything stands out.

_### System configuration_
Sonoff Basic
A float switch across GPIO14, with a rule to stop an On Clock#Timer early - once float switch changes state.
Now, I 'was' going to post SO much more, but out of fear of 'information overload', thought, re-reading what I have written here, narrowing the time frame down, explained the symptoms in words etc..
Reckon 'an initial' look over, a DIFF between Dev5 & Dev6, in the relevant source files which could possible cause the symptoms I have been seeing - is a damn good start.
**Standing by**
Ask me anything ? Produce a file for me to test ? Upload something ? Show logs ?
Ready to assist in any way...

[Changed_Files_From_Dev5_++_ESPEasy_v2.0.0-dev6.zip]
(https://github.com/letscontrolit/ESPEasy/files/1866911/Changed_Files_From_Dev5_._ESPEasy_v2.0.0-dev6.zip)
Any file which is different to Dev5 is in this archive, all matching files were removed.
Only the changed files in this archive.

TD-er, 2018-04-02:
But you did describe something of the problem you're seeing.
Just a question. The problems you're seeing, are they related to a ESP8285 build?
See this piece of code:
https://github.com/letscontrolit/ESPEasy/blame/5d41bf9e738655aae146183da050fa0e37b137c1/src/WebServer.ino#L2383-L2423
I've seen this strange line 2412 and it is on my to-do list to look at it implications.
It has been in there for a year now (to the date).

closed 2018-04-02 by Oxyandy

Oxyandy, 2018-04-02:
```
  optionValues[6] = 5;
  optionValues[7] = 7;
  optionValues[8] = 7;
```
Umm strange sequence..
I use 8266EX based Sonoff basic modules.
The reason I wiped this post out was simple really, before I mastered OTA & built a complete set of OTA friendly firmwares to suit my Sonoff, I was flashing via serial.
This meant the Dupont Cables connectors were getting a work out..
Strange things they are - sometimes you turn them 90 degrees and then they make awful connections..
in/out so many times they get weaker & make a poor electrical connection..
Weakened every time module was removed for serial cable flashing...
Well after I made this post here, I flashed the module I had been working with all day today with the same firmware version I had on a stable system - Dev4 - it has been working perfectly for days.
Soon as I did that - I saw the devices page showing my float switch as '1', 
checked those Male & Female Dupont Cables I was using to join switch wires to the module & sadly one was not as tight as I would have expected..
These were only every going to be temporary , once I had things working right, I was going to use better.
I now no longer trust anything I did today.. 
So annoyed with myself, have done electronics since a small child, I should have known better 👎 
Now soldered & heat shrinked as OTA means it can stay where it is..
Cant get a better physical electrical connection now.

[image]
The symptoms I described were correct, only this time I am sure some of what I was seeing was my own fault, never mind - each time I learn something new and move onward..
returning to OTA upgrade, test ? Upgrade, test ?
Glad your side of the world was sleeping when I posted this :D

Oxyandy, 2018-04-07:
@TD-er for few days I have been using and testing Dev_13
GPIO14 most definitely changes from '0' to '1' where it should not..
Somewhere during the cycle ?? 
When it does, it does not report the change from '0' to '1'
But shows as '1' on device..

Did you say you joined in at Dev_12 👍 ?? haha
Still hope for you yet.. 
Now 'finally' downgraded the firmware on the device to Dev_12...
Going to watch and see for a while, so nothing to report yet..
Still been testing last few days, just busy doing other things, so have not had the time to devote to this that I should have...
If problem goes away with Dev_12, I will return to Dev_13, 
if problem comes back with Dev_13 again..
I will return to Dev_12..
Problem gone with Dev_12, then a report will come !
Otherwise step down to Dev_11 and try that..
I will make no dumb or should I say 'rushed' observations
Cheers

Oxyandy, 2018-04-07:
Almost a conclusion: 
At least I know, the strange behaviour starts in Dev_11...
Dev_10 is ok
I ran out of time and had other tasks to do. So left things run.
My stable system was running Dev_10 for days, the best log of them all, is my reservoir full of water.
It should stay full of water when everything is working ok and only slowly lower.
I changed my stable system from Dev_10 to Dev_11 yesterday, according to logs, somewhere yesterday
it dropped wifi & has not been seen on wifi for at least 12 hours...
>ping 192.168.0.201 = unreachable.
Late yesterday afternoon, I looked at both reservoirs, 2 different systems & sadly both were empty.
So both failed on Dev_11
Log of float switch in Domoticz, 
On Clock times *00, *15, *.30, *.45 (every 15 minutes) using On Clock rule 5 minutes later for off
float switch ends that early about 3 minutes after ON time, is a bit different to Dev_12, the GPIO isnt stuck in device page, instead it seems tied to my ONClock OFF
NORMAL float rises '1' = ON, 2 seconds later, float lowers '0'=OFF
This log is just the float switch.

```
2018-04-08 05:02:59 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
2018-04-08 05:02:59 | On
2018-04-08 05:03:01 | Off <- Float SW lowers changing from 'open circuit' to 'closed circuit'
2018-04-08 05:03:01 | Off
2018-04-08 05:18:07 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
2018-04-08 05:18:07 | On
2018-04-08 05:18:09 | Off <- Float SW lowers changing from 'open circuit' to 'closed circuit'
2018-04-08 05:18:09 | Off
2018-04-08 05:32:58 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
2018-04-08 05:32:58 | On
2018-04-08 05:33:00 | On <WRONG should be OFF
2018-04-08 05:33:01 | On
(Missing *.45 ON cycle, remember firmware does not log change from wrong '1' to correct '1')
2018-04-08 05:50:04 | Off <WRONG -unexpected time, this is my OnClock OFF
2018-04-08 05:50:05 | Off
2018-04-08 06:02:58 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
2018-04-08 06:02:58 | On
2018-04-08 06:03:00 | On <WRONG should be OFF
2018-04-08 06:03:00 | On
(Missing *.15 ON cycle, remember firmware does not log change from wrong '1' to correct '1')
2018-04-08 06:20:04 | Off <WRONG -unexpected time, this is my OnClock OFF
2018-04-08 06:20:04 | Off
```
 <WRONG -unexpected time, this is my OnClock OFF (but is stopping stuck '1')
No way the switch ever physically moved at this time, 
the switch movements all happen around 3 minutes after ON time
But this log (<WRONG -unexpected time, this is my OnClock OFF)
is not in Dev_12, once the GPIO14 is '1' in Dev_12 it stays '1' on device page.
Dev_11 is more subtle, I thought it was OK, because I was not seeing stuck '1' in device page like Dev_12
BUT It's change from '1' to '0' some how tied to my OnClock OFF ?
Is this new ? More closer observation required.

Oxyandy, 2018-04-07:
@TD-er Dev_4, Dev_5, Dev_6, Dev_7, Dev_8, Dev_9, Dev_10 
work as expected......
Dev_11 problems starts, all the way through to the most recent fimwares 
(not tried absolute latest releases as I have been using old firmwares)
What next ? 
Pinpoint fault in Dev_11 & make a working Dev_11
DIFF changes from Dev_10 to Dev_11

Oxyandy, 2018-04-07:
Another clue ?
The not responding to WiFi Dev_11 system.
I went out pump was ON, manually moved float up/down -  up/down - up/down
It did not stop the pump, but seems On Clock rules work...

Oxyandy, 2018-04-07:
Yeah ok TD-er.. I hope you can see I am doing my best to supply as much info as possible, something in what I say maybe a clue..

TD-er, 2018-04-07:
You are certainly supplying as much info as possible, but at the moment it is a bit too much for me to parse and grasp it all. So maybe if you could summarize it a bit, it will make it easier for me to process it.
Something like _rubber duck debugging_ and consider me the duckling on the corner of the keyboard. :)

Oxyandy, 2018-04-07:
No plugins really: 
A Sonoff basic
a float switch on GPIO14, as a switch
The internal relay on GPIO12 as a switch
[image]
The last post showing switch log is really a summary, shortened version, when switch changes, it sometimes gets stuck as '1', the next cycle being falsely '1' causes problems 
(see switch log for better understanding)
It is expected that a change in the switch results in firmware changing it state of switch too
in other words switch goes from '0' to '1' OK, next change results in '1' to '1', next cycle starts wrongly '1' = problems

Oxyandy, 2018-04-07:
Here, this must be GPIO14 starting at correct state, finishing at wrong state.
```
2018-04-08 05:32:58 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
2018-04-08 05:32:58 | On
(take 2 seconds for float switch to lower)
2018-04-08 05:33:00 | On <WRONG should be OFF
2018-04-08 05:33:01 | On
```
Confused ? So is firmware..
And this does not happen every time....
Although it doesn't happen with older firmwares..
I could try running device on different power source.
Reverse switch 180 degrees so it is normally an 'open circuit' - goes closed on risng, returns to 'open'
But why does it work before Dev_11, stop after Dev_11 ??

Grovkillen, 2018-04-07:
I've seen this problem with gpio 14 together with switch plugin/task (might be affecting other io as well, don't know). My dehumidifier uses gpio 14 for the relay controlling the pump. I use a task to monitor the state of the relay/pump but the value get stuck on "1" even if pin state page say that gpio 14 is "0". I got it working by using taskvalueset to force the task value to be updated but yes, switch plugin is not working reliable.

Oxyandy, 2018-04-07:
Rules1```
on button#switch do
if [button#switch]=1
gpio,12,1
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"MPump_2_ON_%systime%"}
else
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"MPump_2_OFF_%systime%"}
endif
endon

on system#boot do
if [Float_SW_2#switch]=1
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"BOOT_FLOAT_2%systime%"}
else
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"v2.0.0-dev11_%systime%"}
endif
endon
```
Rules2
```
on Float_SW_2#switch do
if [Float_SW_2#switch]=1
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Float_SW_2_1_%systime%"}
else
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Float_SW_2_0_%systime%"}
endif
endon

on Pump_2#switch do
if [Pump_2#switch]=1
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Pump_2_ON_%systime%"}
else
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Pump_2_OFF_%systime%"}
endif
endon

```
Rules3
```
on clock#time=all,**:00 do
gpio,12,1
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_ON_2_%systime%"}
endif
endon
on clock#time=all,**:05 do
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_OFF_2_%systime%"}
endif
endon

on clock#time=all,**:15 do
gpio,12,1
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_ON_2_%systime%"}
endif
endon
on clock#time=all,**:20 do
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_OFF_2_%systime%"}
endif
endon

on clock#time=all,**:30 do
gpio,12,1
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_ON_2_%systime%"}
endif
endon
on clock#time=all,**:35 do
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_OFF_2_%systime%"}
endif
endon

on clock#time=all,**:45 do
gpio,12,1
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_ON_2_%systime%"}
endif
endon
on clock#time=all,**:50 do
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Timer_OFF_2_%systime%"}
endif
endon
```
Excuse the extra fluff in rules, is for debug purposes and has taught me lots.

Oxyandy, 2018-04-07:
The 2 devices pages stitched together
[image]

Grovkillen, 2018-04-07:
And you have deactivate serial on gpio3? And the button in gpio0 is always high on boot?

Oxyandy, 2018-04-07:
Yes, deactivated serial, otherwise GPIO3 can not be assigned.
that screenshot was taken when a cycle was active, GPIO 0 is always 0 at boot
everything at boot is '0'
Low_Res_2, should be '1' - Float_SW_2 as '1' is wrong... again
[image]

Oxyandy, 2018-04-07:
Changing boot state, seems to make no difference.
[image]

Oxyandy, 2018-04-07:
@giig1967g  - I can upload a whole back-up fimware 'as is'
I use static
IP 192.168.0.202
SUB 255.255.255.0
GW 192.168.0.254
Can change these and Wifi before I make a copy ?
I am open to anything.. just ask

Oxyandy, 2018-04-07:
And so, earlier I said this:

> If problem goes away with Dev_12, I will return to Dev_13,
> if problem comes back with Dev_13 again..
> I will return to Dev_12..
> Problem gone with Dev_12 'again', then a report will come !
> Otherwise step down to Dev_11 and try that..
> Dev_10 is 'confirmed' as not having the problem, so really, really close ;)
> I will make no dumb or should I say 'rushed' observations
> Cheers
'Doing' this ^^^^ means my observations are absolutely 'confirmed.. 
Well should be... If I stick to that.
This has been a tough problem to **_absolutely_** 'pin down'...

My stable system stopped responding to WiFi yesterday afternoon...
I thought I had OTA upgraded it to Dev_11 ??
But yesterday I had to stop playing IoT.. (life / wife) gets in the way.
I just cold booted it now..
It is not Dev_11 it's actually Dev_10 - however my logs show it was Dev_10 since the 4th..
It failed yesterday, and as I am becoming quite an expert at observation, I now have the vague feeling it is possible I tried an OTA, so I could test Dev_11 on this 'stable system', did that and ran away to do other tasks that keep wife happy, haha
Never try an OTA while a task is active, it can lock up the device. Did I do this ?
EG. If pump was on while I was attempting OTA, it would lock up & pump would stay on, flooding system, overflowing & emptying reservoir. Anyway, doesn't leave much room for doubt because, I can say before yesterday, as Dev_10 it was stable - it has been Dev_10 since the fourth of April...
If Dev_10 was an issue like Dev_11, I would have noticed it sooner.

To make my physical operation 'safe' - I think I need to add a second float switch, with a second Sonoff, it's purpose is to detect when level is too high (before it overflows) & cut power to the 'Main' Sonoff that controls the pump, long enough to make it cold boot, then start again.
I was thinking of doing this via a command via Wifi, but what if 'Main' has locked up ? It wont respond. 
I need to 'cut power' to main.
Shame a Sonoff does not have NC relay contacts, just open those for short time, pump & device would stop, then a few seconds later, cold boot.
At least this way I will never lose all the water in the tanks..

Tests & observations will continue, refilling tank running Dev_10 now, cold boot and allow to run for days without any interference.
Take system 2, which failed on Dev_11 & downgrade to Dev_10 to test..
Confirm stable ? Yes?
Use Dev_11 again...
Double, triple, what ever, but CONFIRM everything to death.

Oxyandy, 2018-04-08:
2 systems, 6 hours+ running Dev_10 and both doing fine..
I need to go away in a few days for 5 days & these 2 systems have plants..
That gives a bit of time to confirm stable before I leave for a mini holiday..
Back next weekend, no more posts from me says "All OK !"

giig1967g, 2018-04-08:
Hi @Oxyandy,
I have 4 comments:
1) Please send your config file, removing passwords.
2) How can you disable serial (to use GPIO3)? 
3) Is this rule correct (you set gpio,12 to 0 in both cases):

```
on Float_SW_2#switch do
if [Float_SW_2#switch]=1
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Float_SW_2_1_%systime%"}
else
gpio,12,0
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Float_SW_2_0_%systime%"}
endif
endon
```
4) O am sure if this is yours case too, but my experience is that since version 2 (but could be dev10 as you say because I joined at dev12), if you use the switch plugin task as a Output it eventually crashes. 
Probably because as the plugin name says - SWITCH INPUT - the plugin is supposed to control INPUT only and not output. 
I have opened several forum requests, but found no solution.
So I ended up using a Dummy task to keep trace of the status of the OUTPUT pin, as suggested also by @Grovkillen .
That is rock solid.

Oxyandy, 2018-04-08:
1.  [config.dat.zip](https://github.com/letscontrolit/ESPEasy/files/1887919/config.zip)
2.
[image]
3. Yes, my Float Switch is meant as an OFF only.
So it was a very simple rule, but because of issues, I added extra to report the state of GPIO14 to a custom log. And want 'any change' in the Float Switch to only turn off, never on
the extra gpio,12,0 never helped, but did no harm so was left there :) 
```
on Float_SW_2#switch do
if [Float_SW_2#switch]=1
gpio,12,0
endif
endon
```
4. I have no experience with a "Dummy Task" yet, any examples ?
Cheers
*****Dev_10, 
just checked uptime 23 hours, tank levels are as they should be, there has been no overflowing. 👍

giig1967g, 2018-04-08:
Hi @oxyandy
I haven't been able to run your firmware, as I couldn't change the IP address (probably because the serial port is disabled... or maybe AP is disabled...)

As I understand, you have three INPUTS (GPIO14,3,0) and one OUTPUT (GPIO12).
If so then try this:
a) delete task number 4 (the GPIO14)
b) add task number 7 as a DUMMY device, as per image attached (but keep task number = 7)
c) modify the rules as per attached file: [rules.txt](https://github.com/letscontrolit/ESPEasy/files/1887958/rules.txt)
d) reboot, doing a power off cycle.
e) should work :)

[image]

Oxyandy, 2018-04-08:
Excellent, I had been meaning to look deeper into events, 
that example just makes it super clear for me to understand !
Yes static is set, I did try to delete those values, but I see a bug when I do - old dev_10
and yeah without serial enabled you can not push commands via serial console..
To get WiFi access, I could set WiFi
SSID TEMP
PASS 12345678
Serial is enabled, GPIO3 disabled
[config.zip](https://github.com/letscontrolit/ESPEasy/files/1887970/config.zip)

giig1967g, 2018-04-08:
Try my suggestion first.
If it doesn't work I will try to load your config.

Oxyandy, 2018-04-08:
```
Rules2

on Float_SW_2#switch do
if [Float_SW_2#switch]=1
gpio,12,0
taskvalueset,7,1,0
event,pump2switch
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Float_SW_2_1_%systime%"}
else
gpio,12,0
taskvalueset,7,1,0
event,pump2switch
publish domoticz/in,{"idx":49,"nvalue":0,"svalue":"Float_SW_2_0_%systime%"}
endif
```
Maybe I am not understanding this right ?
But [Float_SW_2#switch] is my GPIO14, if I delete that how is the switch going to trigger ?
GPIO0 is the button on the Sonoff, manual on/off of Relay GPIO12
GPIO3 is just a sensor for "Low Fluid Level", and for the moment not used.

giig1967g, 2018-04-08:
Delete task number 4, GPIO 12.

Oxyandy, 2018-04-08:
@TD-er  -- In an attempt to make your life just a little easier, here is an archive
Inside is only the changed files from Dev_10 to Dev_11, in this format
ESPEasy_Dev_10.ino
ESPEasy_Dev_11.ino
I assume you would have some sort of visual text tool.
Selecting both, compare...
If you have the time ? Maybe something stands out ?
Dev_10 works, Dev_11 the problem started.

[CHANGED_v2.0.0-dev11.zip](https://github.com/letscontrolit/ESPEasy/files/1888033/CHANGED_v2.0.0-dev11.zip)
Cheers & Goodluck.

giig1967g, 2018-04-09:
@Oxyandy : did you try dev11 with my rules?

Oxyandy, 2018-04-09:
>try dev11 with my rules
Looking at the rule, I see that GPIO14 will still get stuck, it is a great example & teaches me something new. I tried a similar thing before & it did not work.
I will try your rule, but have run out of spare time, when I return it will be the first thing I do... 
But I need to be here to watch it's impact.
Dev_10 works, so while away it will take care of things.
I am not going to be online or be able to test anything until Saturday..
Sorry for delayed response, I was hoping TD-er would see my previous post, 
hope I am not now in ignore mode.
Something changed between Dev_10 & Dev_11 that stopped my scenario from working & every firmware I have tried after Dev_11 has the same problem.
It's a tricky problem & it isnt always there, so have done my best to include as much information as possible to show what I see & it's impact, every clue I can, I can not do more than I have right now, no time.

giig1967g, 2018-04-10:
Hi @Oxyandy ,
the reason why it is important to test my suggestion is try to limit the issue to something more specific.
In other words, if with the new rules and suggestions (delete the OUTPUT task, create a dummy device, etc...) the problem is gone, then the developer can concentrate on a specific behaviour of the plugin, saving time.
Please make this test and report.

Also, which is the more recent firmware version you tested your unit with?

Oxyandy, 2018-04-12:
I am away from home.... On Saturday or Sunday, I will have the time to investigate more..

Oxyandy, 2018-04-16:
I have returned home.. Weather has been awful, so I haven't done much, but getting ready.
I think because I posted 'only the changed files' between Dev_10 & Dev_11, the change should be easy to stop in the source, I made the archive so it can be extracted to a single folder & the changes quickly compared.. Thinking I should open a reduced 'new issue' with much of the information summarised..
```
2018-04-07 11:17:14 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
2018-04-07 11:17:14 | On
(takes float 2 or 3 seconds to lower)-The physical change in the switch still happens at the right time BUT
2018-04-07 11:17:17 | On <- wtf ???
2018-04-07 11:17:18 | On
```

Rules3 is too many characters, so I will need to trim it down to suit...
I still struggle to understand how it will help, the issue is not with the relay but the switch handling..
I have done further reading of other issues & feel it also has something to do with the Domoticz side too..
Is night now & weather has settled, first I will do an indoor test with your re-written rules @giig1967g

giig1967g, 2018-04-16:
@Oxyandy : if rules3 is too long, split them into rules4.

Oxyandy, 2018-04-16:
On my own long and complex review of what I was experiencing with a switch on GPIO14 and yes Domoticz MQTT controller, I tracked an error back to beginning with Dev_11. #1216
I have tested versions prior to Dev_11 all the way back to Dev_4 & those all work just fine.
Versions Dev_10 & prior have proven stable over days on end, Dev_11 fails quickly
Something changed between Dev_10 & Dev_11 and a bug was introduced.
A bug which was is still present in very recent releases (not sure if still present in very latest firmwares)
Are these all related #673 #674 #675 #924?
I created an archive to help
Inside are only the changed files from Dev_10 to Dev_11, all renamed in the format below to make the origin of each file perfectly clear - Extract contents of this archive to a single a folder

> _P029_Output_Dev_10.ino
> _P029_Output_Dev_11.ino

With a visual text compare tool.
Selecting both versions, compare...
If someone has the time ? Maybe something stands out ?
Dev_10 works, Dev_11 the problem started.

[CHANGED_v2.0.0-dev11.zip](https://github.com/letscontrolit/ESPEasy/files/1888033/CHANGED_v2.0.0-dev11.zip)

TD-er, 2018-04-16:
Can you summarize in just a few lines what is happening and what should happen?
So what plugin, what plugin config (screenshot?) and what is connected to this GPIO.
And what should happen and what does happen?

Oxyandy, 2018-04-16:
Yeah @TD-er, I guess I could read over everything I have posted here, 
Open a fresh "Issue" trying to keep everything minimal..
The whole thing is much clearer to me now than it was when I started looking into this

TD-er, 2018-04-16:
Clarity is always nice when describing issues :)