letscontrolit/ESPEasy #1219: Feature request for dummy/virtual switch

easypeasi, 2018-04-02:
I hope I do not offend anyone, I was advised to post this request here as devs do not look at forum.
May I put in a request for a dummy switch, it would save me having to put ugly hardware switches on my devices, for anyone like me that resides all control locally, it would be a great addition. Hoping this is easily achievable.

TD-er, 2018-04-02:
What should be the difference between a dummy/virtual switch and a normal switch?
The option to not configure a GPIO pin?

uzi18, 2018-04-02:
I think it is about to add switch to dummy device :)

TD-er, 2018-04-02:
@uzi18 Maybe I'm a bit fuzzy in my head, but that doesn't make it more clear to me. :(
Is there a link to the forum, perhaps the use case is described there?

easypeasi, 2018-04-02:
More information on what interpretation of a dummy/virtual switch is:
 Configuration of a software switch that can be accessed via config page or dashboard to change state and attach it to a real output or a dummy sensor variable.
Hope I have explained OK.

uzi18, 2018-04-02:
Looks like what I mean but any real life example please?

easypeasi, 2018-04-02:
@uzi18, A real life example:
A device controlling a central heating boiler via time programme and room temperature, today would be called a 'smart thermostat'.
I currently have to use a hardwired push button to override the time programme, it would be much neater to use a dummy/virtual switch, and everything is controlled from your smartphone, with a user friendly dashboard.

Budman1758, 2018-04-02:
Correct me if I'm wrong here but if you are attempting to simply turn the gpio on or off you can already do that with an http get request. What exactly is your manual push button doing? How does your button "override" the time program?

elmicha, 2018-04-02:
I think you could use the built-in [Mini Dashboard](https://www.letscontrolit.com/wiki/index.php/Mini_Dashboard) (which I didn't know until now). 

Take the HTML after "A sample web page for a light control switch: " and put it in a file called switch.esp. Remove the `dashboard.esp` from the two `href`s (so it's e.g. `href="?cmd=gpio,12,1"`). Obviously adapt the needed gpio settings and the [Switch#State]. Upload the file as switch.esp with Tools > Flash. Find two nice images of white and yellow light bulbs, and upload them, too, naming them "Lamp0.png" and "Lamp1.png" (or remove the img tag).

Then you can use http://your-esp-easy/switch.esp to see the dashboard.

Another option would be a plain old HTML page with Javascript and CSS. Upload all the files (the HTML page needs to have a .htm suffix, not .html, I think), then you can call e.g. http://your-esp-easy/switch.htm. You can get the sensor values from /json. But the Mini Dashboard is probably easier, as it already knows about the [Switch#State] etc.

easypeasi, 2018-04-03:
@Budman1758 & elmicha, thanks for your responses.
@Budman1758 The manual pushbutton is used in the logic within the rules engine, without going into too much detail, basically if time schedule is off, and the space is below temperature setpoint then boiler output is on until either of those inputs change state.
I am not attempting to change state of a 'real world' gpio, I do not know if it is even possible without assigning a pin to it, and if did work it would be a shame to use a gpio that may be needed to connect to the real world.
Think of the virtual switch as a variable (0 or 1) that can be changed by gui or rules, it may be that the existing dummy sensor plugin can be used, and just add an enhancement so the variable can be adjusted from gui?

TD-er, 2018-04-03:
The existing switch does support these things, as far as I know.
Not sure whether it is possible to assign a "non existent" pin.

Budman1758, 2018-04-03:
So this "thermostat" has no physical connection to the "boiler" its controlling?  If not then how is it controlling it?  Is there another ESP connected to the boiler?

easypeasi, 2018-04-03:
@Budman1758 Just one ESP, 3 GPIO's used, DHT22 on one, override switch on the second and boiler relay on the third.
Yes I can do whatever I want in rules , but how would you get your wife, tennent, next house owner etc. to change the values in rules, it has to be user friendly so anybody can access and understand it, as a smart thermostat would be.
I am looking at the possibilities of using the Mini Dashboard, do you think I could create a switch there that would directly override the 'Output-Clock state, which by the way has 1st GPIO (None) as it is a virtual output, used in rules only.

Budman1758, 2018-04-03:
Sounds like you need an interface more than any change in ESPEasy.. All "smart" thermostats and other devices like that have their own interface to interact with.  ESPEasy has the web configuration pages.  If you need an elegant way to present it to your wife or tenant you will need to create your own or find one that can send the web requests necessary to control it.  You do not change the rules to achieve what you want. You set it up to do what you want once and use web requests to "fire" events to do what you want. There are lots of smartphone apps that can do this. Probable a lot for tablets too.

easypeasi, 2018-04-03:
@Budman1758 No I am happy with Mini Dashboard. It would appear you get the general idea.
Espeasy is a clever firmware, and I am sure it would be quite capable of presenting the data in a user friendly way with the Mini Dashboard.
You did not respond to my question about using the Mini Dashboard, does that mean you do not think it is possible?

Budman1758, 2018-04-03:
I have not used the mini dashboard so I'm not qualified to answer that.

uzi18, 2018-04-04:
@easypeasi please show us Your configuration and dashboard.
I'm realy curious how it looks like and how it work 👍

TD-er, 2019-11-04:
Is this still a valid request?
We now have variables to store stuff.
Also a dummy plugin can be set to output in some format to send it to a controller, just like a switch.
Values can be set from rules via `taskvalueset` and executed via `taskrun` to send to a controller.

Is that enough for what was originally intended?

I will close the issue now, since it has been inactive for 18 months, let me know if it is still needed.

closed 2019-11-04 by TD-er

linuxhublinuxhub-dev, 2020-10-22:
I always get this no matter what

URI: /edited.esp
Method: GET
Arguments: 0

TD-er, 2020-10-22:
Can you give an example of the URL you're using to get this reply?

linuxhublinuxhub-dev, 2020-10-26:
http://ip_of_esp/edited.html
or
http://ip_of_esp/edited.htm 
or 
http://ip_of_esp/edited.esp

the file name is edited.htm

TD-er, 2020-10-26:
What build are you using?

linuxhublinuxhub-dev, 2020-10-26:
mega-20191208

TD-er, 2020-10-26:
Well, there have been "some" fixes in the past 10 months, so please test also with the most recent build.

linuxhublinuxhub-dev, 2020-10-27:
thanks I will give it a go