letscontrolit/ESPEasy #1222: Munin Plugin to monitor ESPEasy states

soif, 2018-04-04:
For those using the well-known, easy to use and excellent [Munin](http://munin-monitoring.org/) , I've developed a [**Munin plugin**](https://github.com/soif/munin_plugins/tree/master/iot-json_) that works out of the box to monitor ESPEasy devices:
- cpu load, + LC (thank to PR #1217 now merged)
- uptime
- free RAM
- RSSI
- and any sensor value in the JSON

This is **VERY useful** when debugging plugins to watch when/why ESPEasy crashed... Or to easily watch any sensor value evolution druring tome

soif, 2018-04-04:
Here is a screenshot of one of my ESPEasy device:
[image]

soif, 2018-04-04:
using this, i can easily see that
- CPU load is most of the time 100%
- RAM does not vary that much
- Device reboot often, using a repetitive pattern.

Pretty informative isn't it?

TD-er, 2018-04-04:
Please note that the CPU load is a bit tricky, since it does an estimate on how many times the loop() is run.
This isn't constant for the same CPU load.

But still this is quite informative :)

Grovkillen, 2018-04-06:
Is this something you could write for the wiki? I don't see this as a request of issue per say?

soif, 2018-04-06:
>I don't see this as a request of issue per say?

As this is mainly intended  to be useful for developers, I posted it at GH rather than on the forums.

>Is this something you could write for the wiki? 

Just unlock this Github Repo wiki 😉 😜

Grovkillen, 2018-04-06:
Nah, I'll add it to the user wiki.. this is a really great service and I will start using it for sure! :)

closed 2018-04-06 by Grovkillen

soif, 2018-04-06:
Great! 
If you use it and experience it, you're welcome to enhance the included ReadMe.md (PR) that explain how to use it (this a bit tricky for any one not already used to Munin, and virtual nodes)