letscontrolit/ESPEasy #1223: Rules and timer ESP32

flexiti, 2018-04-04:
it looks like the timer is inaccurate and unstable. I do not know if it's the same for ESP8266
Is this normal?

Simple example:

```
On System#Boot do    

  timerSet,1,1      //Set Timer 1 for the next event in 1 seconds
endon

On Rules#Timer=1 do  
  gpio,2,0
  timerSet,2,1       //Resets the Timer 2 for another 1 seconds
endon

On Rules#Timer=2 do  
  gpio,2,1
  timerSet,1,1       //Resets the Timer 1 for another 1 seconds
endon
```

TD-er, 2018-04-04:
How much is the timer off?
A factor of 2?

Does the reported uptime (with NTP disabled) also show a constant offset?

I've noticed the ESP8266 boards to show a drift of up to order 10 seconds per 24h at room temperature.
The one hanging outside didn't have a firmware with accurate timing data during the winter. :(

TD-er, 2018-04-04:
Please have a look at the set frequency.
It is shown in the Tools => info page.

The thing is, the ESP32 can operate at 80 and 160 MHz (and perhaps some more) and I think it can be set somewhere during compile time.
If for some reason the compile time doesn't match the real clock speed, you'll get quite some offset.

But I've not used the ESP32 for real. I just made it compile and tested some WiFi issues.

flexiti, 2018-04-04:
unfortunately there is no speed information there. Tomorrow I will do an "investigation" :)

Budman1758, 2018-04-04:
I'm using the build thats available on the wiki page and there is no "set frequency" at all.
Have not timed the timers yet but it's always showing the correct time. (probably due to ntp enabled)

TD-er, 2018-04-04:
On the ESP8266 there is something like this:

> **ESP board**
> ESP Chip ID | 1649343 (0x192ABF)
> ESP Chip Freq: | 80 MHz

If it isn't on the ESP32, then there is probably not a similar command for the ESP32 (or a known command)

Budman1758, 2018-04-04:
[image]

Budman1758, 2018-04-05:
Did some random timer testing and for the most part seemed to be as close as possible to accurate considering that ESPEasy is not a "realtime" OS. Maybe a second longer every so often. Never shorter.
Nothing as far as a factor of 2. 10 to 30 second timers. Latest ESP32 release.
Its been said here ind in the forums that if you need split second, absolute timing accuracy ESPEasy is not for you.  Seems accurate enough for most purposes IMHO.  :-)

flexiti, 2018-04-05:
I described the problem, because the time of 1s for the processor ESP32  is long, if it works on threads, it seems to me that something is wrong with the system kernel ... instead of 1 s, they are 2 seconds and sometimes freeze longer

flexiti, 2018-04-05:
but this rule in this example is only one real job for this powerfull procesor.
So I ask what this processor is still doing besides switching threads, that there is no time to properly handle timers in rule in example above

flexiti, 2018-04-13:
ESP32
This is log for rule example above
instead of a second there are two, at other times there is also a discrepancy

```
**2265365** : EVENT: Rules#Timer=2
2265392 : ACT : gpio,2,1
2265393 : SW : GPIO 2 Set to 1
2265394 : ACT : timerSet,1,1
**2267365** : EVENT: Rules#Timer=1
2267371 : ACT : gpio,2,0
2267372 : SW : GPIO 2 Set to 0
2267373 : ACT : timerSet,2,1
**2269365** : EVENT: Rules#Timer=2
```

flexiti, 2018-04-28:
#1315

TD-er, 2018-05-02:
I have changed the way how timers are now being set again.
Could you check if this is an improvement?

flexiti, 2018-05-02:
ok, I will

flexiti, 2018-05-02:
Still 
`if (timeDiff(millis(),RulesTimer[x].timestamp)  < 100)`
works better

Note that anything that is asynchronous to rulesTimers () (is called every 1 second) will be affected by a error. eg. if millis () will be increased in 1001ms, and rulesTimers () call in 1000ms, the next second will be added

TD-er, 2018-10-13:
It should work a lot better now, with the scheduler.
I'll close this one now.
If it is still a bug, please re-open.

closed 2018-10-13 by TD-er