letscontrolit/ESPEasy #1226: Add support for the Wemos ST7735S display

mattlward, 2018-04-09:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Not a problem, would like to know if this display will be supported...

https://www.aliexpress.com/store/product/TFT-1-4-Shield-V1-0-0-for-WeMos-D1-mini-1-44-inch-128X128-SPI/1331105_32848274716.html?spm=2114.12010608.0.0.7a2e719f6EssdU

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:
Wemos D1
Software or git version:

Grovkillen, 2018-04-09:
It's for sure a nice display and if any developer is interested it might get supported. It's SPI communication and a e-paper display using SPI is something that is on the horizon. 

PS. a good way of getting developers attention is to sponsor them with units.

mattlward, 2018-04-09:
I could do that, I am going to order 1 to try and would not be opposed to ordering 2.  Where do you live?  Would it be easier to paypal you for one with shipping?

Grovkillen, 2018-04-09:
I'm not a developer, only wiki, support and tools for deployment from me :wink:

mattlward, 2018-04-09:
Well... the offer stands if a dev wants to tackle this I would furnish a board and display to them.

realwax, 2020-02-15:
+1 on the support of ST7735S :)

PR #3761 (2022-03-31): [P116] ST7735/ST7789/ST7796 TFT Plugin and AdafruitGFX helper
- New plugin P116: ST77xx TFT
  - Supports ST7735s/r 128x128, 128x160, 80x160 pixel TFT 65535 color displays
  - Supports ST7789vw 240x320, 240x240, 240x280, 135x240 pixel TFT 65535 color displays
  - Supports ST7796s 320x480 pixel 65535 color displays
- Introducing `AdafruitGFX_helper` class, as a generic interface for all displays using the Adafruit_GFX library
- Add some optional fonts (1 proportional, 5 fixed width), disabled by default
- Plugin is included in `DISPLAY` builds, that have been set to `LIMIT_BUILD_SIZE` for ESP826 [... cut]
closed 2022-03-31 by TD-er