letscontrolit/ESPEasy #1228: MQTT lost connection and device lost ping for 3 sec after gpio,x,x command

radebebek, 2018-04-09:
Hello all,

I am developing home bridge plugin with mqtt broker for up and down blinds.
So, I want "up and down" blinds with Home App, and also I want "up and down" blinds with with a physical wall switch.
Physical wall switch connected to GPIO0 AND GPIO9 (Sonoff Dual r2, lastes firmware v2.0-20180322)
Control even is [Import#moving] that show // 0 = Stoped , 1=Moving Up , Moving Down.
When app start moving down, App publish 1, after blinds moving finished, sonoff publish 2.
So, I have problem with publish 2. 
For some reason after gpio,12,0 mqtt lost connection and dont publish 2, and also mqtt import dont update state moving to 2.
A couple of times pass without problems, but every some time the connection is lost, and this is always on this part after gpio,12,0 or gpio,5,0. Also when mqtt lost connection, sonoff device lose ping and freezing browser for 3 sec.
----
I make some things easier with mqtt, and never lost connection to the same device son off dual r2. (So, The device is okay)

All Rules Code:
`on Import#moving=0 do 
        gpio,12,1
        Delay [Import#time]
        gpio,12,0
        Publish RoletneMalaSobaWC/currentPosition,[Import#target]
        Publish RoletneMalaSobaWC/statePosition,2
endon

on Import#moving=1 do 
        gpio,5,1
        Delay [Import#time]
        gpio,5,0
        Publish RoletneMalaSobaWC/currentPosition,[Import#target]
        Publish RoletneMalaSobaWC/statePosition,2
endon

on TipkaUp#Stanje=0 do
     if [Import#moving]=2 
        Publish RoletneMalaSobaWC/targetPosition,100
        Publish RoletneMalaSobaWC/time,10000
        Publish RoletneMalaSobaWC/statePosition,1
     endif
endon

on TipkaUp#Stanje=1 do 
     if [Import#moving]=2 
        Publish RoletneMalaSobaWC/targetPosition,100
        Publish RoletneMalaSobaWC/time,10000  
        Publish RoletneMalaSobaWC/statePosition,1
     endif
endon

on TipkaDown#Stanje=0 do
     if [Import#moving]=2 
        Publish RoletneMalaSobaWC/targetPosition,0
        Publish RoletneMalaSobaWC/time,10000
        Publish RoletneMalaSobaWC/statePosition,0
     endif
endon

on TipkaDown#Stanje=1 do 
     if [Import#moving]=2 
        Publish RoletneMalaSobaWC/targetPosition,0
        Publish RoletneMalaSobaWC/time,10000  
        Publish RoletneMalaSobaWC/statePosition,0
     endif
endon`

[https://www.letscontrolit.com/forum/download/file.php?id=2319](url)

Grovkillen, 2018-04-09:
Just curious, how come you didn't follow the instructions on how to create an issue? I had to fix the format quite a bit...

giig1967g, 2018-04-09:
@radebebek : can you post the screenshot of the devices page?

radebebek, 2018-04-09:
@giig1967g 
Screenshot
[image]
If you need under device tell me.

giig1967g, 2018-04-09:
First of all, you are not saying which version of the software you are using.

Second, I suspect that you are having the same problem as https://github.com/letscontrolit/ESPEasy/issues/1216
In other words: you are using an OUTPUT pin inside the Input Switch plugin, that should be used only for INPUT pins. Follow the issue linked to see if other user confirms the issue.

In the meantime, try removing the  SwitchUp (task 3) and SwitchDown (task 5) from the devices list, do a power cold reboot and test to see if the issue goes away.

Then please return your feedback

radebebek, 2018-04-09:
@giig1967g  Thanks for advice, I will try tonight and return feedback. 
In first post I wrote version (firmware v2.0-20180322) or We do not mean the same thing?

giig1967g, 2018-04-09:
sorry I missed the firmware version.

I am not sure but I suspect also that the mega version has been updated with several MQTT improvements that have not been included into the V2.0.

Can you try also latest mega relase?

TD-er, 2018-04-09:
There have been some MQTT updates in de last few weeks indeed.
But if the system is unavailable for so long, please also check the logs using serial output to see if something strange is happening.
The logs for MQTT in case of reconnects are quite verbose.
And also look for uptime of the node. 3 seconds is a bit too fast for a complete reboot + reconnect, but 5 sec. is possible with the newer event based wifi.

Grovkillen, 2018-04-10:
As of later versions I need to have a delay of 250mSec between each Publish command. Not sure if 250 is the exact needed time but it works.
```
on MQTT#Connected do 
  Publish,%sysname%/event,MESSAGE_1
  Delay,250
  Publish,%sysname%/event,MESSAGE_2
endon
```

giig1967g, 2018-04-10:
@radebebek : did you test my suggestion and/or @Grovkillen suggestion?

TD-er, 2018-04-10:
@Grovkillen That 250 msec is the default interval to perform MQTT actions.
It was something I chose to be "no idea what it should be, but sounds like OK". You could lower it in ESPEasy.ino to check if that improves the publish behavior.
Maybe I should make it dynamic to set the timeout to 'now' when a publish command was given. That's an easy fix.  If that sounds OK to you, please add an issue for that.

radebebek, 2018-04-10:
@TD-er @giig1967g @Grovkillen 
I have a lot of work in my company :/, I hope that tonight (Central Europe time) test all suggestions from you.

Grovkillen, 2018-04-10:
@TD-er Yes that sounds like a better solution.

TD-er, 2018-04-10:
In Controller.ino, you could add `timermqtt = millis();` like this:
```
boolean MQTTpublish(int controller_idx, const char* topic, const char* payload, boolean retained)
{
  if (MQTTclient.publish(topic, payload, retained)) {
    timermqtt = millis();
    return true;
  }
  addLog(LOG_LEVEL_DEBUG, F("MQTT : publish failed"));
  return false;
}
```

Grovkillen, 2018-04-11:
@TD-er That code snippet did the trick!

```
On TestEvent Do
  Publish,%sysname%/test1,%eventvalue%
  Publish,%sysname%/test2,%eventvalue%
  Publish,%sysname%/test3,%eventvalue%
EndOn
```

And sent this command `event,TestEvent=1...100` worked as a charm.

TD-er, 2018-04-11:
OK, then we should add that asap :)
I was thinking about making the interval to call MQTT's own loop() more dynamic.
What are the use cases data is sent through MQTT?
With this patch we've fixed the publish tasks.
MQTTimport should also be looked at, but I was thinking that one should also get a complete make-over to be less of a hack and use a normal controller, not its own MQTT client.

Grovkillen, 2018-04-11:
"use cases" as in end user or generic esp easy?

Grovkillen, 2018-04-11:
TODAY
----------
From ESP Easy:
- Task/device values
- ..**/status** (not used now?)
- ..**/status/LWT** (last will testament)

To ESP Easy:
- ..**/cmd** (commands)

To MQTT Import:
- float values to be used in rules (essentially a dummy device which is populated using _taskvalueset_?)

TOMORROW?
----------
From ESP Easy:
- ..**/status** (should give "ok" and GPIO information just as in the web gui)
- ..**/log** (just as serial log is today, if enabled)

To ESP Easy:
- ..**/import**/<tasknumber>/<valuenumber> (payload is a float, would be used in conjunction with dummy device to make the MQTT Import obsolete for some use cases)

I would like the #378 discussion to be featured as well....

TD-er, 2018-04-11:
If you're setting a switch in something like Domoticz, how is that data sent to the ESP via MQTT?
Is the switch plugin listening to some subscribed channel?
Polling for incoming messages on MQTT is also related to that interval timer, which is now set to 250 msec.
Maybe the 'nominal' MQTT interval check timer should be user selectable?

Grovkillen, 2018-04-11:
I'm no Domoticz user... haven't looked into that, only the OpenHAB MQTT which is pretty generic... only the starting front slash that is non-generic and easily removed in the settings.

TD-er, 2018-04-11:
I guess OpenHAB MQTT is not so different.
So the switch listens to a subscribed topic?

Grovkillen, 2018-04-11:
for generic MQTT we only send commands or as in my case I use events or MQTT Import to trigger events.

radebebek, 2018-04-11:
Hello all,
**Update:**
1. @giig1967g I was trying remove SwitchUp (task 3) and SwitchDown (task 5). The situation is unchanged. [Picture](https://user-images.githubusercontent.com/29672138/38612817-253b9f50-3d88-11e8-93c1-89af4b145f1d.png)

2. @TD-er , @Grovkillen  I was trying add Delay after all Publish comand. The situation is unchanged. [Picture] (https://user-images.githubusercontent.com/29672138/38612881-5b26ef66-3d88-11e8-9876-e2416d424822.png)

3. @TD-er  Later today, I will try set lasted mega version, after that I set update in topic.

When MQTT connection loss, simply do not change the Import#moving variables that the shutter stop. Or if it was published before lost connection. After establish connection, does not take on real value. App must publish and after that MQTT import take real value.

radebebek, 2018-04-11:
@Grovkillen  I have must wait for a new version firmware or I can add manual that code?
Because I flash .bin firmware file with NodeMcu. And I Am not sure how to add that code.

Grovkillen, 2018-04-11:
Yes you need to wait for the official build, but first someone need to PR that change, then it need to be merged. But until then you can experiment with my custom build found [here](https://www.dropbox.com/s/jtc3p1d7ygt2rg6/MQTT_DELAY_ESP_Easy_mega-20180411_normal_ESP8266_4096.bin?dl=1).

TD-er, 2018-04-14:
@wdonker I'm not sure what you mean by this.
Does this mean, something has to change?

wdonker, 2018-04-14:
No,  I was referring to your question:  

> If you're setting a switch in something like Domoticz, how is that data sent to the ESP via MQTT?
Is the switch plugin listening to some subscribed channel?

Grovkillen, 2018-04-14:
We've added a small delay of 10mSec and the unit is now responding even better. I close this issue now.

TD-er, 2018-04-18:
There is quite a number of issues with MQTTimport, so that is quite high on the to-do list.
But there is also a number of issues with WiFi and that's what I am going to look at this evening.
Maybe this issue and the issues with wifi are somehow related.

Just a question, do you use DHCP or static IP?

radebebek, 2018-04-18:
Sonoff and Mqtt Server both have static ip.
I have noticed since it's new mega-20180417 that I make the opening and closing blinds cca 10 times without lost connection, and after that mqtt import lost connection some time. And so it alternates.
----
For some other things, that might not be a big problem.
However, in my situation, if the mqttimport loses the connection, variable [MQTTImport#moving] stays in the moving state 0 or 1 (2 is stop state and correct state after action DOWN or UP) and after that I have to make a manual pablish number 2. Otherwise, applications are blocked.

radebebek, 2018-04-18:
Bigger log :
[Picture](https://user-images.githubusercontent.com/29672138/38955133-fd13b7f2-4353-11e8-8db9-3b09058c0789.png)
In this case, gpio 12 is correct passed, next acction is gpio 5 and mwtt lost connection. Publish MQTTImport#moving to 2 never come to MQTT Server.

Oxyandy, 2018-04-18:
@radebebek What controllers do you have enabled please ?

radebebek, 2018-04-18:
@Oxyandy 
OpenHAB MQTT

Oxyandy, 2018-04-18:
Thanks, I will try something later
do you have any logs too ?

radebebek, 2018-04-19:
In the picture in previous messages, if you want that log or something else? @Oxyandy

TD-er, 2018-04-19:
Yesterday I made a change in how the wifi is configured with static IP.
In short, there is a bug in core 2.4.0 that will not set the `_use_static_ip` variable and thus starts the DHCP client.
This may lead to several strange issues.
So I created a dummy class which inherits the class that has this static member and added 1 function, to set this boolean.
Now the wifi is switched off, set this boolean as needed and start wifi.

I just realized, it still has to check this flag after changing settings, but at least the wifi should work now with static IP, as long as the settings are not change between static/DHCP.

radebebek, 2018-04-19:
@TD-er There was a misunderstanding.
I have Router "Mikrotik" where I configure Static IP for all sonoff device.
So, my sonoff is DHCP (Leave blank where is IP address) but He always give same address.
Can this be a problem?

TD-er, 2018-04-19:
Nope, should work just fine. There (hopefully was) an issue with connecting with static IP.

radebebek, 2018-04-25:
@TD-er  Any progress?

TD-er, 2018-04-25:
I've not yet looked into the MQTT issues the last week.
Too busy with wifi issues.
Those have to be dealt with first, because these may also lead to other connection related problems.

radebebek, 2018-05-08:
@TD-er, @giig1967g   I installed the latest version mega, I saw nothing improved. :(
I did a little work on this and I have some new knowledge.
So,  when I remove all Publish comannd and leave only this in rules:
`on Import#moving=0 do 
        gpio,12,1
        Delay [Import#time]
        gpio,12,0
endon

on Import#moving=1 do 
        gpio,5,1
        Delay [Import#time]
        gpio,5,0
endon`

When gpio5 or gpio12 set to 0... MQTT Lost connection. (It also sometimes passes without losing the connection.)
I also some other project when use similar and I dont have problem (only difference is thatin one action I use gpio set to high, in other action to low. Code down).

`on Import#Svijetlo=1 do 
      gpio,12,1
endon
on Import#Svijetlo=0 do 
      gpio,12,0 
endon
on Tipka#Stanje=1 do 
         if [Import#Svijetlo]=0 
              Publish RoletneMalaSobaWC/Power,1
        else 
              Publish RoletneMalaSobaWC/Power,0 
        endif 
endon
on Tipka#Stanje=0 do 
         if [Import#Svijetlo]=0 
               Publish RoletneMalaSobaWC/Power,1 
         else 
               Publish RoletneMalaSobaWC/Power,0 
         endif 
endon`

This comparison explains that the problem is created because in the same action gpio is power on and power off, and it's somehow connected with MQTT.
I hope that I helped a little and that this will be solved.

radebebek, 2018-05-08:
After continuing testing variation. I found the cause problem.
If I have this: 
on Import#moving=0 do 
        gpio,12,1
        gpio,12,0
        Publish RoletneMalaSobaWC/currentPosition,[Import#target]
        Publish RoletneMalaSobaWC/statePosition,2 
endon

on Import#moving=1 do 
        gpio,5,1
        gpio,5,0
        Publish RoletneMalaSobaWC/currentPosition,[Import#target]
        Publish RoletneMalaSobaWC/statePosition,2 
endon

TD-er, 2018-05-08:
Maybe @Grovkillen has something intelligent to say about these rules?

giig1967g, 2018-05-09:
Hi @radebebek 
I have similar rule with Delay, publish, etc:
```
on rotarypush#switch=0 do
  timerset,1,0
  oled,2,1,Set=[rotary#counter].
  gpio,0,0
  delay,500
  gpio,0,1
  sendto 16,event,task1=[rotary#counter]
  delay,1000
  publish /%sysname%/dummy/nLocalSetTemp1,[rotary#counter]
endon

```
But I don't have disconnections. See log:
```
30873086 : EVENT: rotaryPush#Switch=0.00
30873103 : ACT  : timerset,1,0
30873111 : Command: timerset
30873119 : ACT  : oled,2,1,Set=23.
30873165 : ACT  : gpio,0,0
30873166 : SW   : GPIO 0 Set to 0
30873248 : ACT  : delay 500
30873255 : Command: delay
30873386 : ACT  : gpio,0,1
30873387 : SW   : GPIO 0 Set to 1
30873395 : ACT  : sendto 16,event,task1=23
30873403 : Command: sendto
30873465 : ACT  : delay,1000
30873471 : Command: delay
30874480 : ACT  : publish /ESPT3/dummy/nLocalSetTemp1,23
30874488 : Command: publish
30874530 : SW   : Switch state 1 Output value 1
30874532 : EVENT: rotaryPush#Switch=1.00
30899298 : EVENT: Clock#Time=Wed,08:05

```

radebebek, 2018-05-09:
```
41572: IMPT : [import#moving] : 2.00
41572: EVENT: import#moving=2.00
60610: WD   : Uptime 1 ConnectFailures 0 FreeMem 17784
90472: IMPT : [import#target] : 100.00
90473: EVENT: import#target=100.00
90610: WD   : Uptime 2 ConnectFailures 0 FreeMem 17784
98575: IMPT : [import#target] : 0.00
98576: EVENT: import#target=0.00
98671: IMPT : [import#moving] : 0.00
98672: EVENT: import#moving=0.00
98682: ACT  : gpio,12,1
98684: SW   : GPIO 12 Set to 1
98687: ACT  : gpio,12,0
98688: SW   : GPIO 12 Set to 0
98693: ACT  : Publish SonoffDual/currentPosition,0
98704: Command: publish
98710: ACT  : Publish SonoffDual/statePosition,2
98722: Command: publish
98772: IMPT : [import#current] : 0.00
98773: EVENT: import#current=0.00
99075: IMPT : [import#moving] : 2.00
99076: EVENT: import#moving=2.00
101372: IMPT : [import#moving] : 1.00
101372: EVENT: import#moving=1.00
101384: ACT  : gpio,5,1
101386: SW   : GPIO 5 Set to 1
101389: ACT  : gpio,5,0
101391: SW   : GPIO 5 Set to 0
101396: ACT  : Publish SonoffDual/currentPosition,100
101407: Command: publish
101413: ACT  : Publish SonoffDual/statePosition,2
101426: Command: publish
101473: IMPT : [import#current] : 100.00
101474: EVENT: import#current=100.00
101673: IMPT : [import#moving] : 2.00
101674: EVENT: import#moving=2.00
105975: IMPT : [import#time] : 10000.00
105976: EVENT: import#time=10000.00
106174: IMPT : [import#moving] : 0.00
106175: EVENT: import#moving=0.00
106181: ACT  : gpio,12,1
106184: SW   : GPIO 12 Set to 1
106187: ACT  : gpio,12,0
106189: SW   : GPIO 12 Set to 0
106194: ACT  : Publish SonoffDual/currentPosition,0
106204: Command: publish
106210: ACT  : Publish SonoffDual/statePosition,2
106220: Command: publish
106473: IMPT : [import#current] : 0.00
106473: EVENT: import#current=0.00
106574: IMPT : [import#moving] : 2.00
106575: EVENT: import#moving=2.00
111272: IMPT : [import#moving] : 1.00
111272: EVENT: import#moving=1.00
111282: ACT  : gpio,5,1
111283: SW   : GPIO 5 Set to 1
111286: ACT  : gpio,5,0
111287: SW   : GPIO 5 Set to 0
111292: ACT  : Publish SonoffDual/currentPosition,100
111302: Command: publish
111308: ACT  : Publish SonoffDual/statePosition,2
111321: Command: publish
111373: IMPT : [import#current] : 100.00
111374: EVENT: import#current=100.00
111472: IMPT : [import#moving] : 2.00
111473: EVENT: import#moving=2.00
114571: IMPT : [import#moving] : 0.00
114572: EVENT: import#moving=0.00
114578: ACT  : gpio,12,1
[... 1699 lines cut ...]
602583: WD   : Uptime 10 ConnectFailures 2 FreeMem 17600
610465: IMPT : [import#moving] : 2.00
610466: EVENT: import#moving=2.00
626366: IMPT : [import#current] : 100.00
626367: EVENT: import#current=100.00
628466: IMPT : [import#target] : 100.00
628466: EVENT: import#target=100.00
632583: WD   : Uptime 11 ConnectFailures 2 FreeMem 17600
662583: WD   : Uptime 11 ConnectFailures 2 FreeMem 17600
692583: WD   : Uptime 12 ConnectFailures 2 FreeMem 17600
722583: WD   : Uptime 12 ConnectFailures 2 FreeMem 17600
752583: WD   : Uptime 13 ConnectFailures 2 FreeMem 17600
782583: WD   : Uptime 13 ConnectFailures 2 FreeMem 17600
812583: WD   : Uptime 14 ConnectFailures 2 FreeMem 17600
842583: WD   : Uptime 14 ConnectFailures 2 FreeMem 17600
872583: WD   : Uptime 15 ConnectFailures 2 FreeMem 17600
902583: WD   : Uptime 15 ConnectFailures 2 FreeMem 17600
932583: WD   : Uptime 16 ConnectFailures 2 FreeMem 17600
962583: WD   : Uptime 16 ConnectFailures 2 FreeMem 17600
992583: WD   : Uptime 17 ConnectFailures 2 FreeMem 17600
1022583: WD   : Uptime 17 ConnectFailures 2 FreeMem 17600
1052583: WD   : Uptime 18 ConnectFailures 2 FreeMem 17600
1082583: WD   : Uptime 18 ConnectFailures 2 FreeMem 17600
1112583: WD   : Uptime 19 ConnectFailures 2 FreeMem 17600
```

radebebek, 2018-05-09:
@TD-er 
Please open issues. I accidentally closed.

TD-er, 2018-05-09:
@radebebek Tomorrow I have more time to look at this.
What do you mean by "set log in spoiler" ?

radebebek, 2018-05-10:
@TD-er  Spoiler for hide text. Similarly to the forum site but it looks like there is no such possibility.

giig1967g, 2018-05-10:
Hi @radebebek 
I can confirm that setting a delay of 10 seconds the UNIT disconnects from MQTT for me too. 
Maybe a timeout?

A workaround could be to set a timer instead of a such long delay.