letscontrolit/ESPEasy #1229: Open wifi requires a space in wifi password to connect.

mattlward, 2018-04-09:
### Summarize of the problem/feature request

Our university runs an IOT wireless network campus wide that is mac authed to devices that are entered into a portal.  When I connect to it via laptop, no password is required.  When I connect a Wemos D1 Mini I must provide a password of " ", or anything else for it to connect.

### Expected behavior

Connect with blank password

### Actual behavior

Requires at least 1 space for a password.

### Steps to reproduce

1. Configure an open AP and connect
2. 
3. 

Yes

### System configuration

Hardware: Wemos D1 Mini V3.0.0, purchased direct from Wemos.  This problem was found today when I flashed the unit with mega-20180409 and it did not come back online.  

ESP Easy version: mega-20180409

ESP Easy settings/screenshots: 

### Rules or log data

mattlward, 2018-04-10:
It will not allow an empty password... really freaks it out if you try.  But, just hitting spacebar on the field fixes it.

TD-er, 2018-04-13:
This one should be fixed. At least I've allowed the password to be empty.

Grovkillen, 2018-04-17:
I close this, please report back here if the issue still persist.

closed 2018-04-17 by Grovkillen