letscontrolit/ESPEasy #1232: Allow us to specify a preferred BSSID

Grovkillen, 2018-04-11:
### Summarize of the problem/feature request

I'd like to be able to specify my "closest" AP and make my unit pick this BSSID before any other AP.

### Expected behavior

I would expect the unit to always try to connect to the best possible AP.

### Actual behavior

With current code everything is working just great with connecting speed really fast. But right after a power out my main AP is broadcasting minutes before the other APs and the result is that all my units will connect to this AP and the closer AP will never be used since the unit will always try to connect to latest (working) BSSID.

TD-er, 2018-04-11:
So it should do a periodic test to see if the preferred BSSID is online and connect to that?
Or only connect to the strongest RSSI?

Currently the BSSID is not stored in the settings.

Grovkillen, 2018-04-11:
Maybe both?

Dropdown with:
1. Preferred BSSID
2. Strongest RSSI

Grovkillen, 2018-04-11:
Yes that is true... not much spare storage? If that's the case I could live with "Strongest RSSI" for now :)

TD-er, 2018-04-11:
... and periodical reconnect to the strongest?
What should be the criterium to switch? >= 10 dB improvement?

Grovkillen, 2018-04-11:
Yes we need quite a big difference like that in order to get the hysteresis working properly.