letscontrolit/ESPEasy #1235: Cannot connect to webserver on Release mega-20180413

mgoepel, 2018-04-13:
### Summarize of the problem/feature request

After upgrading from Release mega-20180412 to Release mega-20180413 I am no longer able to connect to the webserver.

### Expected behavior

Connect to the web server to be able to configure ESPEasy

### Actual behavior

After upgrading from Release mega-20180412 to Release mega-20180413 I am no longer able to connect to the webserver.  The ESP8266 Wemos D1 continues to send data correctly to the MQTT broker, and is executing rules correctly, the web server is just not responding.
I have attempted to connect from various machines, windows and linux.

### Steps to reproduce

1. Load Release mega-20180413 onto an ESP8266
2. Attempt to connect
3. 

### System configuration

Hardware:
Yes I have powered off and on again
Wemos D1

Still cannot connect.

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

Grovkillen, 2018-04-13:
The problem is not mega-2018041**3** but mega-2018041**2**. A bug was introduced in yesterdays release and I quickly deleted the release but you seem to have beat me to it.

The bug was corrupting the settings file and that is what is causing your problem. Really sorry for that!

TD-er, 2018-04-13:
You could try to call "Save" in the terminal.
Then the settings will be saved again.

Also please try a power off/on (or press reset on the node itself) to restart the node.

mgoepel, 2018-04-13:
No worries, 
I saw yesterdays was pulled, I have now downgraded back to mega-20180412 and it all works well again.
Do AFAIAC it is not an issue with 12 ;-)

TD-er, 2018-04-13:
There was an issue with the settings struct.
Some new values were added to the struct, which is just fine. However these were added at the end of the struct, where the checksum is also stored.
So every time the settings are saved, the checksum is added at the position of a boolean which may only be set for ESP32.
If that bit is activated for an ESP8266, it will no longer boot.
So it may work, but at some moment the settings will have a checksum where that bit is set to 1 and then your locked out of your settings.

mgoepel, 2018-04-13:
Wow, that is excellent, thank you.  I have not looked through that part of the code yet...

I have upgraded again to 13 and attempted to issue a save in the terminal (as you suggested), with no success.  This is my dev/test device, so I will leave it on 13 as I am still receiving all my test data...

I have been curious why the config file is binary, and not json text?

TD-er, 2018-04-13:
That's smaller to store. There are plans for JSON REST API like interface.

dsabathi, 2018-04-16:
Same issue here but started with 20180404. After flashing the firmware I can ping the wemos D1. As soon as I access the webpage even ping stops working.
Flashing 20180403 makes it work again

mattlward, 2018-04-20:
So, I just upgraded to mega-20180420 dev and now I connect to wireless but cannot ping the device and never can connect to my mqtt broker and yes it is up...   As you can see I have the Settings CRC fail going on here as well.  I had a feeling this one was going to go south...

Here are my logs:

INIT : Booting version: mega-20180420 (core 2_4_0)
86 : INIT : Warm boot #3
87 : FS   : Mounting...
112 : FS   : Mount successful, used 78061 bytes of 957314
431 : CRC  : program checksum       ...OK
436 : CRC  : Settings CRC           ...FAIL
438 : CRC  : SecuritySettings CRC   ...OK 
493 : INIT : Free RAM:17472
493 : INIT : I2C
493 : INIT : SPI not enabled
596 : INFO : Plugins: 72 [Normal] [Testing] [Development] (core 2_4_0)
596 : EVENT: System#Wake
694 : WIFI : Switch on WiFi
695 : WIFI : Set WiFi to STA
696 : WIFI : Connecting dd-wrt_vap attempt #0
809 : EVENT: System#Boot
813 : ACT  : timerSet,1,15
864 : WD   : Uptime 0 ConnectFailures 0 FreeMem 14784
992 : IMPT : MQTT 037 Intentional reconnect
1347 : IMPT : MQTT 037 Intentional reconnect
1991 : IMPT : MQTT 037 Intentional reconnect
2345 : IMPT : MQTT 037 Intentional reconnect
3852 : IMPT : MQTT 037 Intentional reconnect
4205 : IMPT : MQTT 037 Intentional reconnect
4557 : WIFI : Connected! AP: dd-wrt_vap (BA:A3:86:50:6B:00) Ch: 6 Duration: 3090 ms
4557 : EVENT: WiFi#ChangedAccesspoint
4762 : IMPT : MQTT 037 Intentional reconnect
5115 : IMPT : MQTT 037 Intentional reconnect
5747 : Dummy: value 1: 0.00
5747 : Dummy: value 2: 0.00
5748 : Dummy: value 3: 0.00
5748 : Dummy: value 4: 0.00
5749 : EVENT: D1#Real=0.00
5796 : EVENT: D1#FeelsLike=0.00
5836 : EVENT: D1#Wind=0.00
5873 : EVENT: D1#WindGust=0.00
6033 : IMPT : MQTT 037 Intentional reconnect
6386 : IMPT : MQTT 037 Intentional reconnect
6748 : Dummy: value 1: 0.00
6748 : Dummy: value 2: 0.00
6748 : Dummy: value 3: 0.00
6748 : Dummy: value 4: 0.00
6749 : EVENT: D2#WindDeg=0.00
6787 : EVENT: D2#Humidity=0.00
6827 : EVENT: D2#=0.00
6866 : EVENT: D2#=0.00
7140 : IMPT : MQTT 037 Intentional reconnect
7493 : IMPT : MQTT 037 Intentional reconnect
7846 : WIFI : DHCP IP: 192.168.1.186 (ESP-Spare-9) GW: 192.168.1.2 SN: 255.255.255.0   duration: 3033 ms
7857 : EVENT: WiFi#Connected
8236 : IMPT : MQTT 037 Intentional reconnect
8569 : IMPT : MQTT 037 Intentional reconnect
8621 : IMPT : MQTT 037 Connection lost
8936 : MQTT : Intentional reconnect
54351 : MQTT : Connection lost
59408 : MQTT : Failed to connect to broker
65883 : MQTT : Failed to connect to broker
70897 : MQTT : Failed to connect to broker
78777 : MQTT : Failed to connect to broker
83964 : MQTT : Failed to connect to broker
90424 : MQTT : Failed to connect to broker
95541 : MQTT : Failed to connect to broker
102016 : MQTT : Failed to connect to broker
107034 : MQTT : Failed to connect to broker
113489 : MQTT : Failed to connect to broker
118593 : MQTT : Failed to connect to broker
125071 : MQTT : Failed to connect to broker
130235 : MQTT : Failed to connect to broker
136698 : MQTT : Failed to connect to broker
141716 : MQTT : Failed to connect to broker
149534 : MQTT : Failed to connect to broker
154653 : MQTT : Failed to connect to broker
161125 : MQTT : Failed to connect to broker
166223 : MQTT : Failed to connect to broker
172685 : MQTT : Failed to connect to broker
177701 : MQTT : Failed to connect to broker

TD-er, 2018-04-20:
I think we are having multiple issues here.
I never see "Webserver: start" in the logs.
This means your setup doesn't finish WiFi connecting like it should.

mattlward, 2018-04-20:
My last post was the first boot off of a flash erase and upload of espeasy, had no settings yet.  The point of the last post was that the settings CRC was bad from a fresh erase and load.  It is working with a config on it now, but the settings CRC is still bad.

TD-er, 2018-04-20:
Like I said before, the CRC report may be a false positive.
We really have to look into that and I think that will be the major issue for me this evening.
It may or may not solve many other issues and to say the least it is scary to see in the logs :)
I do believe it is very likely related to the issue of not being able to load saved settings.

Grovkillen, 2019-12-29:
I close this one for now... open if still an issue.

closed 2019-12-29 by Grovkillen