letscontrolit/ESPEasy #1251: [wifi] webserver becomes unresponsive after a while with static IP

TD-er, 2018-04-15:
Setup wifi with static network config, which works just fine.
After a while, the node is no longer accessible via the web, but is still connected and sending data over MQTT to Domoticz.

wdonker, 2018-04-15:
I had this same problem almost a year ago, using R120 but it dissapeared after some time. I thought it was solved by an update of my router but maybe it was a bug in ESPEasy that was solved and now returned...

[http://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3373#p17834](url)

TD-er, 2018-04-15:
Just a question, what brand of wifi products are you using?

wdonker, 2018-04-15:
My router (and DHCP-server) is a Technicolor 7210 provided by Ziggo.

TD-er, 2018-04-16:
I hope I fixed it now. Could you also try to see if the wifi stays connected?

TD-er, 2018-04-18:
OK, it seems to be a bug in core 2.4.0
See https://github.com/esp8266/Arduino/issues/4114
I will try to think of a work-around to fix this.

pwassink, 2018-04-19:
I am not sure (again) if thuis is the same issue bu.

Somewhere after version mega-0407 things went from bad to worse in the daily binary section.
I have several nodes and testnodes in een fixed-ip wifi on a networkt with an seperate wifibackup-network. 
Symptoms include no web-access of the esp at all, only able toe get it listening toe serial (debug) commands at all,  unable to force factory reset bij connectingt tx&rx on the nodemcu.
After a factory default and reflash one time i enterred the WifiSSID and WifiKey via serial added with an Wificonnect, and it started looping. and was never reachable, even not through serial commands.
Today i picked up the 419 version, read something on the fixed ip workaround but tht version did not change a thing, not via Ota, not via a direct flash and even not with a blank,bin over all config it did respond via web or serial .. 
A fresh compiled version pio upload, normal, 4096, esp8266 from a synced git copy also went wrong.

Is this something i just have against me or is there something a bit wrong in the code at the moment ?

Peter

pwassink, 2018-04-19:
Well, reflashed and powercycled the esp8266 (nodemcu) with Mega-20180419 normal 4096 daily build.
Initially it responded to serial commands, used that tor Reset and watched it format its config area 
Gave in the WifiKey And WifySSID via cmdline, saved that, used WifiConnect, still seems ok, NO webresponse on the still Dhcp address mentioned in logging yet.
It is telling me it is using Core 2_4_0
settings check CRC fails i see
Wifi goes into STA mode
it does connect on Wifi, gets an Ipv4 addres there, and now it seems responding on Web
was able to get into web config menu, add some lines of config nodename then hangs again
and stops responding on serial commands completely

Hope this helps?

TD-er, 2018-04-19:
I will also look into the 'start webserver' routine to see if it looks for some IP config to bind to.
Maybe the webserver should be restarted when a (new) IP is received.

pwassink, 2018-04-19:
That does not sound unreasonable, question in my perception is at that point 
why does it work inittially and stops doing so (web and serial based) after a thirty-some seconds ?

TD-er, 2018-04-19:
I have no idea.
If it is after +/- 60 seconds after boot, then it could be that the node starts the AP mode first, which will be disabled after 60 seconds no client connected.

pwassink, 2018-04-20:
Well, Resetted the esp8266, rebooted after formatting flash, enterred the SSID and Pass through serial, enterred a Debug 4 , Save, Wificonnect also. No response on ipv4 address mentioned in logging, 
Started a Nmap poll from other terminal to this esp, Then it crashed during that poll, caught that in tracing.  Used params in Nmap : nmap -sV -T4 -O -F --version-light 192.168.3.207

Perhaps this might help, if you require more info, will be back this evening.

Serial log-trace of Esp8266 :

>Reboot

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
⸮U90 : 

INIT : Booting version: mega-20180419 (core 2_4_0)
91 : INIT : Warm boot #7
92 : FS   : Mounting...
120 : FS   : Mount successful, used 75802 bytes of 957314
409 : CRC  : program checksum       ...OK
414 : CRC  : Settings CRC           ...FAIL
442 : CRC  : SecuritySettings CRC   ...OK 
442 : WIFI : Set WiFi to STA
550 : INIT : Free RAM:20312
551 : INIT : I2C
551 : INIT : SPI not enabled
564 : INFO : Plugins: 47 [Normal] (core 2_4_0)
565 : WIFI : Connecting vucnet attempt #0
573 : WD   : Uptime 0 ConnectFailures 0 FreeMem 17264
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with vucnet, channel 13
dhcp client start...
3637 : WIFI : Connected! AP: vucnet (34:31:C4:5C:91:7E) Ch: 13 Duration: 3071 ms
ip:192.168.3.207,mask:255.255.255.0,gw:192.168.3.1
4190 : WIFI : DHCP IP: 192.168.3.207 (ESP-Easy-14) GW: 192.168.3.1 SN: 255.255.255.0   duration: 553 ms
pm open,type:2 0
>Debug 4

Ok
30573 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16480
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
state: 5 -> 2 (2c0)
rm 0
pm close 7
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x402404a6, epc2=0x00000000, epc3=0x00000000, excvaddr=0x015200dd, depc=0x00000000

Exception (9):
epc1=0x402404a6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x015200dd depc=0x00000000
[... 88 lines cut ...]
442 : CRC  : SecuritySettings CRC   ...OK 
442 : WIFI : Set WiFi to STA
550 : INIT : Free RAM:20312
551 : INIT : I2C
551 : INIT : SPI not enabled
565 : INFO : Plugins: 47 [Normal] (core 2_4_0)
565 : WIFI : Connecting vucnet attempt #0
574 : WD   : Uptime 0 ConnectFailures 0 FreeMem 17264
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with vucnet, channel 13
dhcp client start...
3636 : WIFI : Connected! AP: vucnet (34:31:C4:5C:91:7E) Ch: 13 Duration: 3070 ms
ip:192.168.3.207,mask:255.255.255.0,gw:192.168.3.1
6215 : WIFI : DHCP IP: 192.168.3.207 (ESP-Easy-14) GW: 192.168.3.1 SN: 255.255.255.0   duration: 2579 ms
7908 : Ram usage: Webserver only: 0 including Core: 0
8144 : Ram usage: Webserver only: 0 including Core: 0
pm open,type:2 0
30573 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16240

TD-er, 2018-04-20:
You're using channel 13.
I've seen (at least in the ESP32 api code) that there is support for setting a country code to enable those higher channels.
Above channel 11, wifi is country specific.
Currently there is no code related to setting country specific settings, so I don't know how well the ESP will work on channels above ch. 11.

pwassink, 2018-04-20:
Fine, 

have been using that channel 13 for many years already. also the esp-easy 1.x versions and esp32 have no problems with that up until last week , after mega-407 it went wrong

TD-er, 2018-04-20:
OK, just checking, since I read about this code a few days ago.
And just want to make sure I'm not staring into unreproducable bugs related to some "region specific" differences. :)

TD-er, 2018-04-20:
So after 0407 all kinds of issues?
And CRC already failed on 0319 build?

Oxyandy, 2018-04-20:
ESP_Easy_v2.0-20180319_normal_ESP8266_1024.bin
Just checked, it has no problems 'saving' settings & no crc fail issues..
Occasionally on a firmware change the msg will show a fail, 
issuing a save command always resolves it until
ESP_Easy_mega-20180405_normal_ESP8266_1024.bin

pwassink, 2018-04-20:
Just heard that "mega 0410" was the last known good version, maybe you can find out / trace back what was alterred ?

TD-er, 2018-04-20:
Last known "good version" as in what was wrong since then? Disconnect, Static IP or CRC error?

pwassink, 2018-04-20:
Last know good as in : creating a working esp8266, with webinterface, which does save its config and surives a reboot, even with fixed ip.  

Not sure on the config-crc issue maybe someone can confirm that ?

TD-er, 2018-04-20:
OK, about the settings file. :)
Good to know.

pwassink, 2018-04-21:
This morning i found the mega-421normal 4096 binary,  got that and tested with it, tried hat on another wifi network on channel 3 with similar results as before, APmode primary setup was (visually changed) avaiable and working. still DHCP based setup, almost virginal esp, Still crc errors on the config even after a local save, I did some tracing, output below. 

>reset
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 0
RESET: formatting...
RESET: formatting done...
70319 : FILE : Saved config.dat
70351 : FILE : Saved security.dat
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
⸮⸮⸮U89 : 

INIT : Booting version: mega-20180421 (ESP82xx Core 2_4_0)
90 : INIT : Warm boot #8
91 : FS   : Mounting...
116 : FS   : Mount successful, used 75802 bytes of 957314
406 : CRC  : program checksum       ...OK
410 : CRC  : Settings CRC           ...FAIL
412 : CRC  : SecuritySettings CRC   ...OK 
519 : INIT : Free RAM:20416
519 : INIT : I2C
519 : INIT : SPI not enabled
533 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_0)
533 : WIFI : Switch on WiFi
534 : WIFI : Set WiFi to STA
536 : WIFI : No valid wifi settings
537 : WIFI : Connection Failed
639 : WIFI : Set WiFi to AP
1528 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1535 : WD   : Uptime 0 ConnectFailures 0 FreeMem 18144
31534 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18376

### inbetween it failed to connect after enterring ssid and key via serial,  on channel 13 .
### was able to get in via primary web config menu on the 192.168.4.1 address and entered my second wifinet, channel3

172644 : AP Mode: Client connected: D4:63:C6:47:88:59 Connected devices: 1
172647 : Webserver: start
181534 : WD   : Uptime 3 ConnectFailures 0 FreeMem 14848
185118 : WIFI : Set WiFi to AP
211534 : WD   : Uptime 4 ConnectFailures 0 FreeMem 13568
221541 : WIFI : Set WiFi to AP+STA
221544 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
221545 : Webserver: stop
221545 : WIFI : Connecting Vuc-Link attempt #0
221547 : WIFI : Credentials Changed, retry connection. SSID: Vuc-Link
221600 : WIFI : Connecting Vuc-Link attempt #0
225076 : WIFI : Connected! AP: Vuc-Link (00:02:6F:66:51:57) Ch: 3 Duration: 3475 ms
226124 : WIFI : DHCP IP: 192.168.3.207 (ESP-Easy-0) GW: 192.168.3.1 SN: 255.255.255.0   duration: 1048 ms
226126 : Webserver: start
226182 : FILE : Saved config.dat
226214 : FILE : Saved security.dat
226219 : Webserver: stop
226220 : WIFI : Connecting Vuc-Link attempt #0
241534 : WD   : Uptime 4 ConnectFailures 0 FreeMem 11832
271534 : WD   : Uptime 5 ConnectFailures 0 FreeMem 11832
301534 : WD   : Uptime 5 ConnectFailures 0 FreeMem 11832
331534 : WD   : Uptime 6 ConnectFailures 0 FreeMem 11832
361534 : WD   : Uptime 6 ConnectFailures 0 FreeMem 11832

### No response from webserver in browser , esp seems allive, does respond on icmp ping, after a hard reset it did not change.

INIT : Booting version: mega-20180421 (ESP82xx Core 2_4_0)
84 : INIT : Warm boot #12
85 : FS   : Mounting...
110 : FS   : Mount successful, used 75802 bytes of 957314
400 : CRC  : program checksum       ...OK
405 : CRC  : Settings CRC           ...FAIL
433 : CRC  : SecuritySettings CRC   ...OK 
539 : INIT : Free RAM:20416
539 : INIT : I2C
540 : INIT : SPI not enabled
553 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_0)
554 : WIFI : Switch on WiFi
554 : WIFI : Set WiFi to STA
mode : sta(5c:cf:7f:9d:6e:da)
add if0
556 : WIFI : Connecting Vuc-Link attempt #0
565 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19560
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (4)
4657 : WIFI : Disconnected! Reason: '(4) Assoc expire' Connected for 4097 ms
4657 : WIFI : Connecting Vuc-Link attempt #1
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with Vuc-Link, channel 3
dhcp client start...
6830 : WIFI : Connected! AP: Vuc-Link (00:02:6F:66:51:57) Ch: 3 Duration: 2171 ms
ip:192.168.3.207,mask:255.255.255.0,gw:192.168.3.1
7733 : WIFI : DHCP IP: 192.168.3.207 (ESP-Easy-14) GW: 192.168.3.1 SN: 255.255.255.0   duration: 902 ms
7736 : Webserver: start
pm open,type:2 0
30565 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16184
>debug 4

Ok
60565 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16448
121919 : Ram usage: Webserver only: 0 including Core: 0
121924 : WD   : Uptime 2 ConnectFailures 0 FreeMem 16160