letscontrolit/ESPEasy #1252: ESP32 web pages

flexiti, 2018-04-16:
Does anyone use ESP32?
The version from two weeks ago was weak but it worked, the current version is practically not working,
i.e., for example, you can not add a device (web page hangs), side operations on the web side hangs up, you need to do a reset) (source 15.04.2018)

TD-er, 2018-04-16:
I have not tested ESP32 the last few days, I will have a look at it this evening.

waspie, 2018-04-18:
yeah, i compiled from april 16th source for one today and a good 50% of the time just going into a task causes it to crash. 

99844 : FILE : Saved /config.dat
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x4008c914  PS      : 0x00060c36  A0      : 0x4008172d  A1      : 0x3ffd4050  
A2      : 0x3ffd0be8  A3      : 0x3ffd4450  A4      : 0x3ffd433c  A5      : 0x00000010  
A6      : 0x00000003  A7      : 0x3ffd4374  A8      : 0x801305f4  A9      : 0x3ffd40f0  
A10     : 0x3ffe3804  A11     : 0x3ffd4331  A12     : 0x00000003  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x4008c914:0x3ffd4050 0x4008172a:0x3ffd4140 0x40125f49:0x3ffd4450 0x401036ef:0x3ffd4510 0x400df4f2:0x3ffd4530 0x400df5d1:0x3ffd4590 0x400e0966:0x3ffd45c0 0x400e0b71:0x3ffd4620 0x400e0c5a:0x3ffd4660 0x400e0d03:0x3ffd4910 0x400ecc66:0x3ffd4940 0x4017ad39:0x3ffd5eb0 0x40104143:0x3ffd5ed0 0x401041b9:0x3ffd5ef0 0x40104202:0x3ffd5f20 0x401045bd:0x3ffd5f60 0x400f3443:0x3ffd5fa0 0x40101863:0x3ffd5fc0 0x40160a3c:0x3ffd5ff0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
20 : WIFI : Set WiFi to STA
ªU31 : 

INIT : Booting version: (custom)
31 : INIT : Cold Boot
31 : FS   : Mounting...

linuxnico, 2018-04-18:
i have the same issue without plugins.
i update esp32-webserver, but same issue

TD-er, 2018-04-18:
There has been a commit to allow for some experimental feature to let processes run on both cores.
But I was under the impression that feature was disabled by default.
Maybe it is enabled on your settings due to the recent bug in the settings where the MD5 was stored at the position of this setting?
Could you try with empty settings and the latest build?

linuxnico, 2018-04-18:
how can i erase settings, because, when i uploard R20100 build, the settings is always here

linuxnico, 2018-04-18:
that's the return of 20100 buil upload:
esptool.py v2.1
Connecting........_____....._____..
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 256000
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 9065.6 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 11072 bytes to 7420...
Wrote 11072 bytes (7420 compressed) at 0x00001000 in 0.3 seconds (effective 291.2 kbit/s)...
Hash of data verified.
Compressed 860064 bytes to 541670...
Wrote 860064 bytes (541670 compressed) at 0x00010000 in 22.1 seconds (effective 311.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 2240.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

and the esp32 return :

INIT : Booting version: (custom)
INIT : Cold Boot
FS   : Mounting...
CRC  : No program memory checksum found. Check output of crc2.py
FS   : Error while reading/writing /config.dat in 985

PID:0

Version:0
INIT : Incorrect PID or version!
RESET: Resetting factory defaults...
RESET: Cold boot
RESET: formatting...
RESET: formatting done...
FS   : Error while reading/writing /config.dat in 917
FS   : Error while reading/writing /config.dat in 985
FS   : Error while reading/writing /config.dat in 940
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)
ets Jun  8 2016 00:22:57

linuxnico, 2018-04-18:
en reboot always

TD-er, 2018-04-18:
I can not check before this evening

linuxnico, 2018-04-18:
ok i restart all process, 
i git clone the mega branhc of github
build
upload
and i'll see

waspie, 2018-04-18:
i tried compiling the source from april 02 and i was stuck in a boot loop. using the esptool i was able to erase the flash and upload the binary build from march 30th and it seems stable but i can't apply it to the esp at home with stuff actually hooked up to it until tonight. i'll probably open another issue later regarding hc-sr04 sensor not working on current build from source (problem with newping?)

TD-er, 2018-04-21:
Is this also right after reboot, or only after some uptime?
I know there is a memory leak somewhere in the esp32 code

flexiti, 2018-04-21:
right after reboot and also after some uptime - always

flexiti, 2018-04-21:
Generally, pages are very lazy (in the case of ESP8266 they run faster)

flexiti, 2018-04-21:
I see that some errors are reported ( #1282 #1279 )for ESP32, how is it possible, even if the configuration does not work? :)

flexiti, 2018-04-22:
Well, tomorrow I'll do a test with new ESP32, I'll let you know.
By the way, often after a reset, the system hangs at this point
(I am using ESP32 devkit V1 from DOIT)

INIT : Booting version:  (ESP32 SDK v3.1-dev-661-gf586f5e6)
37 : INIT : Cold Boot
38 : FS   : Mounting...
96 : CRC  : No program memory checksum found. Check output of crc2.py
99 : CRC  : Settings CRC           ...FAIL
114 : CRC  : SecuritySettings CRC   ...OK

waspie, 2018-04-22:
Go back to April 2nd source

TD-er, 2018-04-22:
I just made a commit to hopefully fix the settings related issues.
Could you please also test the latest source?

flexiti, 2018-04-22:
The latest sources:
- new (fresh)  ESP, works badly
- I found ESP which had the previous (good) version loaded, these works now best from 3 I have, but still bad

Often after a reset, all stops as below:

> rst: 0x1 (POWERON_RESET), boot: 0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP: 0xee
> clk_drv: 0x00, q_drv: 0x00, d_drv: 0x00, cs0_drv: 0x00, hd_drv: 0x00, wp_drv: 0x00
> mode: DIO, clock div: 2
> load: 0x3fff0018, flax 4
> load: 0x3fff001c, linen: 812
> load: 0x40078000, flax 0
> load: 0x40078000, linen: 11584
> entry 0x40078a60
> ŞU34:
> 
> INIT: Booting version: (ESP32 SDK v3.1-dev-661-gf586f5e6)
> 34: INIT: Cold Boot
> 35: FS: Mounting ...
> 92: CRC: No memory checksum found. Check output of crc2.py
> 110: CRC: SecuritySettings CRC ... OK

A repeated reset causes the program to run normaly, but still with web problems

TD-er, 2018-04-22:
I will add some information about the used flash chip and settings to the boot log.
Maybe we're looking at some other issue related to the used flash chip.
Wouldn't be the first time ;)

flexiti, 2018-04-22:
What about version before April 2nd ? it works quite good. At the evening I will test all 3 pcs with this version

waspie, 2018-04-22:
I'll try a newer build tonight after the kids go down. I'll erase flash first to be sure no settings are carried forward

waspie, 2018-04-23:
I just cloned the repo and built the firmware and uploaded it after "erase_flash" and I'm not seeing the constant crashing I was before. I haven't loaded up 15 devices or anything yet but previously it was tripped *Really* easily so it looks like its working much better now. I can't stay on this build though, because i need to use hc-sr04 which doesn't work with the newping for whatever reason. but anyway, this looks to be working better now.

flexiti, 2018-04-23:
I found that :

> Often after a reset, all stops as below:
>
> rst: 0x1 (POWERON_RESET), boot: 0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP: 0xee
> clk_drv: 0x00, q_drv: 0x00, d_drv: 0x00, cs0_drv: 0x00, hd_drv: 0x00, [...]
**problem gone if I remove this linie from code**  (ESPEasy.ino)

>   if (Settings.UseSerial)
>   {
>     //make sure previous serial buffers are flushed before resetting baudrate
>     Serial.flush();
>     Serial.begin(Settings.BaudRate);
> //    Serial.setDebugOutput(true);
>   }

System allways restart properly
(only web problem still exist - when I open device page - OLED display)
```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (RTOS_TaskServer) 
Core 1 register dump:
PC      : 0x4008fb93  PS      : 0x00060b36  A0      : 0x4008cb72  A1      : 0x3ffe4300  
A2      : 0x3ffe4370  A3      : 0xb33fffff  A4      : 0x00060b20  A5      : 0x3ffd2c10  
A6      : 0x3ffe4460  A7      : 0x00000000  A8      : 0x80082b94  A9      : 0x4008c964  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x40081771  A13     : 0x00000001  
A14     : 0x00060923  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4008fb93:0x3ffe4300 0x4008cb6f:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11584
entry 0x40078a60
ŞU33 : 
```

TD-er, 2018-04-23:
Yesterday I could reproduce your crashes on an ESP32.
That was after updating the ESP library in PlatformIO.
So we should fixate the version of the library to some version in the PlatformIO.ini file.
That still leaves the Arduino IDE users who build their own versions, since that platform does not check the installed library versions.

TD-er, 2018-04-23:
A stack overflow is often the result of some recursion not terminated and thus a programming bug.
If you really need a bigger stack, then it's almost always the best moment to start looking at your code to see what you're actually doing ;)
Of course there are some exceptions to this rule, but it is a really good rule-of-thumb.

flexiti, 2018-04-25:
@TD-er It looks good now, you are great 😃 

ps. Maybe you can check why the diode flashes 2s / 2s? :)
```
On System#Boot do    
  timerSet,1,5      //Set Timer 1 for the next event in 5 seconds
endon

On Rules#Timer=1 do  
  gpio,2,1
  timerSet,2,1       //Resets the Timer 2 for another 1 seconds
endon

On Rules#Timer=2 do  
  gpio,2,0
  timerSet,1,1       //Resets the Timer 1 for another 1 seconds
endon
```

closed 2018-04-25 by flexiti

TD-er, 2018-04-25:
Someone else also reported the timer of ESP32 was off by a factor 2.
I have not yet looked into that.
I cannot find the issue anymore, maybe you can?

flexiti, 2018-04-25:
ok, I will try

flexiti, 2018-04-28:
II made a correction, it is unfortunate for me is under #1315 

So far, because rulesTimers () is checked asynchronously to timePassedSince time could be by 1 second from the set one
(especially important in short times)
This workaround decreases error is 100 ms.
@TD-er  can You fix it ? :)

TD-er, 2018-04-28:
I will look into it.
But just setting a fixed offset is generally not a good idea.
Especially since we're on multiple platforms (ESP32 and ESP8266), there is no such thing as a constant time offset.