letscontrolit/ESPEasy #1255: Settings CRC  Fail

DittelHome, 2018-04-16:
### Summarize of the problem/feature request
After a clean flash, i get this error: 

**405 : CRC  : Settings CRC           ...FAIL**

In addition, i cant access the web gui, after i  typed my values (serial consol) in the form:
1. WifiKey xxxxxxxxxxxxx
2. WifiSSID yyyyy
3. ip 192.168.1.17
5. unit 6
4. save
5. Reboot  

### Steps to reproduce
You should have the same result, just after flashing

### System configuration
Only the wemos, without any pins connected
Hardware:
WEMOS mini Pro V1

ESP Easy version: 
ESP_Easy_mega-20180416_normal_ESP8266_4096

### Rules or log data

86 : INIT : Warm boot #1
87 : FS   : Mounting...
112 : FS   : Mount successful, used 75802 bytes of 957314
400 : CRC  : program checksum       ...OK
405 : CRC  : Settings CRC           ...FAIL
433 : CRC  : SecuritySettings CRC   ...OK 
539 : INIT : Free RAM:19624
539 : INIT : I2C
540 : INIT : SPI not enabled
553 : INFO : Plugins: 47 [Normal] (core 2_4_0)
554 : WIFI : No valid wifi settings
554 : WIFI : Connection Failed
654 : WIFI : Set WiFi to AP
1544 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1553 : WD   : Uptime 0 ConnectFailures 0 FreeMem 15192
31553 : WD   : Uptime 1 ConnectFailures 0 FreeMem 15424
61553 : WD   : Uptime 1 ConnectFailures 0 FreeMem 15424
91553 : WD   : Uptime 2 ConnectFailures 0 FreeMem 15424
>WifiKey xxxxxxxxxxxxxxxxxxxxxx
Ok
>WifiSSID yyyyyyyyyy
Ok
>ip 192.168.1.17
Ok
>unit 6
Ok
>save
121594 : FILE : Saved config.dat
121625 : FILE : Saved security.dat
Ok

INIT : Booting version: mega-20180416 (core 2_4_0)
86 : INIT : Warm boot #2
88 : FS   : Mounting...
112 : FS   : Mount successful, used 75802 bytes of 957314
400 : CRC  : program checksum       ...OK
405 : CRC  : Settings CRC           ...FAIL
407 : CRC  : SecuritySettings CRC   ...OK 
513 : INIT : Free RAM:19624
513 : INIT : I2C
513 : INIT : SPI not enabled
527 : INFO : Plugins: 47 [Normal] (core 2_4_0)
528 : WIFI : Connecting im6shop attempt #0
536 : WD   : Uptime 0 ConnectFailures 0 FreeMem 16576
3575 : WIFI : Connected! AP: yyyy (30:B5:C2:EB:DB:7D) Ch: 9 Duration: 3046 ms
3576 : IP   : Static IP : 192.168.1.17 GW: 0.0.0.0 SN: 0.0.0.0 DNS: 0.0.0.0
30536 : WD   : Uptime 1 ConnectFailures 0 FreeMem 15840
60536 : WD   : Uptime 1 ConnectFailures 0 FreeMem 15840
90536 : WD   : Uptime 2 ConnectFailures 0 FreeMem 15840
120536 : WD   : Uptime 2 ConnectFailures 0 FreeMem 15840
150536 : WD   : Uptime 3 ConnectFailures 0 FreeMem 15840
180536 : WD   : Uptime 3 ConnectFailures 0 FreeMem 15840
210536 : WD   : Uptime 4 ConnectFailures 0 FreeMem 15840
240536 : WD   : Uptime 4 ConnectFailures 0 FreeMem 15840
270536 : WD   : Uptime 5 ConnectFailures 0 FreeMem 15840
300536 : WD   : Uptime 5 ConnectFailures 0 FreeMem 15840
330536 : WD   : Uptime 6 ConnectFailures 0 FreeMem 15840
360536 : WD   : Uptime 6 ConnectFailures 0 FreeMem 15840
390536 : WD   : Uptime 7 ConnectFailures 0 FreeMem 15840

**Ping is not possible:**

Ping wird ausgeführt für 192.168.1.17 mit 32 Bytes Daten:
Antwort von 192.168.1.3: Zielhost nicht erreichbar.
Antwort von 192.168.1.3: Zielhost nicht erreichbar.
Antwort von 192.168.1.3: Zielhost nicht erreichbar.
Antwort von 192.168.1.3: Zielhost nicht erreichbar.

Ping-Statistik für 192.168.1.17:
    Pakete: Gesendet = 4, Empfangen = 4, Verloren = 0
    (0% Verlust),

`

DittelHome, 2018-04-16:
I have played around and found the following:
After typing "wifidisconnect" from serial i get access to the gui, and of course the esp is pingable.
It seems to be a timing problem.
`210570 : WD   : Uptime 4 ConnectFailures 0 FreeMem 15840
wifidisconnect

Ok
232017 : WIFI : Disconnected! Reason: 'Assoc leave' Connected for 3 m 48 s
232018 : WIFI : Connecting iyyyy attempt #0
234272 : WIFI : Connected! AP: yyyy (30:B5:C2:EB:DB:7D) Ch: 9 Duration: 2254 ms
234273 : IP   : Static IP : 192.168.1.17 GW: 0.0.0.0 SN: 0.0.0.0 DNS: 0.0.0.0
234276 : WIFI : Static IP: 192.168.1.17 (HMUART2-2-6) GW: 0.0.0.0 SN: 255.255.255.0`

TD-er, 2018-04-17:
And a restart using the reset button?
Does that work?

I noticed you did not enter any subnet, that may affect the IP filtering.
I will test your setup later this evening.

DittelHome, 2018-04-17:
Correct, i am using in the first step no subnet. I think there is no command for that.
In the past time it worked like a charm. (Of course a assign a subnet, if the device is reachable)
I have tried both methods, reset button and power off.

DittelHome, 2018-04-17:
by the way..
I have assigned a subnet and a gateway/dns, after the manual wifidisconnect, same result.
The message:
05 : CRC : Settings CRC ...FAIL
makes me nervous..

TD-er, 2018-04-17:
There are indeed more frequent reports of CRC errors.
I am not sure all of these reports are correct

DittelHome, 2018-04-17:
Definitly, my ESP needs after every reboot a "wifidisconnect". Then it works fine...

DittelHome, 2018-04-17:
Maybe this helps...
```
112 : FS   : Mount successful, used 75802 bytes of 957314
400 : CRC  : program checksum       ...OK
405 : CRC  : Settings CRC           ...FAIL
434 : CRC  : SecuritySettings CRC   ...OK 
rules1.txt 1
rules2.txt 0
rules3.txt 0
rules4.txt 0
540 : INIT : Free RAM:20584
540 : INIT : I2C
541 : INIT : SPI not enabled
554 : INFO : Plugins: 47 [Normal] (core 2_4_0)
555 : WIFI : Connecting im6shop attempt #0
564 : WD   : Uptime 0 ConnectFailures 0 FreeMem 17536
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with im6shop, channel 9
dhcp client start...
3601 : WIFI : Connected! AP: im6shop (30:B5:C2:EB:DB:7D) Ch: 9 Duration: 3045 ms
3602 : IP   : Static IP : 192.168.1.17 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
3608 : WIFI : Static IP: 192.168.1.17 (ESP-Easy-6) GW: 192.168.1.1 SN: 255.255.255.0
pm open,type:2 0
30564 : Memtrace
0: lowest: 16672  parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 checkSensors-> 17352 parseTemplate-> 16736 parseTemplate3-> 16672 parseTemplate-> 16736 parseTemplate3-> 16672 
30585 : 1: lowest: 16672  parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 checkSensors-> 17352 parseTemplate-> 16736 parseTemplate3-> 16672 
30617 : 2: lowest: 16672  parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 parseTemplate-> 17240 parseTemplate3-> 17176 checkSensors-> 17352 parseTemplate-> 16736 parseTemplate3-> 16672 parseTemplate-> 16736 parseTemplate3-> 16672 parseTemplate-> 16736 parseTemplate3-> 16672 
30649 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16800
>wifidisconnect
state: 5 -> 0 (0)
rm 0
pm close 7

Ok
40221 : WIFI : Disconnected! Reason: 'Assoc leave' Connected for 36 s
40221 : WIFI : Connecting im6shop attempt #0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with im6shop, channel 9
ip:192.168.1.17,mask:255.255.255.0,gw:192.168.1.1
42475 : WIFI : Connected! AP: im6shop (30:B5:C2:EB:DB:7D) Ch: 9 Duration: 2253 ms
42477 : IP   : Static IP : 192.168.1.17 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
42485 : WIFI : Static IP: 192.168.1.17 (ESP-Easy-6) GW: 192.168.1.1 SN: 255.255.255.0
pm open,type:2 0
```

Feuerreiter, 2018-04-17:
I have the same Problem after i set my ip to static (192.168.178.202). After reboot it shows 2 ip addresses. The second one (192.168.178.70) came from the dhcp server of my router.
Settings CRC also FAIL

INIT : Booting version: (custom) (core 2_4_1)
75 : INIT : Warm boot #5
77 : FS   : Mounting...
101 : FS   : Mount successful, used 75802 bytes of 957314
108 : CRC  : No program memory checksum found. Check output of crc2.py
112 : CRC  : Settings CRC           ...FAIL
140 : CRC  : SecuritySettings CRC   ...OK 
246 : INIT : Free RAM:23720
246 : INIT : I2C
246 : INIT : SPI not enabled
260 : INFO : Plugins: 47 [Normal] (core 2_4_1)
261 : WIFI : Connecting Bitches!!! attempt #0
269 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20672
4031 : WIFI : Connected! AP: Bitches!!! (34:31:C4:1F:2D:E3) Ch: 1 Duration: 3769 ms
4032 : IP   : Static IP : 192.168.178.202 GW: 192.168.178.1 SN: 255.255.255.0 DNS: 192.168.178.1
4037 : WIFI : Static IP: 192.168.178.202 (ESP-Easy-0) GW: 192.168.178.1 SN: 255.255.255.0
5049 : WIFI : Static IP: 192.168.178.70 (ESP-Easy-0) GW: 192.168.178.1 SN: 255.255.255.0   duration: 1017 ms
30269 : WD   : Uptime 1 ConnectFailures 0 FreeMem 19576
60269 : WD   : Uptime 1 ConnectFailures 0 FreeMem 19576

After that i can't reach any ip address.
After i type wifidisconnect i reach the 192.168.178.202 and i can call the webinterface.

I think there is a problem with the configuration of the static ip. Static IP and DHCP at the same time?

TD-er, 2018-04-17:
Hmm, did not know "!" was accepted for an SSID ;)

But it's a good suggestion about the DHCP.
Thing is that the core library is trying to start reconnecting as soon as it boots.
That's before setup() is called.
That was one of the issues last week, that I received a disconnect event before even trying to connect.
It may wel be the DHCP should be disabled as soon as possible when using static IP.
Or perhaps turn wifi off, setup static IP, start wifi again in STA mode.

uzi18, 2018-04-18:
Have some reliable repeatable reports about task/controller settings malformed, but it was not possible for me to reproduce it on my unit,  need to investigate it but have no time.

DittelHome, 2018-04-18:
> I think there is a problem with the configuration of the static ip. Static IP and DHCP at the same time?

@Feuerreiter 
You use your own build with core 2.4.1. I have tested with the ready to use builds including core 2.4.0.
Therfore the behavior is maybe differnet....

TD-er, 2018-04-18:
There is no platformio build yet for 2.4.1

Feuerreiter, 2018-04-18:
Sorry. My fault. I builded it with the arduino ide. Today i would try it with platformio. Then i post the result here.

TD-er, 2018-04-18:
The CRC error reports also seem to occur with PlatformIO builds (you could try the nightly builds we make)
So I think it is an incorrect reported error.
It is simply not possible all those ESP's have faulty flash.

DittelHome, 2018-04-20:
OK, i noted that we should wait to gitcore 2.4.1 for the crc problem.
The static IP problem is not solved for my devices, using firmware ESP_Easy_mega-20180420_normal_ESP8266_4096 .
See here:
```
81 : INIT : Warm boot #3
83 : FS   : Mounting...
107 : FS   : Mount successful, used 75802 bytes of 957314
398 : CRC  : program checksum       ...OK
402 : CRC  : Settings CRC           ...FAIL
404 : CRC  : SecuritySettings CRC   ...OK 
405 : CRC  : binary has changed since last save of Settings
509 : INIT : Free RAM:20496
510 : INIT : I2C
510 : INIT : SPI not enabled
523 : INFO : Plugins: 47 [Normal] (core 2_4_0)
524 : EVENT: System#Wake
528 : WIFI : Switch on WiFi
529 : WIFI : Set WiFi to STA
530 : WIFI : Connecting im6shop attempt #0
532 : EVENT: System#Boot
539 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19528
3579 : WIFI : Connected! AP: im6shop (30:B5:C2:EB:DB:7D) Ch: 9 Duration: 3048 ms
3580 : IP   : Static IP : 192.168.1.17 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
3584 : EVENT: WiFi#ChangedAccesspoint
3590 : WIFI : Static IP: 192.168.1.17 (ESP-Easy-6) GW: 192.168.1.1 SN: 255.255.255.0   duration: 5 ms
3596 : EVENT: WiFi#Connected
3603 : Webserver: start
30539 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16448
60539 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16448
90539 : WD   : Uptime 2 ConnectFailures 0 FreeMem 16448
120539 : WD   : Uptime 2 ConnectFailures 0 FreeMem 16448
150539 : WD   : Uptime 3 ConnectFailures 0 FreeMem 16448
180539 : WD   : Uptime 3 ConnectFailures 0 FreeMem 16448
210539 : WD   : Uptime 4 ConnectFailures 0 FreeMem 16448
240539 : WD   : Uptime 4 ConnectFailures 0 FreeMem 16448
>wifidisconnect

Ok
261979 : EVENT: WiFi#Disconnected
261981 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4 m 18 s
261982 : Webserver: stop
261982 : WIFI : Connecting im6shop attempt #0
264234 : WIFI : Connected! AP: im6shop (30:B5:C2:EB:DB:7D) Ch: 9 Duration: 2249 ms
264235 : IP   : Static IP : 192.168.1.17 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
264239 : WIFI : Static IP: 192.168.1.17 (ESP-Easy-6) GW: 192.168.1.1 SN: 255.255.255.0   duration: 5 ms
264248 : EVENT: WiFi#Connected
264252 : Webserver: start
270539 : WD   : Uptime 5 ConnectFailures 0 FreeMem 16728
```
After wifidisconnect it is working...

TD-er, 2018-04-20:
Good to know some things do work after fixing :)
I have not looked at CRC issue yet, since I had a very busy day yesterday.

DittelHome, 2018-04-20:
> Good to know some things do work after fixing :)

It is early in the morning..., and you have had a very busy day ... ;)
Have youe read my last post ?

> The static IP problem is **not** solved for my devices, using firmware ESP_Easy_mega-20180420_normal_ESP8266_4096 .
> After wifidisconnect it is working...

What i want to say, same behavior, still wifidisconnect is needed in irder to work ..

TD-er, 2018-04-20:
OK, that's indeed something I noticed on one of my test nodes and maybe related to the 'known issues' I posted at the last commit.

Feuerreiter, 2018-04-20:
I do some research. WiFi.begin is called before WiFi.config is called. WiFi.begin try to connect with dhcp. In processConnect () WiFi.config should change the ip to a static one. If the dhcp reply from the Server arrives after the config command, it leads to this strange behavior. I think there is a problem with esp core library. For testing i changed the code. I call WiFi.config befor Wifi.begin. It solved the problem. Sry i am new at github and my english is not the best. Maybe it help you to solve the problem

TD-er, 2018-04-20:
(I only added markup correction to your post to get the layout of the code indented)

DittelHome, 2018-04-21:
Good job !!
Problem with static IP is fixed !!
I close, because the CRC thing is discussed on #1272

closed 2018-04-21 by DittelHome

Oxyandy, 2018-04-21:
@DittelHome Sorry about the dupe report, I should have searched first ;) 
still learning the whole Github environment..
I am using DHCP.. I am now getting crashes..