letscontrolit/ESPEasy #1256: Errors compiling playground plugins with mega-20180416

mstjerna, 2018-04-16:
### Summarize of the problem/feature request

Issues compiling plugins from Playground on later versions of MEGA.

### Expected behavior

No issues when compiling, the plugins where working in jannuary.
Did something major change?

### Actual behavior

C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino: In function 'boolean Plugin_118(byte, EventStruct*, String&)':
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:206:116: error: no matching function for call to 'addFormSelector(String&, const __FlashStringHelper*, const __FlashStringHelper*, int, String [2], int [2], byte&)'
addFormSelector(string, F("I2C Address"), F("plugin_118_i2c_address"), 2, options, optionValues, choice);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:206:116: note: candidates are:
C:/Temp/ESPEasy-mega/src/WebServer.ino:2562:6: note: void addFormSelector(const String&, const String&, int, const String*, const int*, int)
void addFormSelector(const String& label, const String& id, int optionCount, const String options[], const int indices[], int selectedIndex)
^
C:/Temp/ESPEasy-mega/src/WebServer.ino:2562:6: note:   candidate expects 6 arguments, 7 provided
C:/Temp/ESPEasy-mega/src/WebServer.ino:2567:6: note: void addFormSelector(const String&, const String&, int, const String*, const int*, const String*, int, boolean)
void addFormSelector(const String& label, const String& id, int optionCount, const String options[], const int indices[], const String attr[], int selectedIndex, boolean reloadonchange)
^
C:/Temp/ESPEasy-mega/src/WebServer.ino:2567:6: note:   candidate expects 8 arguments, 7 provided
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:212:147: error: no matching function for call to 'addFormSelector(String&, const __FlashStringHelper*, const __FlashStringHelper*, int, String [3], int [3], int&)'
addFormSelector(string, F("Take reading every"), F("plugin_118_read_frequency"), 3, frequencyOptions, frequencyValues, frequencyChoice);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:212:147: note: candidates are:
C:/Temp/ESPEasy-mega/src/WebServer.ino:2562:6: note: void addFormSelector(const String&, const String&, int, const String*, const int*, int)
void addFormSelector(const String& label, const String& id, int optionCount, const String options[], const int indices[], int selectedIndex)
^
C:/Temp/ESPEasy-mega/src/WebServer.ino:2562:6: note:   candidate expects 6 arguments, 7 provided
C:/Temp/ESPEasy-mega/src/WebServer.ino:2567:6: note: void addFormSelector(const String&, const String&, int, const String*, const int*, const String*, int, boolean)
void addFormSelector(const String& label, const String& id, int optionCount, const String options[], const int indices[], const String attr[], int selectedIndex, boolean reloadonchange)
^
C:/Temp/ESPEasy-mega/src/WebServer.ino:2567:6: note:   candidate expects 8 arguments, 7 provided
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:214:36: error: cannot convert 'String' to 'int' for argument '1' to 'void addFormSeparator(int)'
addFormSeparator(string);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:216:163: error: invalid conversion from 'const __FlashStringHelper*' to 'boolean {aka unsigned char}' [-fpermissive]
addFormCheckBox(string, F("Enable temp/humid compensation"), F("plugin_118_enable_compensation"), Settings.TaskDevicePluginConfig[event->TaskIndex][1]);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:216:163: error: too many arguments to function 'void addFormCheckBox(const String&, const String&, boolean)'
C:/Temp/ESPEasy-mega/src/WebServer.ino:2750:6: note: declared here
void addFormCheckBox(const String& label, const String& id, boolean checked)
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:217:122: error: too many arguments to function 'void addFormNote(const String&)'
addFormNote(string, F("If this is enabled, the Temperature and Humidity values below need to be configured."));
^
C:/Temp/ESPEasy-mega/src/WebServer.ino:2716:6: note: declared here
void addFormNote(const String& text)
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:221:121: error: invalid conversion from 'const __FlashStringHelper*' to 'int' [-fpermissive]
addTaskSelect(string, F("plugin_118_temperature_task"), Settings.TaskDevicePluginConfig[event->TaskIndex][2]);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:221:121: error: too many arguments to function 'void addTaskSelect(String, int)'
C:/Temp/ESPEasy-mega/src/WebServer.ino:2876:6: note: declared here
void addTaskSelect(String name,  int choice)
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:224:181: error: invalid conversion from 'const __FlashStringHelper*' to 'int' [-fpermissive]
addTaskValueSelect(string, F("plugin_118_temperature_value"), Settings.TaskDevicePluginConfig[event->TaskIndex][3], Settings.TaskDevicePluginConfig[event->TaskIndex][2]);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:224:181: error: too many arguments to function 'void addTaskValueSelect(String, int, byte)'
C:/Temp/ESPEasy-mega/src/WebServer.ino:2941:6: note: declared here
void addTaskValueSelect(String name, int choice, byte TaskIndex)
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:238:118: error: invalid conversion from 'const __FlashStringHelper*' to 'int' [-fpermissive]
addTaskSelect(string, F("plugin_118_humidity_task"), Settings.TaskDevicePluginConfig[event->TaskIndex][4]);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:238:118: error: too many arguments to function 'void addTaskSelect(String, int)'
C:/Temp/ESPEasy-mega/src/WebServer.ino:2876:6: note: declared here
void addTaskSelect(String name,  int choice)
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:241:178: error: invalid conversion from 'const __FlashStringHelper*' to 'int' [-fpermissive]
addTaskValueSelect(string, F("plugin_118_humidity_value"), Settings.TaskDevicePluginConfig[event->TaskIndex][5], Settings.TaskDevicePluginConfig[event->TaskIndex][4]);
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:241:178: error: too many arguments to function 'void addTaskValueSelect(String, int, byte)'
C:/Temp/ESPEasy-mega/src/WebServer.ino:2941:6: note: declared here
void addTaskValueSelect(String name, int choice, byte TaskIndex)
^
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:244:36: error: cannot convert 'String' to 'int' for argument '1' to 'void addFormSeparator(int)'
addFormSeparator(string);
^
Archiving .pioenvs\dev_ESP8266_1024\lib00c\libAdafruit_Sensor.a
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino: In member function 'String CCS811::getSensorError()':
C:/Temp/ESPEasy-mega/src/_P118_CCS811.ino:1009:1: warning: control reaches end of non-void function [-Wreturn-type]
}

### Steps to reproduce

1.  Download the latest build and add "_P118_CCS811.ino"
2. 
3. 

Hardware:
N/A

ESP Easy version: 
 mega-20180416

Br
Michael Stjerna

TD-er, 2018-04-16:
Yes those functions were changed 2 days ago.
I was not aware the playground got merged too.

mstjerna, 2018-04-16:
I just downloaded two plugins that I need for my solution:
_P117_Nextion.ino
_P118_CCS811.ino

Last time I build with the plugin it worked but as described in the issue it's not working anymore...

TD-er, 2018-04-16:
I have the Nextion plugin prepared to integrate into the main codebase.
A lot has changed in the last few months, so it may take some time and effort to integrate the CSS811 into the main codebase.
We may need to go through all those playground plugins to adapt them to work with the current codebase.

mstjerna, 2018-04-16:
Hi

I understand that it may not be a quick fix :-)
The CSS811 is a nice and cost effective chip. I would love to see this in the "standard" delivery so I don't need to custom-compile the build to get it in with every update..
Appreciate all the hard work everyone put in, and the newer releases have a loot of nice updates all the time

TD-er, 2018-04-16:
Was the implementation of the CSS811 working fine on the playground, in other words, should we mark it "testing" or could it be used by anyone?

mstjerna, 2018-04-16:
Yes it was working fine, I have bought Sensair S8 and a MH-Z19 to compare the Co2 values with the CSS811.
If I can get the CSS811 compiled I can post some graphs where the 3 sensors are put in the same room over a weeks time to see the differences.

TD-er, 2018-04-16:
You may test it: https://github.com/letscontrolit/ESPEasy/pull/1264

TD-er, 2019-11-04:
Nextion has been ported a while ago and CCS811 is in PR #2630 
So I will close this one.

closed 2019-11-04 by TD-er