letscontrolit/ESPEasy #1258: feature request : possibility to add a custom port to the syslog IP

Zixim, 2018-04-16:
My syslog server is listening on port 5140 , there's no way to specify a syslog port in ESPEasy.

TD-er, 2018-04-16:
Adding a portnumber to the settings is a little bit tricky.
It isn't that hard, but let me think about it for now.
I was planning to change the log entirely, but maybe the re-design of that should wait until after the next release and then it should not be a problem to add the port now.

cybermaus, 2019-08-15:
I would like to make the same request / bump this request up: Please allow for custom port for syslog.

I use cloud service "Papertrail" for syslog services, for years already, and they assign each of their customers a set of ports so they can distinguish between their customers. 

Small extra request: be allowed to use hostname as well as IP for the syslog IP

Build:⋄ | 20103 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20190813
Plugins:⋄ | 48 [Normal]
Build Md5: | 74939a896fe719b99f217da49cfa12
Md5 check: | passed.
Build Time:⋄ | Aug 13 2019 02:13:44
Binary Filename:⋄ | ESP_Easy_mega-20190813_normal_ESP8285_1M.bin

uzi18, 2019-08-16:
@TD-er it is simply possible by extending host address like host:port, then check in software if it contains ':'

TD-er, 2019-08-16:
If it is already a string and not an IP being stored, then sure it is a very simple fix.
But since @cybermaus did ask for a hostname, I don't think it is that easy.
Also check for max length of host being stored?

cybermaus, 2019-08-16:
While I understand you meant it as hint/proof the field is not a string, but please note  the hostname request is just a nicety. The port is the real request.

Also FYI: I did try the host:port and IP:port syntax already. But upon refresh the field back blank. Indeed only an IP address is accepted as value.

IoTPlay, 2020-04-07:
@TD-er - another vote from me. Ip address:port no for udp syslog.

Please..... pretty please?

mozgovs, 2020-06-29:
One more vote from me.
IP:PORT will be enough;)

dl2suf, 2020-07-10:
and another vote from me.

TD-er, 2020-07-19:
It has been implemented and will be merged soon.

PR #3148 (2020-07-20): [WiFi] Reduce wifi reset calls and call full wifi init at reset (#3147)
Fixes: #2931
Fixes: #1258
Fixes: #3155
Fixes: #3141
Fixes: #3111 

Connected state is now based on the state reported by the internal WiFi object.
Frequent calls to `reset()` of WiFi are now rate limited to make sure we don't end up in a loop.
commit eff8fd48 (2020-07-20): [Syslog] Add custom port to the syslog IP (#1258)

Fixes: #1258
closed 2020-07-20 by TD-er