letscontrolit/ESPEasy #1265: GPIO14 changes from '0' to '1' where it shouldn't..

Oxyandy, 2018-04-17:
### Summary of the problem
GPIO14 most definitely changes from '0' to '1' where it should not..
GPIO14 shows wrongly as '1' on device page & reports to Domoticz the same '1' as the ESP
This is a **working** log from Domoticz of the float SW I have across GPIO14
```
13:17:56 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
13:17:56 | On
(takes Float SW 2 or 3 seconds to lower)
13:17:58 | Off <- Float SW lowers changing from 'open circuit' to 'closed circuit'
13:17:58 | Off
```
Same log on >Dev_11 Firmware (or greater), but has a problem.
```
11:17:14 | On <- Float SW rises changing from 'closed circuit' to 'open circuit'
11:17:14 | On
(takes float 2 or 3 seconds to lower)-Change in the switch still happens at the right time BUT
11:17:17 | On <- Wrong, should be OFF
11:17:18 | On
```

### Expected behaviour
Have the change in the physical switch match what is represented on the Device page 
### Actual behavior
The wrong state is shown on the Device page & wrong state is reported to Domoticz 
### Steps to reproduce
Dev_11 Firmware default settings
Enable a Switch I use GPIO14, internal pulldown & Domoticz MQTT
I then using the Devices web-page & a jumper cable from GPIO14 to GND
Simulate On-Off-On, and then replace jumper as a 'Closed Circuit'
A few F5 presses while doing above, and it will soon show wrongly as '1'
### Does the problem persists after powering off and on?
Yes
### System configuration
In my case a Sonoff Basic with Dev_11 or greater firmware
Dev_4, 5, 6, 7, 8, 9 & 10 are confirmed to not have this issue.

So I tried all other controllers & could not replicate the issue
100% Domoticz MQQT code related. 
_C002.ino did not change between Dev_10 & Dev_11
_CPlugin_DomoticzHelper.ino did not exist
_P029_Output.ino changed in Dev_11

I created an archive to help
Inside are only the changed files from Dev_10 to Dev_11, all renamed in the format below to make the origin of each file perfectly clear - Extract contents of this archive to a single folder

> _P029_Output_Dev_10.ino
> _P029_Output_Dev_11.ino

Selecting both versions, compare with a visual text compare tool.
If someone has the time ? Maybe something stands out ?
Dev_10 works, Dev_11 the problem started.

[CHANGED_v2.0.0-dev11.zip](https://github.com/letscontrolit/ESPEasy/files/1888033/CHANGED_v2.0.0-dev11.zip)
Maybe it is time I tried a more recent release of the firmware too, 
see if any other changes have impacted anything..

The last firmware of the more recent one I have tried was 
ESP_Easy_mega-20180322_normal_ESP8266_1024.bin
then I started back tracking to find a point the bug vanished.
I have not yet tried anything released after this version.

Oxyandy, 2018-04-17:
Are these all related #330 #673 #674 #675 #924 ?
I have tried to reduce the amount of information on this new issue, 
but for more reading there is more detailed info here #1216

Oxyandy, 2018-04-18:
OK, completely related to this bug & I am thinking maybe it should be addressed before this one described above... because it relates completely to the Domoticz MQTT controller
However, it needs more research by me..
I want to backtrack older versions of firmware to see if it has always existed OR was introduced.

Normal reporting looks like this with Domoticz MQTT controller _**disabled**_.
Serial output from removing a jumper cable between GPIO14 & GND - wait - Then replacing it
SW   : Switch state 0 Output value 0
EVENT: FULL_Tray_2#Switch=0.00
SW   : Switch state 1 Output value 1
EVENT: FULL_Tray_2#Switch=1.00
Now enable Domoticz MQTT controller..
Serial output from removing a jumper cable between GPIO14 & GND - wait - Then replacing it
and here is that 'too busy' reporting I spoke about
```
SW   : Switch state 0 Output value 0
EVENT: FULL_Tray_2#Switch=0.00
EVENT: FULL_Tray_2#Switch=0.00
EVENT: FULL_Tray_2#Switch=0.00
EVENT: FULL_Tray_2#Switch=0.00
EVENT: FULL_Tray_2#Switch=0.00
SW   : Switch state 1 Output value 1
EVENT: FULL_Tray_2#Switch=1.00
EVENT: FULL_Tray_2#Switch=1.00
EVENT: FULL_Tray_2#Switch=1.00
EVENT: FULL_Tray_2#Switch=1.00
EVENT: FULL_Tray_2#Switch=1.00
```

TD-er, 2018-04-18:
Do you also use MQTT importer? (P037)

Oxyandy, 2018-04-18:
Ah finally found the point this 'excessive reporting bug' was introduced 
ESP_Easy_mega-20180212_normal_ESP8266_1024_DOUT.bin **OK**
```
Subscribed to: domoticz/out
EVENT: System#Boot
SW   : Switch state 0 Output value 0
EVENT: FULL_Tray_2#Switch=0.00
WD   : Uptime 0 ConnectFailures 0 FreeMem 20744
EVENT: Clock#Time=Thu,05:32
```
ESP_Easy_mega-20180213_normal_ESP8266_1024_DOUT.bin **BAD** starts too much reporting - 
```
Subscribed to: domoticz/out
EVENT: System#Boot
SW   : Switch state 0 Output value 0
EVENT: FULL_Tray_2#Switch=0.00
WD   : Uptime 0 ConnectFailures 0 FreeMem 20448
EVENT: Clock#Time=Thu,05:29
EVENT: FULL_Tray_2#Switch=0.00
EVENT: FULL_Tray_2#Switch=0.00
EVENT: FULL_Tray_2#Switch=0.00
EVENT: FULL_Tray_2#Switch=0.00
```
There was no need to even disconnect jumper cable on GPIO14, simply tick "Send Boot state"

TD-er, 2018-04-18:
Just to be clear, those builds with a date stamp in them are way later than dev-10 and dev-11.

So are we on the same topic again, or is this a completely different topic?
You lost me here.

Oxyandy, 2018-04-18:
No, I do not use "MQTT importer? (P037)"
I reduced everything to minimal to make your life easy...
mostly default, NTP, Wifi connection, DHCP, Domoticz MQTT controller
a switch with send boot state, not inverted, internal pull up ticked
That's all

Oxyandy, 2018-04-18:
Ok Dev_10 is stable, it does not have any bug.
Dev_11 is when GPIO14 gets stuck / wrong state shown, either 1 or 0, but because I normally have a closed circuit across it I didn't notice as much that it was also happening in reverse..
Every bug is related to Domoticz MQTT being enabled.
the 'excessive reporting bug' was introduced with
ESP_Easy_mega-20180213_normal_ESP8266_1024_DOUT.bin
I am going to try make a 
ESP_Easy_mega-20180213_normal_ESP8266_1024_DOUT.bin (FIXED wrong GPIO state build)

Oxyandy, 2018-04-18:
I try to compile ESP_Easy_mega-20180212 source code, in Atom
I get errors about lib/IRremoteESP8266 :(

TD-er, 2018-04-18:
Then you have probably a different version of this library installed.
I remember to have changed these library calls to min() and max().
Perhaps this was after feb 12, I guess

You can also disable (remove) the plugins calling this lib.

Oxyandy, 2018-04-18:
Yeah, I was just trying to "disable (remove) the plugins", but still not familiar enough with the source.

Oxyandy, 2018-04-18:
_P016_ & _P035_ got them

Oxyandy, 2018-04-18:
Ha, it doesn't seem to have the GPIO stuck / wrong bug on new compile :D
ESP_Easy_mega-20180212_normal_ESP8266_1024_DOUT.bin (FIXED "wrong GPIO state" build)
That's very satisfying - first build since Dev_10 that works right...

Oxyandy, 2018-04-18:
@TD-er 
Yes there are 2 Bugs relating to controller "Domoticz MQTT" being enabled
Closely related... 
I am happy to split them from this page and make a "New Issue" if you want ??

Oxyandy, 2018-04-18:
Ha, I am still awake, just dropped kid off to school.
I have a check to do before I sleep
No rush TD-er, I can use Dev_10, it is really only missing the email feature, which I do miss as an emergency alert..
Otherwise is very stable

Oxyandy, 2018-04-19:
10.45AM still awake.. 
The second bug which I called 'excessive reporting' was in:
ESP_Easy_mega-20180213
Controller.ino, line 172
was
 `     return;  // end loop if successful`
now
`      break; // end loop if successful`
and that stopped the "EXTRA" below.
```
Subscribed to: domoticz/out
EVENT: System#Boot
SW   : Switch state 0 Output value 0
EVENT: FULL_Tray_2#Switch=0.00 <- Correct: as above
WD   : Uptime 0 ConnectFailures 0 FreeMem 20448
EVENT: Clock#Time=Thu,05:29
EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
```

TD-er, 2018-04-19:
One tip, you can cite sourcecode from a specific branch.
The text you quoted does no longer exist in Controller.ino, so the context is hard to see without finding the version etc.

For example:
https://github.com/letscontrolit/ESPEasy/blob/40eb3e7ceb9586488c7c4c968ec1bd8ff15e3571/src/Controller.ino#L157-L174

Oxyandy, 2018-04-19:
Yeah I see what you have just done, no idea how to do that just yet, will try later..
So an easy fix for the 'aggressive reporting' ?
Think needs Domoticz MQTT enabled, not sure what else it impacts, steps to repeat are simple enough

TD-er, 2018-04-19:
I think I have to take a proper look at MQTT, as well as the controller as the importer.
It is just a bit funky how it's working now.

And about the link to source-code. Browse through the code on Github, select the right branch (or tag for this project) and click on a line number.
Then hold shift and select the last line to select.
Then on the [...] and select "use permalink" (or how it's called)
Just paste such a link in a topic here and press preview.

Oxyandy, 2018-04-19:
You are doing great work TD-er, keep it up !
Thank for the tips, between Github & PlatformIO there is so much I want to learn.
I am burnt out, only a few hours sleep today, 
now 10.48PM my local time I think it's obvious what comes next for me..

TD-er, 2018-04-19:
10.49 PM?

Oxyandy, 2018-04-21:
I finally go a newer release to load
Serial log from 
ESP_Easy_mega-2018**0420**_normal_ESP8266_1024.bin
Yeah I finally have a new release that loads.
Removing jumper from GPIO14, which creates a closed circuit to GND
```
16489 : SW   : Switch state 1 Output value 1
16491 : EVENT: FULL_Tray_2#Switch=1.00
16963 : EVENT: FULL_Tray_2#Switch=1.00 <-EXTRA
17002 : EVENT: FULL_Tray_2#Switch=1.00 <-EXTRA
17084 : EVENT: FULL_Tray_2#Switch=1.00 <-EXTRA
17121 : EVENT: FULL_Tray_2#Switch=1.00 <-EXTRA
18090 : SW   : Switch state 0 Output value 0
18092 : EVENT: FULL_Tray_2#Switch=0.00
18704 : EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
18775 : EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
18786 : EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
18801 : EVENT: FULL_Tray_2#Switch=0.00 <-EXTRA
```
I know you haven't had time, that's ok, but because I tested it, I wanted to let you know
The 'aggressive reporting' is still there
And as an extra note, those "EVENT:" then results in "double Wifi reporting transmissions"
back to Domoticz, resulting in 8 transmissions each time the switch changes.
So I went to Controllers, unticked enable "Domoticz MQTT" then serial log is correct.
```
877128 : SW   : Switch state 1 Output value 1
877130 : EVENT: FULL_Tray_2#Switch=1.00
879428 : SW   : Switch state 0 Output value 0
879430 : EVENT: FULL_Tray_2#Switch=0.00
```
So everything is as it was.

Oxyandy, 2018-04-29:
There are 2 bugs on this page "GPIO state wrong" + "Excessive reporting"
ESP_Easy_mega-20180429_normal_ESP8266_1024.bin
Still has the "Excessive reporting" which I [mentioned](https://github.com/letscontrolit/ESPEasy/issues/1265#issuecomment-382574727) above as the point it was introduced. 
Not sure about the other as it is more fiddly to test.
Should I split the 2 bugs ?

TD-er, 2018-04-29:
I think splitting the bugreports is best, just to get a good summary.

Oxyandy, 2018-04-29:
Oh my, I found the GPIO wrong state cause...
More on that later..
In the interest of trim & tidy I have begun to split the 2 issues, as I result I have deleted everything about the second bug... as that is all going to a second issue, if you like bug history minimal, you may wish to delete your now fragmented replies, sorry

TD-er, 2018-04-29:
I'm very curious about your findings :)
At the very least, no-one may accuse you of not preparing a bug report ;)

Oxyandy, 2018-04-29:
Ok, so I just flashed ESP_Easy_mega-20180429_normal_ESP8266_1024.bin onto a blank node
Get Wifi setup via AP then go into controllers I see

[image]

I only want to setup Domoticz MQTT, so I just ignore the first controller Domoticz HTTP, after all it's disabled right ? So just edit controller 2 & set that up as Domoticz MQTT, enter IP & submit - Done !

[image]

No problems right ?
@Grovkillen have you ever done this, but with a different controller..
Left the existing first entry Domoticz HTTP ?
Set up the second with what you need ?

TD-er, 2018-04-29:
There used to be a limitation on the MQTT controllers.
A MQTT controller could only be the first one.

I changed this a while back, to try and make it possible to have multiple MQTT controllers present.
Currently only one MQTT controller may be active and for some functions only the data of the first enabled MQTT controller will be used.

Your setup as shown in the screenshots should be a valid one, but I hope all calls to controllers will check for the 'enabled' checkmark.

Grovkillen, 2018-04-29:
@Oxyandy Yes I've done that, it's a 50/50 thing for me 😉 Shouldn't be any problem. 

@TD-er we would not have the Domoticz HTTP as a default controller, leave it blank as the other two.

Oxyandy, 2018-04-29:
Great  I am so glad you said "Yes I've done that, it's a 50/50 thing for me"
Cause it is a problem, well "confirmed" for me... I would need to test with your controller though.
What so you use ? Thanks
An explanation will come later, but for me it is a part of the cause of the wrong GPIO state bug.
(I will write it up tomorrow - while your side of the globe sleeps)
And your second comment saves me from saying that too, 
it should not be showing Domoticz HTTP on a fresh flash, all controllers should be blank.
I believe it was put there for a reason as a temp fix, that never was improved.
Dates back to Dev_11

Grovkillen, 2018-04-29:
I'm using openHAB MQTT on the second controller.

Oxyandy, 2018-04-29:
Great I will test quickly now

Oxyandy, 2018-04-29:
The other 'big' reason I was glad you said you have left it 'as is' and used the second 
was that I was feeling a little stupid for leaving it there too, 
so hearing you've done it as well - makes me feel suddenly 'smarter', haha
Anyway moving on..
I was hoping to find a repeatable scenario where this happened in Grokillen's usage, 
I couldn't last night - was falling asleep, I didn't want to wake up with the reverse imprint of letters on my face & forehead so crashed in bed instead..
TD-er I know what causes this now, but the real fix is beyond me
But I can point to the area in the code it happens
and yeah it seems the red X 'disable' is ignored with Domoticz HTTP,
it just needs to be there either 1st or 2nd controller 
I do not even know what "Domoticz HTTP" does ? (look more later)
Last night while testing to see if bug exists in 0429, I couldn't repeat it - Until I worked out what I just mentioned above, still happens in 0429, let's fix it.. more later, have work to do

Oxyandy, 2018-04-30:
Dev-10 works, Dev-11 this was added, how does it break GPIO state ?
Commenting the displayed lines out on 0429's source 
makes a firmware which I am sick of trying to get a stuck / wrong state  
This is not a proper fix, but it does solve the problem 
- but it seems so does editing Domoticz HTTP controller to become Domoticz MQTT, so Domoticz HTTP
isn't showing in devices even as "disabled"
### OLD
https://github.com/letscontrolit/ESPEasy/blob/4d28d2b8370e51833836b4d10ed071dc3ef13bc6/src/_P029_Output.ino#L42-L58
### CURRENT
https://github.com/letscontrolit/ESPEasy/blob/e60c0dc5f02c57ac7b460a3267f2880628215b7a/src/_P029_Output.ino#L43-L60

TD-er, 2018-04-30:
It is a very strange change.
It is 'fixing' something by not fixing it.
The protocol was compared with a plugin ID.
And now it is compared with a fixed number, which will break is the order of protocols is ever changed.

And my first impression is the `controllerNr` should be initialized with `CONTROLLER_MAX`, which can be considered as "not found".

I have to look at the code a bit longer to get it, what's happening here.

Oxyandy, 2018-04-30:
Yeah, the Domoticz code is a mess
in _C002.ino has references like ` // temp solution, if plugin 029, set gpio`
Have a fresh firmware show 3 controllers
The disabled controller still isnt really disabled
No delete button in controllers, is it needed ?

Grovkillen, 2018-04-30:
Set the first on to "stand alone", that's equal to none/delete.

Oxyandy, 2018-04-30:
Oh yeah,,, of course, that would work, it just I have never considered removing a controller until now..
In your case I wonder how things work, no Dom HTTP in first - no wrong state ?

Grovkillen, 2018-04-30:
I haven't tested that, maybe? Need to test when I get around to it.

Oxyandy, 2018-04-30:
So this whole issue could have been avoided for me by simply doing thing slightly differently..
Anyhow I am pleased I reached the point, that I know how to avoid it. ;)
Maybe will save others grief one day & makes a better ESpeasy for the future..
Looking forward to a close for this..
Now to complete the New Issue - "Excessive Reporting", cause I have a better understanding of that too.

Domosapiens, 2018-05-01:
> no Domoticz HTTP in first
Seems to me related to this comment from Hexenmeister:

[https://github.com/letscontrolit/ESPEasy/issues/590#issuecomment-344511468](url)
Task-data that overwrites Controller-data because of non-related fixed block sizes.
Hexenmeisters solution relates them to each other.

Grovkillen commented on Nov 15, 2017:
> Not in 2.0.0 since we're trying to close the version asap. But I'll add it to 2.1.0 :)

Oxyandy, 2018-05-02:
Nice, thanks for your input, I will read more later : https://github.com/letscontrolit/ESPEasy/issues/590#issuecomment-344511468

TD-er, 2018-05-02:
Nope, not yet.
The idea is to get some json import/export later.

Grovkillen, 2019-10-10:
Is this still an issue?

closed 2019-10-11 by Grovkillen