letscontrolit/ESPEasy #1270: feature request: Clock Time set using http

79kiko, 2018-04-17:
`http://xxx.xxx.xxx.xxx/control?cmd=event,clocktimeset=02:37`

Save the time value from http command to variable or dummy device.

Grovkillen, 2018-04-17:
This is a good suggestion. Storing it in a dummy is not possible currently since only float values are accepted. But definitely a good feature to have.

giig1967g, 2018-05-01:
What about: 
Rules:
setTime,23,20,03
SetDate,2018,05,01

Http:
http://xxx.xxx.xxx.xxx/control?cmd=setTime,23,20,03
http://xxx.xxx.xxx.xxx/control?cmd=setDate,2018,05,01

sassod, 2018-05-02:
Definitely a feature to have but....

It needs to be possible to set multiple time sets not just one. Kind of
dummy string for time in a manner of [TimeSetName#Value] so it can be used
in rules, cmd, lcd, etc.

I'm currently using 2 dummy values to concatenate to a time format in rules
for timerset... 😁 It works but too much time to implement and a waste of
resources for dummy strings

giig1967g, 2018-05-02:
@sassod : please explain better what you need the time for. I didn't understand...
@79kiko: is my suggestion what you were looking for?

79kiko, 2018-05-03:
@giig1967g : I do not care about the solution. I have android app for pool control and I need set start/end time of pump from my app. The solution can be but must work in rules.

giig1967g, 2018-05-03:
So you do not want to set the "system time" with this request, but to set a timer.
I will think about it.

79kiko, 2018-05-03:
Yes, I want set timer in rules.

giig1967g, 2018-05-03:
I can think of this simple approach that is compliant with the current limitations of the architecture (only numbers in device values):

Develop a new plugin (CustomTimer) that stores 4 timers in the following way:
Time=14:15, Day=Monday -> it will store the number (float) 11415 or 114.15
(where the first digit indicates the day of the week when the timer should fire [1=monday, 2=Tuesday...8=All] and the other digits indicate the time)
It will have its own http command: setCustomTimer,timer_number,timer_time.
Or we can even use: taskvalueset,x,y,114.15

This plugin could trigger the rules using the same syntax as now -> on [taskname#valuename] do and using the same logic developed for Clock#Time command.
Also we could use [taskname#valuename] in lcd and rules statements.

The only open issue could be the possibility of enabling and disabling a specific timer. 
We could use value=-1, and this could mean that that timer is disabled.

79kiko, 2018-05-03:
Now, I have Timer fixed in rules and activate it using dummy device. Im sending value to dummy device using http and rules. I activate The timer using dummy device, value=0 is OFF and value=1 is ON.
But your solution is good too.

giig1967g, 2018-05-03:
As we have a limit of 4 values for every device, we could define a device with two timers only where each timer uses two values:

Value1 = Timer1 enable/disable
Value2 = Timer1 time
Value3 = TImer2 enable/disable
Value4 = Timer2 time

In this case we will not overwrite everytime the timer time, but we get only 2 timers per device instead of 4.

giig1967g, 2018-05-03:
But it will only work if the unit is connected to the internet, isn't it?

TD-er, 2018-05-03:
Nope, given a date and a geo position, I can compute sunset and sunrise

giig1967g, 2018-05-03:
interesting. But isn't it more complex than what requested by the OP.
Especially in terms of user interface, with the current limitations?

79kiko, 2018-05-03:
@giig1967g That would be great. Two Timer value is super. 
For example:
1. Start Timer 
2. End Timer.

When I need more Timer so I use more device.

TD-er, 2018-05-03:
@giig1967g Some kind of computed value is not complex. Then you could write in the rules something like "sunset-30m", which is a value that's fixed during that day, so you have to compute it only once per day.

Adding some kind of complex notation to be able to read it from a float is far more complex and prone to errors and mis-understanding.  We have the source, and can change it the way we like, so I would like to stay away from complex solutions to make things work taking limitations into account.
I don't like limitations :)

giig1967g, 2018-05-03:
In this case I think we need two logics:
1) absolute timers, where you set start and end time of a timer
2) relative timer, where you set the number of minutes/hours/days starting from a specific date/time event.

The first can be done easily.
For the second we should consider the complexity of passing the string "sunset-30m" from http as requested by the OP.

susisstrolch, 2018-05-03:
I already thought about / started a DuskTilDawn Plugin with
```
    Config parameters:
      Longitude
      Latitude
      Zenith: official     90° 50' - sunup
              civil        96°
              nautical     102°    - twilight
              astronomical 108°
    
    Sensor values:
      Dusk    abs. seconds of day
      Dawn    abs. seconds of day
      isDay   true if Dawn <= now <= Dusk
      isNight !isDay
    
    Based on https://www.mikrocontroller.net/attachment/30393/sun.c
```
Format of Dusk / Dawn in absolute seconds of day (syssec_d).
So you can easily build rules like
```
on Clock#Time do
  if [zenith#Dawn] <= %syssec_d% and %syssec_d% <= [zenith#Dusk}
```
or alternatively 
```
  if [zenith#isDay]
```

Currently I set dusk/dawn via FHEM and taskvalueset, so the pond is controlled (nearly) self-sustained by the Sonoff 4CH.
```
on setDawn do taskValueSet,1,2,%eventvalue%
on setDusk do taskValueSet,1,3,%eventvalue%
on Clock#Time do
  if [zenith#Dawn] <= %syssec_d% and %syssec_d% <= [zenith#Dusk}
```

TD-er, 2020-06-28:
There is now a command to set the system time, so this one can be closed.

closed 2020-06-28 by TD-er