letscontrolit/ESPEasy #1274: No reconnect after a short failure of the wifi Network

micropet, 2018-04-18:
After a short failure of the wifi (about 0.2 seconds) there is no reconnect.
The device wants to start in AP mode.
Hardware Wemos D1 
Software ESP_Easy_mega-20180417_test_ESP8266_4096.bin

12584 : INFO : Plugins: 70 [Normal] [Testing] (core 2_4_0)
12584 : EVENT: System#Wake
12629 : WIFI : Connecting SMC attempt #0
12644 : EVENT: System#Boot
12652 : ACT  : 
12665 : ACT  : Publish ESP-201/IP,0.0.0.0
12676 : ACT  : timerSet,1,60
12686 : ACT  : 
12711 : WD   : Uptime 0 ConnectFailures 0 FreeMem 15368
12742 : DS   : Temperature: 20.62 (28-ff-b8-ea-b4-16-3-ed)
12744 : EVENT: DS18b20#Temperature=20.62
13688 : SW   : Switch state 0 Output value 0
13691 : EVENT: Motion#Pir=0.00
13706 : ACT  : gpio,14,0
13707 : SW   : GPIO 14 Set to 0
13966 : SW   : Switch state 0 Output value 0
13969 : EVENT: Led#Led2=0.00
15794 : WIFI : Connected! AP: SMC (78:8A:20:D1:9B:D9) Ch: 6 Duration: 3061 ms
15795 : EVENT: WiFi#ChangedAccesspoint
16463 : WIFI : DHCP IP: 192.168.0.201 (ESP-201-1) GW: 192.168.0.3 SN: 255.255.255.0   duration: 772 ms
16474 : EVENT: WiFi#Connected
16574 : MQTT : Intentional reconnect
16620 : MQTT : Connected to broker with client ID: ESPClient_5C:CF:7F:0B:68:52
16622 : Subscribed to: ESP-201/#
16623 : EVENT: MQTT#Connected
16932 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
16933 : EVENT: Time#Initialized
16953 : EVENT: Clock#Time=Wed,23:18
42710 : WD   : Uptime 1 ConnectFailures 0 FreeMem 14152
49770 : EVENT: Clock#Time=Wed,23:19
50909 : EVENT: WiFi#Disconnected
50925 : WIFI : Disconnected! Reason: 'Unspecified' Connected for 35 s
50925 : WIFI : Connecting SMC attempt #0
50941 : MQTT : Connection lost
50941 : EVENT: MQTT#Disconnected
52714 : EVENT: WiFi#Disconnected
52729 : WIFI : Disconnected! Reason: 'Auth fail' Connected for 1784 ms
52730 : WIFI : Connection Failed
52830 : WIFI : Set WiFi to AP
52835 : WIFI : Error while starting AP Mode with SSID: ESP-201_1 IP: 192.168.4.1
72710 : WD   : Uptime 1 ConnectFailures 0 FreeMem 14896
72865 : DS   : Temperature: 20.62 (28-ff-b8-ea-b4-16-3-ed)
72866 : EVENT: DS18b20#Temperature=20.62
72986 : EVENT: Rules#Timer=1
72995 : ACT  : Publish ESP-201/IP,0.0.0.0
73007 : ACT  : Publish ESP-201/MAC,5C:CF:7F:0B:68:52
73018 : ACT  : Publish ESP-201/Time,23:19:24
73030 : ACT  : Publish ESP-201/Uptime,1
73040 : ACT  : Publish ESP-201/RSSI,0
73050 : ACT  : Publish ESP-201/SSID,--
73061 : ACT  : Publish ESP-201/BSSID,00:00:00:00:00:00
73071 : ACT  : Publish ESP-201/CH,0
73082 : ACT  : Publish ESP-201/SYSHEAP,14520
73092 : ACT  : timerSet,1,60
77008 : SW   : Switch state 1 Output value 1
77012 : EVENT: Motion#Pir=1.00
79341 : SW   : Switch state 0 Output value 0
79343 : EVENT: Motion#Pir=0.00
79358 : ACT  : gpio,14,0
79360 : SW   : GPIO 14 Set to 0

micropet, 2018-04-20:
I do not know if you already did something.
The mistake is still there.

16388 : WIFI : Connected! AP: SMC (78:8A:20:D1:9B:D9) Ch: 6 Duration: 3787 ms
16389 : EVENT: WiFi#ChangedAccesspoint
16426 : WIFI : DHCP IP: 192.168.0.201 (ESP-201-1) GW: 192.168.0.3 SN: 255.255.255.0   duration: 34 ms
16437 : EVENT: WiFi#Connected
16456 : Webserver: start
16537 : MQTT : Intentional reconnect
16570 : MQTT : Connected to broker with client ID: ESPClient_5C:CF:7F:0B:68:52
16573 : Subscribed to: ESP-201/#
16574 : EVENT: MQTT#Connected
37538 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
37539 : EVENT: Time#Initialized
37562 : EVENT: Clock#Time=Fri,03:01
42678 : WD   : Uptime 1 ConnectFailures 0 FreeMem 20232
47470 : EVENT: WiFi#Disconnected
47489 : WIFI : Disconnected! Reason: '(1) Unspecified' Connected for 31 s
47490 : Webserver: stop
47491 : WIFI : Connecting SMC attempt #0
47511 : MQTT : Connection lost
47511 : EVENT: MQTT#Disconnected
49283 : EVENT: WiFi#Disconnected
49301 : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 1782 ms
49302 : WIFI : Connection Failed
49402 : WIFI : Set WiFi to AP
49408 : WIFI : Error while starting AP Mode with SSID: ESP-201_1 IP: 192.168.0.250

TD-er, 2018-04-20:
@workgroupengineering Is core 2.4.0 already using LWIP2?
I thought it was still 1.x

workgroupengineering, 2018-04-20:
Yes [this](https://github.com/esp8266/Arduino/releases/tag/2.4.0) is changes log.

TD-er, 2018-04-20:
On my Android phone, I have to tap an extra notification bar on the top of the screen to allow connecting to the ESP, since it is not connected to the internet.
That's on Android 8.1. (Nexus 5x) Not sure how older versions of Android handle this.

blb4github, 2018-04-22:
version mega-20180422 is working now for me on my Wemos D1 mini

TD-er, 2018-07-29:
Is this still an issue? If so, please re-open.

closed 2018-07-29 by TD-er