letscontrolit/ESPEasy #1279: ESP32 newping / sr-hc04 doesn't work

waspie, 2018-04-19:
*** ESP32 ***

compiling any of the new builds using the newping library results in a non working distance sensor.
arduino throws this warning:
WARNING: library NewPing claims to run on (avr, arm) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).

and serial log:
576363 : ULTRASONIC : Distance: No reading!

waspie, 2018-04-23:
I'm assuming I can just replace the sr04 plugin with one previous to newping and i'll be using the old plugin that works with the esp32 so I can use newer builds?

waspie, 2018-04-23:
i'll play around with it a bit more today. i brought an esp32 and the needed stuff with me today to do some testing. i tried copying the old plugin and got lots of compilation errors so that's out.

waspie, 2018-04-23:
*seems* to be working now. may have been the gpios i was using. will test on my controller at home that is known working with 2 distance sensors.

waspie, 2018-04-24:
i'm going to close this. while i'm not brave enough to try the current release on my now perfectly working esp32 it worked fine all day on the bench.

closed 2018-04-24 by waspie

MaxESP, 2018-09-06:
hello guys,
I want to merge an important project to esp32 from esp8266

I have seen some where that i have to put 

#include "NewPingESP8266.h" for esp32

instead of normal .

When i compile i have an error on :

NewPingESP8266 sonar(TRIGGER_PIN, ECHO_PIN, MAX_DISTANCE);