letscontrolit/ESPEasy #1282: ESP32 i2c stops working

waspie, 2018-04-19:
*** ESP32 ***

I'm still working on what conditions this happens. For sure it happens with a TSL2561 connected.
I don't think i'm seeing the same behavior with just 2 MCP23017s connected but I need to disconnect the TSL2561 when I get home and see if it still happens but I'm pretty sure the TSL2561 is the cause. The particular piece of hardware has been working reliably for months on end so I don't think its the TSL2561 itself but something software/esp32 related.

I've used the precompiled binary from late march and my own compiled versions and the behavior is the same. 

With my 3 i2c devices connected they work for a couple minutes (at best) but soon after they all disappear and an i2c scan shows no devices. 

On my own compiled versions i'm running just 6 plugins. 

Build | 20100  - Mega32
-- | --
GIT version | (custom)
Plugins | 6 [Normal]
Build Md5 | 4d44355f4d44355f4d44355f4d44355f
Md5 check | fail !
Build time | Apr 18 2018 23:03:07  **<----built from April 2 source**
Binary filename | ThisIsTheDummyPlaceHolderForTheBinaryFilename...
ESP board
ESP Chip ID |  
Storage
Flash Chip ID |  
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash Writes | 0 daily / 0 boot
Sketch Size

i2c is on pin 4/5 (default) 
No I2C devices found

[image]

Budman1758, 2018-04-19:
I have the same problem on a test unit also. The I2C stops working after what seems to be a random amount of time. I have an OLED display and an MCP23017 connected.  I have seen anywhere from 800 to about 1400 minutes uptime until it quits.  When this happens I have the same results in the I2C scanner page. No I2C devices found.  This is the same unit that I reported issue #1029 with.  There is also a Dallas temp sensor connected but it is not affected as far as I can tell.

Budman1758, 2018-04-19:
I have resistors on my setup.

TD-er, 2018-04-19:
The devices pull the signal to ground to communicate, so in order to get a good signal quality, you may add pull up resistors to the signal lines.
Maybe the esp32 does not do internal pull up like the esp8266 or other Arduino chips do.

It is advisable to add these when using more I2C devices on the same bus and some chips need it more than others.
Also when using longer signal lines.

waspie, 2018-04-19:
ok, i will try that. it was never an issue with all the 8266s i've hooked up so i had no idea!

TD-er, 2018-04-19:
Pull up between I2C line and 3v3 line.
Just to make sure :)

waspie, 2018-04-19:
I unhooked the TSL and the MCPs stayed working for a few hours. I just now hooked up the 10ks and plugged the TSL back in and immediately both MCPs and TSL are visible in the scan which hasn't been the case previously. cautiously optimistic at the moment. I'll check back in on this topic tomorrow.

waspie, 2018-04-19:
spoke too soon. drat

ferazambuja, 2018-04-20:
I tested with two different kinds of boards and the I2C scanner reads the wrong addresses. I'm using a Sparkfun ESP32 Thing and a Wemos Lolin32. How do I build for ESP32 with more plugins (like dev or test)?

waspie, 2018-04-20:
I may be way off base here, but while searching around for pullup resistor values on the ESP32 i2c lines I found this topic: https://esp32.com/viewtopic.php?f=13&t=1308&sid=8429201731480ecffca91d3ab827d59a&start=10

and it seemed like useful information.

TD-er, 2018-04-20:
Doesn't have to be that far-fetched. The suggestion of the pull-ups was more to make sure it wasn't some signal quality issue.
It may very well be related to timing, which may affect some sensors more than others.

waspie, 2018-04-20:
i modified the freertos definition from 1000 to 100 which was a bad idea. everything ran much faster, so i instead modified I2CDEV_DEFAULT_READ_TIMEOUT from 1000 to 100 in i2cdev.h and uploaded it an esp32 on my desk at work and the TSL has been working continuously  for about 20 minutes with no PU resistors (didnt think to bring any to work) which is far better luck than i've had thus far.

i wonder if the problem is we're not using enough pull up (i see a lot of references to using 4.7k). so at 10k the pull up is too lazy still but 4.7k is much faster (or modify the timeout).
modifying the timeout would affect other things you compile with i2c though so the resistors would seem like the better option?

TD-er, 2018-04-20:
10k pull up is indeed a bit conservative.
But compared to having none, it is already an improvement.
You also don't want to have too strong pull up, since the sensors and the ESP must still be able to pull it down to GND.
That "10k" was just a "better than nothing, if it's not working at least it doesn't harm anything".
If you need 4k7, please do and let's hope that makes it operate more stable.

TD-er, 2018-04-20:
If you have multiple 10k's you could place 2 parallel to get 5k ;)
I guess the data line is the most sensitive, so you could set both 10k's to the data line.

waspie, 2018-04-20:
also found this. i'm sure there's a lot more and your google works as well as mine so i'll try not to keep bugging you too much. i dont know a lot of what they're saying but maybe you(? or whoever?) can make sense of what they did to fix it?
https://github.com/espressif/esp-idf/issues/680

waspie, 2018-04-20:
further down the rabbit hole... i found this
https://github.com/stickbreaker/arduino-esp32/commit/798e73b9b3665a93157f0abd35763a3c85ac7a88

its referenced in a very long thread about i2c bus hanging but this time there is a file changed that i actually see on my system! i'm going to replace the current file with this and retest.

waspie, 2018-04-20:
tried with 4.7k PUR. tried with modified i2c hal c file. doesn't matter, they all stop working. kind of at the end of my ability for the moment.

TD-er, 2018-04-20:
Can you monitor the memory usage?
My ESP32 board was crashed (completely frozen) and in the Domoticz log, I could see the free memory was quite a bit less than before.
The same setup on an ESP8266 gave a nice flatline (no trend) with the occasional dip or peak, which is perfectly fine, which indicates there is no leak in our part of the software, so it must be in the core library.

waspie, 2018-04-21:
I'm once again sticking my foot in my mouth. The changes are necessary in several more files.  You're supposed to replace the esp32 core with this guy's entire rewrite. I tried it earlier but my esp became rather slow,  albeit I kept seeing measurements come across mqtt.  I rebuilt it all again,  fresh,  tonight and it's been responsive and working for 3 hours already which is far better than before.  I was usually lucky to make it 30 minutes.  

I'll check in later

waspie, 2018-04-21:
12 hours,  still working. Previously 20 minutes was a milestone.

TD-er, 2018-04-21:
Can you explain what you did, or post a link to instructions what to do?

waspie, 2018-04-21:
I fully intend to when I'm satisfied that it has resolved the issue

waspie, 2018-04-21:
Do you still want me to start logging memory usage?

TD-er, 2018-04-21:
Well it may help in determine the cause of the next crash :)
Some report crashes after N minutes and my ESP32 does seem to crash after N hours, probably due to memory issues.
Although I am not entirely sure what to monitor, since the free reported memory may be quite high, but allocations on the heap may fail due to heap fragmentation.

waspie, 2018-04-23:
I don't know where you want to post this or what editing you want to do...

These instructions are for a Windows machine. OS X and Linux will be different (naturally). If you're using linux you ought to be clever enough to adapt these instructions to your needs. 

What is this fix and why do I need it? 
If you use the ESP32 and are using i2c devices, you need this fix.
Why?
Because espressif needs to fix their code regarding i2c devices releasing the bus after information is transmitted. At some point espressif will fix it or it may get merged from this guy's fixes.
What do you need to do to "fix" it?
Download and overwrite your existing arduino ide build environment for the esp32 with this code.
Go to: https://github.com/stickbreaker/arduino-esp32 and clone the archive (download as zip, whatever).

If you're using the arduino ide as a "portable" install your esp32 build environment will be c:\whateverfolderarduinoisin\hardware\espressif\esp32
[image]

The archive that you cloned above will have this same folder structure. Extract those files overwriting the files in the above directory in this example. 

If you installed arduino your folder *should* be c:\users\YOURUSERNAME\documents\arduino\hardware\espressif\esp32 (i think, i don't have it "installed" so I don't know off the top of my head for sure).

That's it. Assuming you previously followed the instructions to installed the esp32 environment you're all set. Rebuild your firmware and you'll have no more i2c breakages.

waspie, 2018-04-23:
This is not an ESPEasy problem. Solved with different esp32 core

Budman1758, 2018-04-24:
@waspie Any chance you could make a bin file I could download? I do a lot of testing\jacking around with this stuff but don't have a clue how to compile ESP32.

waspie, 2018-04-24:
Compiled from source downloaded as of 4 24 2018 7:30A EST

[ESPEasy.ino.esp32.zip](https://github.com/letscontrolit/ESPEasy/files/1942578/ESPEasy.ino.esp32.zip)

minus these plugins due to compilation errors (missing libraries on my end)
[image]

Budman1758, 2018-04-24:
Thanks a bunch!!

Budman1758, 2018-04-24:
@waspie  Are you having any problems with the version in your bin file?  I keep getting a crash and reboot when I try to access device 1.  Pretty consistent. I also had a bit of difficulty getting it flashed at first.  What do you use to flash your 32's?

waspie, 2018-04-24:
Anything passed April 3rd? Ish seems to have this instability.  I'm
personally running April 2nd source.  If you'd rather have bin from that
source let me know.  I use flash serial script or esptool

waspie, 2018-04-24:
> @waspie <https://github.com/waspie> A bin that you are using would be
> great. An example of your command line for esptool would be pretty awesome
> too. :>)

Yeah hang tight until about 8est. Home w kids now

waspie, 2018-04-25:
i use the flashserial.cmd that came with one of the releases?? plop it in the same folder as esptool.exe
use the bin file as named in the same folder, then run flashserial.cmd. it asks which com port (use just the ##) then which release type "20100" that should do it. this is april 2nd source with most plugins. missing 10 16 35 49 52 65 mostly the same as before. all others included, all controllers all notis. 

it still crashes _occasionally_ when adding devices but probably 95% less than newer builds. allow it to reboot and reload the page or go to the next device (if configuring task 5, try configuring 6 instead... its goofy but whatever. once its configured and working who cares?).
mine has about 48 hours uptime, would have been longer but i pulled the plug accidentally. all i2c devices still work.  i included flashserial.cmd in case you dont have it. good luck
[ESP32flash.zip](https://github.com/letscontrolit/ESPEasy/files/1945078/ESP32flash.zip)

Budman1758, 2018-04-25:
Thanks for the bin file.  Don't know why but it crashes about as much as the other. I think I'm gonna set my 32's aside fir a bit until the code is a bit more mature.  Need to concentrate on getting a bunch of other stuff already built deployed.

waspie, 2018-04-25:
@Budman1758 what are you doing to cause the crashes? what controller do you use and what plugins do you _need?_ I compile mine with only OH controller and about 6 plugins.

Budman1758, 2018-04-25:
@waspie The crashes are happening right away after flashing. I'm just navigating around the web pages. It crashes 99% of the time just clicking on the edit button for task 1 from the devices main page.  Also crashes on other task edit buttons but not as much.  I don't have any devices set up yet.  No controllers.

Budman1758, 2018-04-25:
@waspie Do you happen to use the framed OLED?  Was wondering if the I2C fix fixes this issue. #1029 ?

ferazambuja, 2018-04-25:
Mine also crashes after a few clicks around the web interface. I also don’t have anything hookup. I’m using the compiling formula available on platform IO. I will try to get a log later.

waspie, 2018-04-25:
@Budman1758 , i would imagine.

Like I said, if you tell me what plugins (numbers) and controller I'll whip another bin for you to try. 

You may still get an occasional crash on a working version of my bin but it shouldn't be the norm.

Budman1758, 2018-04-25:
@waspie The plugins in the second bin you made are fine. The seven segment display would be nice if possible.  Can't find the number I don't use a controller right now but if I did it would probably be Domotiz.

waspie, 2018-04-25:
no notis, just domo and oh controllers and plugins in the attached picture. 
[image]

[ESPEasy32_R20100.zip](https://github.com/letscontrolit/ESPEasy/files/1947970/ESPEasy32_R20100.zip)

TD-er, 2018-04-25:
I have the oled framed plugin running on my ESP32 and it works just fine.
I can upload a binary version later this evening.

Budman1758, 2018-04-25:
I'll give it a spin later.  Thanks.

Budman1758, 2018-04-25:
@TD-er That would be awesome!

Budman1758, 2018-04-25:
@waspie Thanks again!!  Gonna give it a spin here in a couple...

TD-er, 2018-04-25:
@Budman1758 
https://www.dropbox.com/s/972eoa9c72bgngw/esp32dev_20180425.rar?dl=0
Just made the build I use on my ESP32 node.
That one is running with an OLED framed plugin, MQTT and some generic plugin to log the memory usage.

DL6DBZ, 2018-08-20:
Is there any new esp easy firmware for the ESP32 available ? I used the ESPEasy32_R20100 but I still has problems with the framed OLED in combination with an BME280 on the same I2C bus. It always stuck in the I2C routines if both devices are active. With the standard OLED 1306 it is working also in combination with the BME280.

Thanks for you help in advance. The dropbox link from Budman1758 does not work.

TD-er, 2018-08-20:
Please only close issues when they need to be closed, not when asking a question.

TD-er, 2018-08-20:
About your question by the way, in the nightly builds there is also a version for ESP32 included.
But it has still issues with I2C, at least when using more than one device.
The last release of PlatformIO for ESP32 I tried about a week ago could not be built, so I cannot test if it has been working in the last core libraries.

TD-er, 2018-08-21:
I have tried to build with the 1.3.0 a few days ago but wasn't able to compile ESP easy.
And 1.2.0 caused all kinds or crashes when accessing the device config.

waspie, 2018-08-21:
Well crap, that stinks. I stopped using esp32s a few months ago because of the instability so I'm of no help then :\ sorry about that

TD-er, 2018-08-22:
I really would like to use the ESP32 also, since a lot of nice boards are available with that CPU, like the M5stack boards.  Also that CPU has so much more potential, so I really really really like to have it supported in ESPeasy.
But the I2C problems are one reason it isn't yet an active supported platform for ESPeasy.
There are other issues too, but those are manageable.

For now I am working mainly on the ESP8266 to get it all stable to get to release a version.
So that's my main focus now.

TD-er, 2019-10-27:
I2C has been stable for a long time on ESP32 So will close this issue.

closed 2019-10-27 by TD-er