letscontrolit/ESPEasy #1284: OTA update not working

wdonker, 2018-04-20:
Tried to load new firmware in my NodeMCU 4MB, from mega-20180417 to mega-20180420 but it did not work. After reboot 20180417 is still active.
Serial log shows:

```
30654 : WD   : Uptime 1 ConnectFailures 0 FreeMem 15952
Update: ESP_Easy_mega-20180420_normal_ESP8266_4096.bin
                                                      sleep disable
                                                                   ......................................................................................................................................................
LmacRxBlk:1
LmacRxBlk:1
etc.

```

Updating with the flash tool works fine.

TD-er, 2018-04-20:
That's strange. I did not change anything related to this and did some OTA update on my nodes a few days ago.

Do you have deep sleep enabled?

Grovkillen, 2018-04-20:
I would test and then test again. Sometimes you get no response from the webserver but a new try must likely will work. If you monitor the serial log during the OTA you'll understand the behavior better.

wdonker, 2018-04-20:
Well, the serial log showed the errors above. That does not make me understand the behavior better ;-).
But I tried another unit that was still on mega-20180319 and that one did the OTA flash fine.
So maybe it is an incidental problem indeed.

Grovkillen, 2018-04-20:
Hehe... true! :)

But as you stated, it's possibly a "from-time-to-time-glitch". I close this issue for now.

closed 2018-04-20 by Grovkillen