letscontrolit/ESPEasy #1291: Webserver unreachable with static IP set.

aero83, 2018-04-21:
### Summarize of the problem/feature request

WEmos D1 Mini. 
When I set static ip, webserver unreachable, serial debug output as follows.  Always shows NTP server unreachable. 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
⸮⸮⸮U60 : 

INIT : Booting version: mega-20180420 (core 2_4_0)
60 : INIT : Warm boot #7
61 : FS   : Mounting...
86 : FS   : Mount successful, used 75802 bytes of 957314
405 : CRC  : program checksum       ...OK
409 : CRC  : Settings CRC           ...FAIL
411 : CRC  : SecuritySettings CRC   ...OK 
517 : INIT : Free RAM:18432
518 : INIT : I2C
518 : INIT : SPI not enabled
534 : INFO : Plugins: 72 [Normal] [Testing] [Development] (core 2_4_0)
534 : WIFI : Switch on WiFi
535 : WIFI : Set WiFi to STA
536 : WIFI : Connecting NSA attempt #0
555 : WD   : Uptime 0 ConnectFailures 0 FreeMem 17696
3614 : WIFI : Connected! AP: NSA (2C:30:33:F6:EA:27) Ch: 6 Duration: 3076 ms
3614 : IP   : Static IP : 192.168.10.60 GW: 192.168.10.1 SN: 255.255.255.0 DNS: 192.168.10.1
3618 : WIFI : Static IP: 192.168.10.60 (ESP-Easy-0) GW: 192.168.10.1 SN: 255.255.255.0   duration: 5 ms
3640 : Webserver: start
14219 : Host unreachable: 0.0.0.0
14219 : NTP  : NTP host 0.us.pool.ntp.org (0.0.0.0) unreachable

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. Flash mega-20180420 (core 2_4_0), dev 4096 also occurred on 2018417 build
2. Set static ip.
3. reboot
4. unreachable webserver and serial debug output above.

### System configuration

Hardware:

ESP Easy version: mega-20180420 (core 2_4_0) also occurred on 2018417 build

ESP Easy settings/screenshots: 

### Rules or log data

```

```

Budman1758, 2018-04-21:
I had the exact same problem.

TD-er, 2018-04-21:
Yesterday I added a commit that should solve this.
Could you test with the latest code?

stafil, 2018-04-21:
I try latest release, bat not working. thx

v-a-d-e-r, 2018-04-21:
No problems here with static IP.  :grin:

TD-er, 2018-04-21:
@stafil What do you mean with "bat not working" ?

ghtester, 2018-04-21:
Hello, I have also encountered an issue after upgrade from mega 2.0.0 dev 11 to mega-20180421 test on NodeMCU Amica. There's a rule which beeps on a boot and use IR receiver plugin to control relay.  No other plugin is used. Also the Fixed IP Octet is configured. After the successfull OTA firmware upgrade the device rebooted itself, then I hearded 3 beeps (so three times rebooted), then I could access it on configyred fixed IP address by PING but the web interface is not accessible anymore.  When unplugged from power and plug again, several beeps heard (first almost immediately, next after about 10 sec) until device get connected to AP. The IR plugin and rule looks working but the web interface is dead.

giig1967g, 2018-04-21:
Hi am facing the same problem: no access to the web interface.
I use static IP.
Everyting was fine until 20180402.
Yesterday's version didn't solve for me.

giig1967g, 2018-04-21:
Tried 20180421 version with DHCP but could not access webinterface either.

v-a-d-e-r, 2018-04-21:
I don't know what you guys are doing, but I flashed the firmware from tonight, just to reproduce your issues, but it works!
- Wifi setup with AP mode: ok
- DHCP: ok
- Static IP: ok
- Web interface: ok
TD-er made all right. So you are making something wrong!

TD-er, 2018-04-21:
Like I said in another thread, there are a lot of issues which are hard to reproduce.
So could you please give more information on what settings are being used.
Also used plugins and controllers and notifications (email/buzzer)

v-a-d-e-r, 2018-04-21:
This was a test under real conditions. I installed it on a fresh and clean flash chip. Only so you can be sure, that there is no crap left from a previous installation or config. ;-)

Feuerreiter, 2018-04-21:
I have check both settings. For me DHCP and Static IP work. I do a reset (factoryreset) between the to settings. My Version is mega-20180421 build in PlatformIo. Hardware is NodeMCU 1.0 Amica ESP8266.

szwigiel, 2018-04-21:
Feuerreiter,
First try to totally clear your ESP through "ESP FLASH DOWNLOAD TOOL V3.6.4". It helped me.

ghtester, 2018-04-22:
Just tried to reset settings through serial console, configured only wifissid and wifikey, saved settings and rebooted.  After reboot ESP connected to AP, reported Settings CRC FAIL, it's accessible via PING but the web console is still dead...

Any hint what could be tested except total ESP clear?

INIT : Booting version: mega-20180421 (ESP82xx Core 2_4_0)
65 : INIT : Warm boot #1
66 : FS   : Mounting...
91 : FS   : Mount successful, used 75802 bytes of 957314
409 : CRC  : program checksum       ...OK
413 : CRC  : Settings CRC           ...FAIL
442 : CRC  : SecuritySettings CRC   ...OK
548 : INIT : Free RAM:18384
548 : INIT : I2C
548 : INIT : SPI not enabled
564 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_0)
565 : WIFI : Switch on WiFi
565 : WIFI : Set WiFi to STA
567 : WIFI : Connecting IOTAP attempt #0
574 : WD   : Uptime 0 ConnectFailures 0 FreeMem 17464
4548 : WIFI : Connected! AP: IOTAP (xx:xx:xx:xx:xx:xx) Ch: 4 Duration: 3980 ms
8196 : WIFI : DHCP IP: 192.168.0.14 (ESP-Easy-0) GW: 192.168.0.1 SN: 255.255.255.0   duration: 3647 ms
8199 : Webserver: start

ghtester, 2018-04-22:
Yes that's the latest release with settings CRC, I did not update to today's code yet just for the case if there's anything to be debugged.
I think there's no problem with connection in my case as the ESP device is still PINGable but the webserver does not work. It's strange as the configuration was reset to defaults, only primary wifi SSID / key was entered through serial console and the IP address is DHCP.

IlluminatMartin, 2018-05-01:
Hi,
I has also instaled today the last version (ESP_Easy_mega-20180501_test_ESP8266_4096) and my static IP is also not working. the modules use the IP from DHCP what they get from the initial phase.

TD-er, 2018-05-01:
Can you add a log via serial output, with the log level set to "Debug More" ?
That level does include the internal log of the Arduino library, which states enable/disable of DHCPclient.

Is it a binary built in the daily build, or one you built yourself?

IlluminatMartin, 2018-05-04:
Hi,
in oposite to older versions it works it works if you (or I) fill every value (GW, Subnet, DNS) .
In older versions it was enought to fill just the IP field.
So I "solved" my problem

thanks
Andy

TD-er, 2018-05-04:
OK, that's an interesting point you make.
Maybe we should make some kind of explicit enable/disable static IP and set some kind of defaults for the static settings, which can be changed if desired.
Or... we must warn the user if not all is set.