letscontrolit/ESPEasy #1296: Issues saving rules - Rules page bug?

giig1967g, 2018-04-22:
Hi,
I am testing latest mega-20180422.
I have issues saving rules in the "Rules" webpage.

When I press "Submit" either all the text of the rule disappear or is trunked to something around 1500 chars (number of chars varies everytime).

I have checked and there are no special chars in the text.

If I upload the same rules1.txt from the Tools menu, it works.
So must be something related to the web page

TD-er, 2018-04-22:
Does the same thing happen now that you've got the rules uploaded to the filesystem and try to save them again?
Can you attach the rules? Also binary file.

giig1967g, 2018-04-22:
Hi,
yes it still happens. If I try to change anything and then press "Submit" the rules section is blank and the file size in the Tools->Flash filelist is 0 bytes long.

[rules2.txt](https://github.com/letscontrolit/ESPEasy/files/1935597/rules2.txt)

[config_ESPT6_U16_Build20102_201814.dat.txt](https://github.com/letscontrolit/ESPEasy/files/1935598/config_ESPT6_U16_Build20102_201814.dat.txt)

Budman1758, 2018-04-22:
I can confirm this. Has happened to me a bunch of times. Tried in Chrome and Internet Explorer.  Uploading files via the "files on flash" page works.

Grovkillen, 2018-04-22:
And if you copy the text to be saved to a Notepad++ window and see if any unseen characters are added?

TD-er, 2018-04-22:
It could have something to do with the size of some buffers.
Either in the webserver, or network stack.
The headers of a POST request are roughly 100 bytes and the MTU size is normally 1492 bytes.
So there is roughly 1400 bytes left as payload in 1 packet.
It could be that exceeding this limit is causing some issues.

micropet, 2018-04-22:
The lower half of the rules disappeared since today's update.

Budman1758, 2018-04-22:
@Grovkillen I use Notepad++ to type the rules and then upload them.  I don't seem to have any problems with the rules working.  How do you know if there are unseen characters?

Grovkillen, 2018-04-22:
You can turn on to have all characters showing. 

View->Show Symbol->Show All Characters

Budman1758, 2018-04-22:
Only unseen in my rules page is  CRLF.  At the end of each line.  (enter)

Grovkillen, 2018-04-22:
Then we can rule that one out. Thanks!

TD-er, 2018-04-22:
@micropet 

> The lower half of the rules disappeared since today's update.

I know you rebuild almost on every commit, so what update do you mean?
What was the last commit you had when building?

v-a-d-e-r, 2018-04-22:
No issue here. Filled up the text area with all 2048 chars and saved it. File size on flash and HD: 2048 bytes. Browser: Firefox.

v-a-d-e-r, 2018-04-22:
Any blocker for Javascript active?

giig1967g, 2018-04-22:
@Budman1758 : same for me.
Sometimes blank rule, sometimes rules truncated.

TD-er, 2018-04-22:
And this wasn't happening on the previous build?
As far as I know, there was nothing changed when saving the rules. Or there must have been something reading the rules before they are saved.

Is any of these also on builds we made, or only on builds self-made?
That may differ on what commits were included.

micropet, 2018-04-22:
Yes, I do an update almost every day.
I am not sure when the problem occurred.

I do not always check the rules.

TD-er, 2018-04-22:
As long as you follow them.... ;)

Budman1758, 2018-04-22:
@TD-er  I actually have seen this i previous builds.  It seemed to be more random though.  I noticed it would do it more when the rules page had been displayed in a browser tab for a long time vs if it was just loaded, edited and then saved.  There were times I had a tab open for say a couple hours and if I hit submit it would come back blank. But if I loaded, edit, and then submit within a couple minutes it would work as it should.  Now its consistent and repeatable.

TD-er, 2018-04-22:
Also with small rule sets?

micropet, 2018-04-22:
I always have the same, quite long rules, sorry.

Budman1758, 2018-04-22:
Does not seem to make any difference as far as I have seen.

TD-er, 2018-04-22:
Just tested here at an ESP32 and that seems to work just fine.
Tested both as rules1.txt and rules2.txt.
I will test it also on an ESP8266

What build do you guys use?
Normal/dev/testing? (guessing free memory may be an issue)

TD-er, 2018-04-22:
Confirmed that ESP8266 does do as reported.
It uses exactly the same source as the ESP32, so it must be something in the core libraries.

giig1967g, 2018-04-22:
It happens also on ESP8285.
I use dev.

Budman1758, 2018-04-22:
I am using the test 4096-VCC versions.

TD-er, 2018-04-22:
I just updated to the staged version of PlatformIO (should be core 2.4.1 since 2 days) and now the uploaded file is reduced to 1471 Bytes. (also had 1405-something bytes once)
That's about the size of a single packet minus some overhead.

Oxyandy, 2018-04-22:
Remember I had that bug, I posted with the red bar at top of webpage... 
something about can not save config ?
I could dig up what version that was...? No, was not my hardware & had it happen on several units..
The version after that, any change to rules, no error, but rules would then often be blank after a text update & submit..
At that point, I started back tracking until I learnt Dev_10 was rock solid for my usage scenario

TD-er, 2018-04-22:
@Oxyandy That sounds like "flash write protection" kicking in.
That happens when you have 100x written in the last 24 hours.
To reset that you have to power-cycle the unit, or enter the flash count reset command.

Oxyandy, 2018-04-22:
Yeah, I get that, but this was happening shortly after an erase, new flash... 
last time, was suggested bad flash memory too 🥁 
I thought was over on my forum post, scrolled through all pages, '404'
So must be here on Github somewhere ?? 
As I said - I could dig it up, reflash that version as see if is 'repeatable'

Oxyandy, 2018-04-22:
And... the build 'after' this... 
If I tried to edit/update rules, 
they would truncate or even become zero byte rules files, but without any error

TD-er, 2018-04-22:
The builds starting to show this behavior, were they based on core 2.4.0 or 2.3.0?

Oxyandy, 2018-04-23:
@TD-er I can answer this, but it will take 'absolute testing' 
- I am happy to do this & give you every answer to every question you can come up with..
I would then after I have positive 'repeatable impact' documented between firmware revisions, 
really appreciate some-else to 'verify' my findings..
Just tell me what you are looking for...

Oxyandy, 2018-04-23:
ESP_Easy_mega-20180403_normal_ESP8266_1024.bin
As above was very basic config, just enough to get online
Enabled Rules, "Submit"
then NTP submit, Serial log OK?
```
WD   : Uptime 20 ConnectFailures 0 FreeMem 16496
FILE : Saved config.dat
FILE : Saved security.dat
done.
done.
done.
done.
done.
IP blocked: 0.0.0.0 Allowed: 192.168.0.0 - 192.168.0.255
IP blocked: 0.0.0.0 Allowed: 192.168.0.0 - 192.168.0.255

IP blocked: 0.0.0.0 Allowed: 192.168.0.0 - 192.168.0.255
done.
WD   : Uptime 21 ConnectFailures 0 FreeMem 12688
IP blocked: 0.0.0.0 Allowed: 192.168.0.0 - 192.168.0.255
IP blocked: 0.0.0.0 Allowed: 192.168.0.0 - 192.168.0.255
IP blocked: 0.0.0.0 Allowed: 192.168.0.0 - 192.168.0.255
IP blocked: 0.0.0.0 Allowed: 192.168.0.0 - 192.168.0.255
done.
WD   : Uptime 21 ConnectFailures 0 FreeMem 13232
done.
WD   : Uptime 22 ConnectFailures 0 FreeMem 16624
WD   : Uptime 22 ConnectFailures 0 FreeMem 16624
done.

WD   : Uptime 23 ConnectFailures 0 FreeMem 16480
FILE : Saved config.dat
FILE : Saved security.dat
Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 600 min
EVENT: Time#Initialized
EVENT: Clock#Time=Mon,10:37
done.
WD   : Uptime 23 ConnectFailures 0 FreeMem 16480
WD   : Uptime 24 ConnectFailures 0 FreeMem 16624
EVENT: Clock#Time=Mon,10:38
```

Oxyandy, 2018-04-23:
We need a very well thought out list of items to check..
EG.
Power down router, wait 30 seconds then ON, unit connects OK ?
etc
etc

TD-er, 2018-04-23:
The question of this issue is about the rules saving.
If you add a rules file, like linked in this issue, (18xx bytes), save that in the rules text field. Does it result in a saved rules file? And if so, is it 0 Bytes, 1471 Bytes or 18xx bytes?
And what core lib was used in that version? (can be seen via Tools => info)

I guess a binary search is the fastest, instead of testing all versions.

Oxyandy, 2018-04-23:
ESP_Easy_mega-20180410_normal_ESP8266_1024.bin 
Was the last 2.3.0

Oxyandy, 2018-04-23:
ESP_Easy_mega-20180410_normal_ESP8266_1024.bin
Was the last 2.3.0

I'd be happy to rules in these more recent versions, 
but is a little hard when I can not get the ESP to connect or stay connected to Wifi..
and / or access web server's pages.
2 Routers (Different) & Mobile android hotspot, Blanked (FF) devices before flash - nothing really helps..
I don't have enough variety of ESP hardware to test with..
Seem others with different hardware to me get away with starting with a blank device & everything works
Tried, wont work, posted log..

TD-er, 2018-04-29:
I added a commit to log some information on what is sent in the POST request.
The output in the log will be something like this:

`8185127 : Rules : Save rulesSet: 1 currentSet: 1 rules.length(): 1519 Write to file: rules1.txt`

The test I did here had a rules entry of 1734 bytes, thus the rules reported in ESPeasy is less.
Sometimes it is even 0 bytes.

There is something fishy about the handling of the POST.
The content-length of the data sent to the ESP device was 2758 bytes, which is all there should be, including url-encoding.

TD-er, 2018-04-30:
I found a few related issues:

https://github.com/esp8266/Arduino/issues/2437
https://github.com/esp8266/Arduino/issues/1872
https://github.com/esp8266/Arduino/commit/199fe0f16d3544a092f2e3622c40502f5a5117cd

Maybe we could embed some javascript to send POST requests in several parts.
But this may probably also affect restore of settings and some update of plugin settings when the POST exceeds 1 TCP packet.

TD-er, 2018-05-01:
Should be fixed by PR https://github.com/letscontrolit/ESPEasy/pull/1346
It looks like the problem was caused by some issue in LWIP 2.0

TD-er, 2018-05-15:
That's a very specific number.
I will try to make sense of that.
First guess is 4 packets of 536 bytes minus overhead.

Budman1758, 2018-05-15:
I kept testing to narrow it down to that number. So far its rock solid consistent.

Budman1758, 2018-08-13:
I hate to rain on the parade but this issue seems to have reared it's ugly head again. I'm using MEGA-20180809.  Went to the rules and made a single digit edit, hit submit and got a blank page back. Went to flash page and sure enough the rules1 file was 0 bytes.  File was  1676 characters.

Using Chrome the file comes back at zero size.  Using IE the file was truncated at 607 bytes. Several times..Same 607 bytes.

ESP-07 module. ESP_Easy_mega-20180809_test_ESP8266_4096_VCC firmware.

Let me know what other info is needed.

TD-er, 2018-08-13:
So it is the 'official build' ?

Budman1758, 2018-08-13:
@TD-er  Yes, official builds are all I generally use. Still haven't mastered the self-compiled thing...

Budman1758, 2018-10-03:
Not sure of the actual status of this but as of the latest Mega-20181003 this is still an issue.  At the least you may want to remove the "Fixed" label....  :>)

Budman1758, 2018-10-03:
Gotta watch that @Grovkillen dude.  He's a sly one...  :)

giig1967g, 2018-10-07:
> Gotta watch that @Grovkillen dude. He's a sly one... :)
Yeah.
Still issues saving rules. Version 20181007 when pressing Save (on Firefox), it saves a blank file (0 bytes).

TD-er, 2018-10-07:
A blank file?
Is it with core 2.4.1 or 2.4.2?

Domosapiens, 2018-10-07:
Version 20181007 Rules Set 1, when pressing Download to file (on Firefox), saved file is correct in text editor.

TD-er, 2018-10-07:
So when you press save, it is destroyed?

TD-er, 2018-10-07:
Could you also try to change 
`-DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY`
to 
`-DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH`
in PlatformIO.ini ?

TD-er, 2018-10-08:
Just as a reference for my last reply on this:
https://github.com/letscontrolit/ESPEasy/pull/1346

wolverinevn, 2018-10-26:
Any news?
I have same situation:

_Rules : Save rulesSet: 2 currentSet: 1
2590736: Webserver args: 0: 'set' length: 1 1: 'rules' length: 1169
2610393: Rules : Save rulesSet: 2 currentSet: 2 rules.length(): 0 Write to file: rules2.txt
2610394: Webserver args: 0: 'set' length: 1 1: 'rules' length: 0
2612358: WD : Uptime 44 ConnectFailures 0 FreeMem 14104_

Hit submit 1 time then the rule was 0. Happend everytime I paste then click submit. If I cut the same rule to be shorter, it's saved.
For more, after issued occured, when I edit the text box, word count did not work. Maybe browser's or web server's issue? (Safari 12.0 & Chrome  69.0.3497.100 on Mac 10.13.6).
Reseted nodemcu and re-submitted rules (same length), it worked as intentional.

Grovkillen, 2018-11-02:
After we the WD issue "under control" we should take care if this issue.

TD-er, 2019-10-10:
This does no longer seem to be an issue, so I'll close it.
If it still is an issue, please let me know.

PR #2921 (2020-03-23): Rules save fix
Now the rules are uploaded as a (virtual) file instead of form data. The previous way resulted in corrupted rules files if the unit didn't get the full POST body.

Should fix #2896
Should fix: #2819
Should fix: #2729
Should fix: #1296
closed 2019-10-10 by TD-er