letscontrolit/ESPEasy #1300: LCD2004 problem, only work as LCD1602 !

Jufo84, 2018-04-22:
Hi,

I tested release v2.0-20180322 and I see a problem with the correct display on the LCD.
The problem is with LCD2004, when we send remotely via LCD commands: <row>, <col>, <text>. As a result, only the first two lines are displayed, limited to 16 characters (as in LCD1002).

Best Regards,
Robert

giig1967g, 2018-04-22:
You are right. Bug found.
Will fix it.

giig1967g, 2018-04-22:
proposed fix in #1298

TD-er, 2018-04-22:
@giig1967g You added the proposed fix to the PR of the refresh command.
Now it cannot be separated to merge the proposed fix and the new command later.

giig1967g, 2018-04-22:
I noticed that..
But I just committed the changes. 
I am still a novice in github...

TD-er, 2018-04-22:
When you create a pull request, then you're offering a branch to be merged.
If you add commits later to that branch, and push them, they will be added to the existing pull request, since they are on the same branch.
If you want to separate PRs, you have to go back to a certain point (e.g. the remote upstream Mega branch) and create a new branch from there,
Then you'll add the changes to that branch and push that + create a PR from it.

TD-er, 2018-04-22:
I will cherrypick the commit tomorrow.

giig1967g, 2018-04-23:
I understand now.
I can modify the code to fix just the LCD bug. But can do it only tomorrow.

TD-er, 2018-10-13:
I'll close this one now.
If it is still a bug, please re-open.

closed 2018-10-13 by TD-er