letscontrolit/ESPEasy #1301: Impossible to upgrade. Must start with "clean" unit.

Budman1758, 2018-04-22:
This is probably related to issue #1292.
I had tried to do an inplace upgrade on my current test unit and nothing I have tried works. I keep ending up with a boot loop.  I can get any of the most recent builds to work as long as I start with a unit that has been flashed with a blank bin file.  Any form of upgrade attempt ends up in a boot loop.  I am using the last few nightly builds for these tests.

Flash blank bin... Flash any recent build... = Working unit exactly as its supposed to.
Upgrade via OTA on above unit  = boot loop
Upgrade via serial on above unit  = boot loop
Start again with working fresh flashed unit... Upload config ifile to restore settings and devices..  =  boot loop.

Consistent and repeatable.  Here is a sample of the serial output.....

INIT : Booting version: mega-20180422 (ESP82xx Core 2_4_0)
76 : INIT : Warm boot #11
77 : FS   : Mounting...
102 : FS   : Mount successful, used 75802 bytes of 957314
422 : CRC  : program checksum       ...OK
452 : CRC  : SecuritySettings CRC   ...OK

Build changed!
Fix settings with uninitalized data or corrupted by switching between versions
614 : FILE : Saved config.dat
614 : INIT : Free RAM:18384
615 : INIT : I2C
615 : INIT : SPI not enabled
1082 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_0)
1083 : EVENT: System#Wake
1109 : WIFI : Switch on WiFi
1110 : WIFI : Set WiFi to STA
1111 : WIFI : Connecting BronzeDeer attempt #0
1225 : EVENT: System#Boot
1238 : WD   : Uptime 0 ConnectFailures 0 FreeMem 15264

Exception (28):
epc1=0x4020832d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont
          sp: 3fff4fb0 end: 3fff5230 offset: 01a0

>>>stack>>>
           3fff5150:  00000010 00000001 3ffe94d6 40259600
3fff5160:  3fff547c 00000000 3fff6fc2 3fff4208
3fff5170:  3fff6f94 0000002f 3fff4148 40255469
3fff5180:  3ffe94d5 3fff4148 3fff4148 40255469
3fff5190:  3fff6f94 00002088 3fff4148 4025561c
3fff51a0:  3fff6f94 3fff4148 3fff4148 3fff4208
3fff51b0:  3fff6f94 3fff339c 3fff2c94 40214f1e
3fff51c0:  3fffdad0 00000000 3fff22d8 3fff4208
3fff51d0:  3fffdad0 00000000 3fff22d8 4023d4f1
3fff51e0:  3fff6f94 0000003f 0000002f 4020bd28
3fff51f0:  00000000 00000000 3fff20d8 4023f731
3fff5200:  feefeffe feefeffe feefeffe 3fff4208
3fff5210:  3fffdad0 00000000 3fff4202 40257000
3fff5220:  feefeffe feefeffe 3fff4210 40100700
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
   ▒U75 :

INIT : Booting version: mega-20180422 (ESP82xx Core 2_4_0)
76 : INIT : Warm boot #12
77 : FS   : Mounting...
102 : FS   : Mount successful, used 75802 bytes of 957314
422 : CRC  : program checksum       ...OK
452 : CRC  : SecuritySettings CRC   ...OK
558 : INIT : Free RAM:18384
559 : INIT : I2C
559 : INIT : SPI not enabled
1026 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_0)
1026 : EVENT: System#Wake
1053 : WIFI : Switch on WiFi
1053 : WIFI : Set WiFi to STA
1055 : WIFI : Connecting BronzeDeer attempt #0
1168 : EVENT: System#Boot
1181 : WD   : Uptime 0 ConnectFailures 0 FreeMem 15264

Exception (28):
epc1=0x4020832d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont
          sp: 3fff4fb0 end: 3fff5230 offset: 01a0

>>>stack>>>
           3fff5150:  00000010 00000001 3ffe94d6 40259600
3fff5160:  3fff547c 00000000 3fff6fba 3fff4208
3fff5170:  3fff6f8c 0000002f 3fff4148 40255469
3fff5180:  3ffe94d5 3fff4148 3fff4148 40255469
3fff5190:  3fff6f8c 00002090 3fff4148 4025561c
3fff51a0:  3fff6f8c 3fff4148 3fff4148 3fff4208
3fff51b0:  3fff6f8c 3fff339c 3fff2c94 40214f1e
3fff51c0:  3fffdad0 00000000 3fff22d8 3fff4208
3fff51d0:  3fffdad0 00000000 3fff22d8 4023d4f1
3fff51e0:  3fff6f8c 0000003f 0000002f 4020bd28
3fff51f0:  00000000 00000000 3fff20d8 4023f731
3fff5200:  feefeffe feefeffe feefeffe 3fff4208
3fff5210:  3fffdad0 00000000 3fff4202 40257000
3fff5220:  feefeffe feefeffe 3fff4210 40100700
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
   ▒U75 :

Oxyandy, 2018-04-23:
OK, I will take mega-20180422, wipe out a Sonoff Basic 1M unit, flash this firmware & report.

Oxyandy, 2018-04-23:
I got AP show, using webpage at 192.168.4.1 saved SSID & pass
Then it all went wrong from there
```
407038 : AP Mode: Client connected: 3C:46:D8:10:7B:92 Connected devices: 1
407042 : Webserver: start
421423 : WD   : Uptime 7 ConnectFailures 0 FreeMem 14896
440034 : WIFI : Set WiFi to AP
451423 : WD   : Uptime 8 ConnectFailures 0 FreeMem 8264
459006 : WIFI : Set WiFi to AP+STA
459113 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
459114 : Webserver: stop
459115 : WIFI : Connecting MAD_MOB attempt #0
459116 : WIFI : Credentials Changed, retry connection. SSID: MAD_MOB
459149 : WIFI : Connecting MAD_MOB attempt #0
462626 : WIFI : Connected! AP: MAD_MOB (18:90:D8:AC:0F:D8) Ch: 7 Duration: 3476 ms

463279 : WIFI : DHCP IP: 192.168.0.103 (ESP-Easy-0) GW: 192.168.0.254 SN: 255.255.255.0   duration: 653 ms
463280 : Webserver: start
463805 : FILE : Saved config.dat
464092 : FILE : Saved security.dat
464099 : Webserver: stop
464099 : WIFI : Connecting MAD_MOB attempt #0
471718 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 7619 ms
471723 : WIFI : AP Mode Disabled for SSID: ESP_Easy_0
471724 : WIFI : Set WiFi to STA
471725 : WIFI : Connecting MAD_MOB attempt #1
471739 : AP Mode: Client disconnected: 3C:46:D8:10:7B:92 Connected devices: 0
474799 : WIFI : Connected! AP: MAD_MOB (18:90:D8:AC:0F:D8) Ch: 7 Duration: 3072 ms
475455 : WIFI : DHCP IP: 192.168.0.103 (ESP-Easy-0) GW: 192.168.0.254 SN: 255.255.255.0   duration: 655 ms
475456 : Webserver: start
481423 : WD   : Uptime 8 ConnectFailures 0 FreeMem 15584
483914 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 9105 ms
483915 : Webserver: stop
483915 : WIFI : Connecting MAD_MOB attempt #0```

Budman1758, 2018-04-23:
Brand new never been flashed by me generic ESP-01 module.
Flashed with normal 1024  mega 20180422
Unit started AP mode right away.
Laptop connect to the AP. My browser went to the wifi connect page (automatically)
Entered my wifi creds. Would not connect to the router and started back in AP Mode.
Entered wifi creds using serial commands.
Immediately connected to router. Up and running fine since.  No wifi problems.
Maybe this is related to the rules save issue. #1296

Oxyandy, 2018-04-23:
Yes, I will re-flash again (they go up/down haha) and then do as you have done..
wifissid SID wifikey bla & save, see if it helps.. 
but the errors '(200) Beacon timeout' & '(201) No AP found'
are likely something else, still I'll do it

Grovkillen, 2018-04-23:
I'm thinking about a external "restore settings" app that could do the setup from older versions to newer versions without the need to upload a config.dat file. I don't know if that's an acceptable solution though?

giig1967g, 2018-04-23:
Wow, that will be an awesome feature!

Budman1758, 2018-04-23:
Sounds better than having to manually reconfigure the unit from scratch.

Grovkillen, 2018-04-23:
I guess it's not an bad idea then ;) I'm a Windows user myself but maybe someone could point me in the right direction on deploying a python version?

I will for myself create such a tool for Windows though...

Budman1758, 2018-04-23:
Why is it that lately we can't upgrade from one daily build to another? All I get is boot loops.

Grovkillen, 2018-04-23:
I don't have that problem with my test unit...?

Budman1758, 2018-04-23:
Thats what this topic is all about.

Grovkillen, 2018-04-23:
Yes, sorry :sweat_drops:

Budman1758, 2018-04-23:
It's been going on for a while now.

Grovkillen, 2018-04-23:
I do not get a boot loop with OTA using NORMAL bin file. Do you need something in TESTING? Not a solution but TESTING shouldn't be in the scope of the release road map of v2.1

```
{$ l£ƒ| ─dÓ|♥♀♦♀ä♀lõ♀#|çâ♥ý←█søbä♀c─‗'o▀$gn£Òý♦c∟pýÃds$sdp¹gÓ►♥♦♦é♦l♦─£♦♀♦c♀nÒ|♥
$ý♦♀Äcî¹o'þ läçd`♥ÿ‼←gn♀d`♥♥orÄøÆn♀♦c♀£d♫{Æøo♦♦b♦£☼l£âõ█‼cô←←d`♥³ân£♥▒ ¬U66 :

INIT : Booting version: mega-20180423 (ESP82xx Core 2_4_0)
66 : INIT : Warm boot #3
68 : FS   : Mounting...
92 : FS   : Mount successful, used 76304 bytes of 957314
398 : CRC  : program checksum       ...OK
431 : CRC  : SecuritySettings CRC   ...OK
511 : INIT : Free RAM:20448
512 : INIT : I2C
512 : INIT : SPI not enabled
530 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_0)
531 : WIFI : Switch on WiFi
531 : WIFI : Set WiFi to STA
654 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19088
768 : SYS  : 0.00
1105 : WIFI : Connecting SA_AP attempt #0
1119 : IMPT : MQTT 037 Intentional reconnect
1768 : SYS  : 31.00
2221 : IMPT : MQTT 037 Intentional reconnect
2783 : IMPT : MQTT 037 Intentional reconnect
4140 : IMPT : MQTT 037 Intentional reconnect
4492 : WIFI : Connected! AP: SA_AP (80:2A:A8:5A:1B:08) Ch: 6 Duration: 3056 ms
5139 : IMPT : MQTT 037 Intentional reconnect
5492 : WIFI : DHCP IP: 192.168.0.192 (Test-20) GW: 192.168.1.1 SN: 255.255.252.0
   duration: 1065 ms
5505 : Webserver: start
5602 : MQTT : Intentional reconnect
5768 : MQTT : Connected to broker with client ID: ESPClient_60:01:94:0A:FD:71
5771 : Subscribed to: Test/#
6175 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minST
D time start: 2018-10-28 03:00:00 offset: 60 min
6566 : IMPT : Connected to MQTT broker with Client ID=Test-Import
6570 : IMPT : [sniff#cmdMQTT] subscribed to cmdMQTT
6882 : IMPT : [sniff#cmdMQTT] : 2.00
28303 : FILE : Saved config.dat
28350 : EVENT: Time#Initialized
30657 : WD   : Uptime 1 ConnectFailures 0 FreeMem 14936
47509 : EVENT: Clock#Time=Mon,10:31
60657 : WD   : Uptime 1 ConnectFailures 0 FreeMem 14936
61508 : SYS  : 1.00
61510 : EVENT: Uptime#min=1.00
90734 : WD   : Uptime 2 ConnectFailures 0 FreeMem 11496
107747 : EVENT: Clock#Time=Mon,10:32
120740 : WD   : Uptime 2 ConnectFailures 0 FreeMem 14936
121746 : SYS  : 2.00
121748 : EVENT: Uptime#min=2.00
150739 : WD   : Uptime 3 ConnectFailures 0 FreeMem 14936
167753 : EVENT: Clock#Time=Mon,10:33
180739 : WD   : Uptime 3 ConnectFailures 0 FreeMem 14936
181752 : SYS  : 3.00
181754 : EVENT: Uptime#min=3.00
210739 : WD   : Uptime 4 ConnectFailures 0 FreeMem 14936
227761 : EVENT: Clock#Time=Mon,10:34
240739 : WD   : Uptime 4 ConnectFailures 0 FreeMem 14936
241761 : SYS  : 4.00
241763 : EVENT: Uptime#min=4.00
270739 : WD   : Uptime 5 ConnectFailures 0 FreeMem 14936
287770 : EVENT: Clock#Time=Mon,10:35
300739 : WD   : Uptime 5 ConnectFailures 0 FreeMem 14936
301772 : SYS  : 5.00
301774 : EVENT: Uptime#min=5.00
330739 : WD   : Uptime 6 ConnectFailures 0 FreeMem 14936
347781 : EVENT: Clock#Time=Mon,10:36
360739 : WD   : Uptime 6 ConnectFailures 0 FreeMem 14936
361782 : SYS  : 6.00
[... 181 lines cut ...]
14829 : IMPT : MQTT 037 Intentional reconnect
15624 : IMPT : MQTT 037 Intentional reconnect
16855 : WIFI : Connected! AP: SA_AP (80:2A:A8:5A:1B:08) Ch: 6 Duration: 3063 ms
16856 : EVENT: WiFi#ChangedAccesspoint
16899 : IMPT : MQTT 037 Intentional reconnect
17644 : WIFI : DHCP IP: 192.168.0.192 (Test-20) GW: 192.168.1.1 SN: 255.255.252.
0   duration: 867 ms
17655 : EVENT: WiFi#Connected
17669 : Webserver: start
17755 : MQTT : Intentional reconnect
18015 : MQTT : Connected to broker with client ID: ESPClient_60:01:94:0A:FD:71
18017 : Subscribed to: Test/#
18019 : EVENT: MQTT#Connected
18562 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minS
TD time start: 2018-10-28 03:00:00 offset: 60 min
18563 : EVENT: Time#Initialized
18582 : EVENT: Clock#Time=Mon,10:41
18967 : IMPT : Connected to MQTT broker with Client ID=Test-Import
18968 : EVENT: MQTTimport#Connected
18984 : IMPT : [sniff#cmdMQTT] subscribed to cmdMQTT
19296 : IMPT : [sniff#cmdMQTT] : 2.00
19297 : EVENT: sniff#cmdMQTT=2.00
19310 : ACT  : Publish Test/Time,10:41:10
43484 : WD   : Uptime 1 ConnectFailures 0 FreeMem 15096
```

Budman1758, 2018-04-23:
After I attempt to get some sleep I will get back to you on that...  :>)

Budman1758, 2018-04-23:
@Grovkillen I have a test unit running mega 20180423 normal 1024 right now. ESP-01 generic module.  Will try an upgrade via serial with the next available nightly build. Fingers crossed... :)

Budman1758, 2018-04-24:
Completely successful upgrade from 20180423 to 20180424.  Normal 1024 build.  Just like its supposed to do!!

@Grovkillen  I use the test build mainly for the VCC feature.  Be nice if there was one in the normal builds.

Grovkillen, 2018-04-24:
Nice, we'll need to make a normal VCC then.  :+1: @TD-er you agree?

TD-er, 2018-04-24:
Yep, that one makes sense to have

Budman1758, 2018-04-25:
Hard to tell if this is still an issue. Close it?

Oxyandy, 2018-04-26:
As an open issue, I think it's a resource of information for others, so maybe still useful ?
Using different hardware to you, the issues we have are quite different, I haven't seen the boot loops
and guessing you haven't seen the '(200) Beacon timeout' + '(201) No AP found' & other wifi connectivity errors.

closed 2023-09-04 by TD-er