letscontrolit/ESPEasy #1303: MPR121 Plugin is causing endless boot loop

happytm, 2018-04-23:
I have been using espeasy for more than 2 years now.I noticed odd event I am experiencing. I am trying to use MPR121 keypad plugin using latest test_4096 from 04/22/2018.If I run software without using mpr121 everything works fine but as soon as I activate MPR121 in device settings page firmware go in endless reboot loop like everybody is experiencing lately.I have tried this several times with different esp8266 and different firmware from last 2 months but every time MPR121 breaks the firmware.

TD-er, 2018-04-23:
You could compare that plugin to other plugins to see if it returns `success` at the same cases as other plugins do.
If I remember correctly there was a similar issue with the "Kamstrup" plugin a few months ago.

happytm, 2018-04-23:
TD-er,

I am sorry I did not understand how I can compare to other plugins for success. Here is my setup on -

rotary encoder connected - works fine
oledssd1306 connected - works fine
pme plugin - works fine
ir transmitter - not tried yet (waiting for parts)
mpr121 - fails when activated in devices page even without connecting hardware to I2C bus

I am trying to make  universal remote input device which can be used in different applications using above parts. MPR121 gives me 12 buttons for remote control and rotary encoder should give up and down control for each button separately which will be seen on oled. I am going to use rules engine with dummy sensor plugin to do it. Here are rules i will use -

//4 CHANNEL REMOTE CONTROL WITH MPR121 TOUCH SWITCHES , ROTARY ENCODER & WEB EVENT COMMAND FROM BROWSER

on  mpr121#switch=1.00 do
    
    TaskValueSet,6,1,[rotary#counter]*100 //scaling (i have my scaling in device setting for encoder as 1 to 10 to make roughly 10 steps dimmer)
    extpwm,13,[dummy#d1] // pwm level value is taken from dummy sensor value
    oled,6,4,FAN1 = [dummy#d1]  // pwm level value is taken from dummy sensor value

endon

on mpr121#switch=2.00 do
   
    TaskValueSet,6,2,[rotary#counter]*100 
    extpwm,13,[dummy#d2] 
   oled,6,4,FAN2 = [dummy#d2] 

endon

on mpr121#switch=3.00 do

    TaskValueSet,6,3,[rotary#counter]*100 
    extpwm,13,[dummy#d3] 
    oled,6,4,FAN3 = [dummy#d3] 

endon

on mpr121#switch=4.00 do

    TaskValueSet,6,4,[rotary#counter]*100 
    extpwm,13,[dummy#d4] 
   oled,6,4,FAN4 = [dummy#d4]

endon

on dimmer1 do // when http://192.168.0.73/control?cmd=event,dimmer1=100 command issued from browser
TaskValueSet 6,1,%eventvalue% // store pwm level value of 0 to 1023 from above command into dummy sensor 
extpwm,13,[dummy#d1] //pwm level value is taken from dummy sensor value
oled,6,4,FAN1 = [dummy#d1] // pwm level value is taken from dummy sensor value
endon

on dimmer2 do
TaskValueSet 6,2,%eventvalue%  
extpwm,13,[dummy#d2]
oled,6,4,FAN2 = [dummy#d2]
endon

on dimmer3 do 
TaskValueSet 6,3,%eventvalue%
extpwm,13,[dummy#d3]
oled,6,4,FAN3 = [dummy#d3]  
endon

on dimmer4 do 
TaskValueSet 6,4,%eventvalue%
extpwm,13,[dummy#d4]
oled,6,4,FAN4 = [dummy#d4] 
endon

happytm, 2018-04-23:
Just to try fresh install on new esp8266 device I loaded today's test firmware 4096 without any other device activated and no rules defined and tried to activate only MPR121 plugin on device page. it still breaks firmware. following is log - 

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
⸮U60 : 

INIT : Booting version: mega-20180423 (ESP82xx Core 2_4_0)
60 : INIT : Warm boot #31
62 : FS   : Mounting...
87 : FS   : Mount successful, used 77559 bytes of 957314
420 : CRC  : program checksum       ...OK
451 : CRC  : SecuritySettings CRC   ...OK 
553 : INIT : Free RAM:18400
554 : INIT : I2C
554 : INIT : SPI not enabled

Soft WDT reset

ctx: cont 
sp: 3fff4ea0 end: 3fff5220 offset: 01b0

>>>stack>>>
3fff5050:  3fff546c 00000158 00000158 00000001  
3fff5060:  3fff3f16 00000180 00000001 40201938  
3fff5070:  3fff3f16 0000005b 00000004 40201b3e  
3fff5080:  00000001 00000000 00000004 40201a74  
3fff5090:  0000005b 00028737 3fff5280 00000000  
3fff50a0:  0000005d 3fff632c 00000001 4024cbae  
3fff50b0:  0000005d 3fff632c 3fff3ee0 4024cbf7  
3fff50c0:  40201d52 00000001 3fff3ee0 402522c7  
3fff50d0:  0000005b 0000005b 3fff632c 40252317  
3fff50e0:  3fff20d4 3fff632c 3fff17e0 4023998d  
3fff50f0:  00000000 00000000 00000000 40255e98  
3fff5100:  3ffe8ed4 00000001 40291e56 4020ad42  
3fff5110:  00000000 00000000 00000000 402085b8  
3fff5120:  3fff20d4 00000002 3fff5130 40213420 <
3fff5130:  00000000 3fff0000 00000000 4025530a  
3fff5140:  00000000 00000000 00000000 00000000  
3fff5150:  00000000 00000200 3fff65bc 0000000f  
3fff5160:  00000000 3fff5d14 0000000f 00000000  
3fff5170:  3fff6fcc 0000000f 00000000 00000000  
3fff5180:  3fff410c 3fff3384 00000000 0000003a  
3fff5190:  3fff3384 00000000 00000000 00000000  
3fff51a0:  00000000 00000000 402a0d30 3fff2320  
3fff51b0:  3fff410c 0000000c 3fff20d4 40213915  
3fff51c0:  3fff410c 00000000 3fff3384 4023e158  
3fff51d0:  00000000 00000000 00000000 40258104  
3fff51e0:  3fff6f84 0000003f 00000015 feefeffe  
3fff51f0:  feefeffe feefeffe feefeffe 3fff41ec  
3fff5200:  3fffdad0 00000000 3fff41e6 40256eb4  
3fff5210:  feefeffe feefeffe 3fff4200 40100700  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
s⸮U61 : 

INIT : Booting version: mega-20180423 (ESP82xx Core 2_4_0)
61 : INIT : Warm boot #32
62 : FS   : Mounting...
87 : FS   : Mount successful, used 77559 bytes of 957314
420 : CRC  : program checksum       ...OK
452 : CRC  : SecuritySettings CRC   ...OK 
554 : INIT : Free RAM:18400
554 : INIT : I2C
555 : INIT : SPI not enabled

Soft WDT reset

ctx: cont 
sp: 3fff4ea0 end: 3fff5220 offset: 01b0

happytm, 2018-04-27:
Just found out it was my mistake. I was  trying MPR121 without plugging it in to esp hardware.Sorry for the confusion.Once I plugged in MPR121 to I2C bus there is no problem anymore.I am using following ruleset for my setup for four dimmer and works as intended -

// 4 channel dimmer control

on key#press=17.00 do
TaskValueSet,12,1,[dummy#d1]+127
if [dummy#d1]>1000 
TaskValueSet,12,1,1000
endif
EXTPWM,6,[dummy#d1] 
oled,2,4,FAN1=[dummy#d1]
endon

on key#press=272.00 do
TaskValueSet,12,1,[dummy#d1]-127
if [dummy#d1]<0 
TaskValueSet,12,1,0
endif
EXTPWM,6,[dummy#d1] 
oled,2,4,FAN1=[dummy#d1]
endon

on key#press=34.00 do
 TaskValueSet,12,2,[dummy#d2]+127
if [dummy#d2]>1000 
TaskValueSet,12,2,1000
endif
EXTPWM,7,[dummy#d2] 
oled,2,4,FAN2=[dummy#d2]
endon

on key#press=544.00 do
 TaskValueSet,12,2,[dummy#d2]-127
if [dummy#d2]<0 
TaskValueSet,12,2,0
endif
EXTPWM,7,[dummy#d2] 
oled,2,4,FAN2=[dummy#d2]
endon

on key#press=68.00 do
TaskValueSet,12,3,[dummy#d3]+127
if [dummy#d3]>1000 
TaskValueSet,12,3,1000
endif
EXTPWM,8,[dummy#d3] 
oled,2,4,FAN3=[dummy#d3]
endon

on key#press=1088.00 do
TaskValueSet,12,3,[dummy#d3]-127
if [dummy#d3]<0 
TaskValueSet,12,3,0
endif
EXTPWM,8,[dummy#d3] 
oled,2,4,FAN3=[dummy#d3]
endon

on key#press=136.00 do
TaskValueSet,12,4,[dummy#d4]+127
if [dummy#d4]>1000 
TaskValueSet,12,4,1000
endif
EXTPWM,9,[dummy#d4] 
oled,2,4,FAN4=[dummy#d4]
endon

on key#press=2176.00 do
TaskValueSet,12,4,[dummy#d4]-127
if [dummy#d4]<0 
TaskValueSet,12,4,0
endif
EXTPWM,9,[dummy#d4] 
oled,2,4,FAN4=[dummy#d4]
endon

on dimmer1 do // when http://192.168.0.73/control?cmd=event,dimmer1=100 command issued from browser
TaskValueSet 12,1,%eventvalue% // store pwm level value of 0 to 1023 from above command into dummy sensor 
oled,2,4,FAN1=[dummy#d1]
endon

on dimmer2 do
TaskValueSet 12,2,%eventvalue%
oled,2,4,FAN2=[dummy#d2]
endon

on dimmer3 do 
TaskValueSet 12,3,%eventvalue%
oled,2,4,FAN3=[dummy#d3]
endon

on dimmer4 do 
TaskValueSet 12,4,%eventvalue% 
oled,2,4,FAN4=[dummy#d4]
endon

melwinek, 2018-10-26:
MPR121 inputs can also work as outputs. Can you include this in the plugin?