letscontrolit/ESPEasy #1306: [Generic HTTP Advanced controller] Newline placeholders are being replaced incorrectly

ziolek, 2018-04-23:
### Summarize of the problem/feature request

Placeholders `%CR%` and `%LF%` are not being replaced according to [the documentation](https://www.letscontrolit.com/wiki/index.php/EasyProtocols#InfluxDB_HTTP_Api_.28via_Generic_HTTP_Advanced.29).

### Expected behavior

I'd expect that  `%CR%` and `%LF%` are replaced with `\r` and `\n` as for example InfluxDB accept only `\n` as line delimiters while [writing multiple points](http://docs.influxdata.com/influxdb/v1.5/guides/writing_data/#writing-multiple-points).

### Actual behavior

Placeholders  are being replaced as follows:
* `%CR%` into `%0d`
* `%LF%` into `%0a`

It looks like they are being treated with `StreamingListener::UrlEncode()`. 

### Steps to reproduce

1. Put a template in HTTP Body of the controller containing %CR% or %LF%. Mine:
```
uptime,thing=%sysname% value=%uptime%%1%%LF%%vname1%,device=%tskname%,thing=%sysname% value=%val1%%/1%
```
2. Set InfluxDB as a backend (or any HTTP catcher, even just a netcat).
3. Check the serial console out and look for HTTP resonse from InfluxDB (or check that catched data by netcat).
You could see that InfluxDB can't parse `%0a` character.
I got the response:
```
{"error":"unable to parse 'uptime,thing=ziolek-forest value=1%0amoisture,device=soil,thing=ziolek-forest value=2.54': invalid number"}
```

### System configuration

Hardware: NodeMCU (ESP8266)

ESP Easy version:  ESP_Easy_mega-20180423_test_ESP8266_4096

Naimadnap, 2018-04-24:
Could you have some workaround for that?

TD-er, 2018-04-24:
This is in the body of the message, so indeed the UrlEncode should not be called here.

JanSpacil, 2018-05-01:
Hallo, same problem here .. any solution?

giig1967g, 2018-05-02:
In order to keep backward compatibility, what if we create 2 more system variables as follows:

%R%  -> "\r"
%N%  -> "\n"

JanSpacil, 2018-05-08:
Hallo, any news? :]
http://www.letscontrolit.com/forum/viewtopic.php?f=6&t=5298

JanSpacil, 2018-05-17:
Once again, humble request .. can you please fix this? I'd like to send multiple values in one POST (temp, humidity, pressure), but I can't .. If I define three controllers (each for one value), system goes to be very unstable ..

InxluxDB 1.52 (raspberry pi, openhabian os)

st3fan87, 2018-07-26:
On dev 7 HTTP advanced it was possible to send multiple values. Is this issue already fixed? Really want to send multiple values within the http body

st3fan87, 2018-07-27:
Or does someone knows a quick fix? @ziolek @TD-er

TD-er, 2018-07-30:
I have just been looking at the code and found this:
```C++
  repl(F("%CR%"), F("\r"), s, useURLencode);
  repl(F("%LF%"), F("\n"), s, useURLencode);
```

This means `%CR%` will be replaced into "\r", which is used in C++ as an alias for ASCII code 13.
I will add [this](https://github.com/letscontrolit/ESPEasy/issues/1306#issuecomment-385889927) suggestion from @giig1967g 

- %R% -> "\\r"
- %N% -> "\\n"

oyla12, 2019-01-11:
I think there is a bug again, or I'm really to dumb to use it. Using the current mega-20190110 test version on my nodeMCU (ESP_Easy_mega-20190110_test_ESP8266_4096.bin) with this 

HTTP-Header: `Content-Type: application/x-www-form-urlencoded`
HTTP-Body: `temp value=%val1%%N%humidity value=%val2%`

The result I get using `nc -l -p 8086 | hexdump -C` as pseudo influxdb http server is the following. The left side is hex, the right side is the ASCII representation. The Pipe is the delimiter between them. 

`3d 31 38 2e 36 30 25 35  63 6e 68 75 6d 69 64 69 |=18.60%5cnhumidi|`

I tested the old, not working way with %LF% and can reproduce the old "problem". Just to be sure there is no encoding-problem, i know it's for backward-compatibility:

`3d 31 39 2e 30 35 25 30  61 68 75 6d 69 64 69 74  |=19.05%0ahumidit|` 

So why do I get `%5cn` as result for the `%N%`? Is this really the expected behavior?

luciust, 2019-02-10:
Same situation here:
ESPEasy 2018-01-17 dev - influxdB multiple values with %LF% - works
ESPEasy 2019-02-02 dev - influxdB multiple values with %LF% nor %N% - doesn't work

DrakenGray, 2020-09-05:
Bringing this back up.  I have been trying to find a workaround for this and simply cannot get two strings to work without %LF% or %N% making a proper newline.  Neither command is getting properly replaced in the body resulting in parsing failure.  I have confirmed that the installation works when posting a single value, but fails with any implementation of newline and multiple values.  InfluxDB fails when using \r returns.  
ESP_Easy_mega_20200829_test_ESP8266_4M1M_VCC on nodeMCU

Body code used
%vname1%,device=%tskname% value=%val1% %unixtime%%LF%%vname2%,device=%tskname% value=%val2% %unixtime%

Captured output
pm1.0,device=Partic value=1 1599332800%0apm2.5,device=Partic value=2 1599332800

Results in a bad request from Influx as its inserting a %0a instead of a newline, causing influx to read the entire string as a single entry.

%N% returns %5cn , %LF% returns %0a.  HTTP was captured on influxdb receiving server via wireshark.

TD-er, 2020-09-06:
So it seems like the order of replacements may be an issue here?

DrakenGray, 2020-09-06:
This ones a little out of my realm, but I'm not sure its a order of replacement issue.  %0a in ASCII should be processed as a new line command in the output and break the value strings into two lines.  Unfortunately using the normal carriage return is not acceptable for influxdb, and will throw a timestamp error as it tries to process the \r at the end of the string.  Attempting to directly insert a \n is just accepted as body text and not processed.  The root problem is the output processor is not producing a new line from the %0a variable that is being inserted using %N%.  So the ascii replacement part is working correctly, processing the replacement into a new line is not.  I do not understand the mechanics of how this is processed to help more, sorry.

Included two pictures of a good output by another device, and the failed output.  
[image]

no line break causes a processing error.  This was set using %N% between strings.
[image]

DrakenGray, 2020-09-06:
I did a bit more digging.  My IoTaWatt installation, the hex for newline shows up as 0a and creates a newline as intended.  While %LF% is inserting the code that should be in the hex (0a), but its actually printed out as body text instead.  Everything else in the hex for a newline shows up as '0a' in the header, its only in the body that it does not work as expected.  

Body code is not accessible on Iotawatt
Good newline hex example
00a0   30 37 35 **0a** 52 65 66 72 69 67 65 72 61 74 6f 72  | 075 **.** Refrigerator
HTTP output
Oven,iota=home value=0.00 1599414065
Refrigerator,iota=home value=120.72 1599414065

Body code
%vname1%,device=%tskname% value=%val1% %unixtime%%LF%%vname2%,device=%tskname% value=%val2% %unixtime%
Failed newline hex
0170   31 35 39 39 34 31 34 32 36 34 **25 30 61** 48 75 6d   1599414264 **%0a** Hum
HTTP output
Temperature,device=DHT22 value=75.02 1599414264%0aHumidity,device=DHT22 value=55.20 1599414264

These appear to be the same findings oyla12 had back in 2019.

Thanks!

TD-er, 2020-09-07:
It looks like the body of the POST is URL-encoded, which should not be done.

See the `true` in this line:
https://github.com/letscontrolit/ESPEasy/blob/3b8b34586eb958148e8427795555a324c3d39734/src/_C011.ino#L327

TD-er, 2020-09-08:
I made a test build with only the URL-encode flag (which is actually a percent encode) set to false.

- [ESPEasy_ESP32_mega-20200829-29-PR_3250.zip](https://www.dropbox.com/s/4ay229fgbr0x38c/ESPEasy_ESP32_mega-20200829-29-PR_3250.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200829-29-PR_3250.zip](https://www.dropbox.com/s/tiuuzla0a07ug2k/ESPEasy_ESP82xx_mega-20200829-29-PR_3250.zip?dl=0)

Can you test if this is a fix for this specific issue?
If it is, then I will add a flag to the controller configuration whether the content of body and/or header should be percent-encoded.
I do think, the way it is done now in this test build is not according to standard, as the `Content-Type: application/x-www-form-urlencoded` suggests the content should be percent-encoded.
But sending ASCII code 0x0A is still not considered "binary data", so for this purpose it is not an issue.
It will be a problem when you try to replace `{D}` in the body, as that will be translated into a degree symbol which should be percent-encoded.

TL;DR this is not the final fix, but I am not sure yet how it should be fixed.

DrakenGray, 2020-09-08:
Unfortunately it does not appear to have changed the output in a usable manner, it still shows up without the proper newline in the capture.  Problem I'm having now is it is sending a double header resulting in influx not reading the payload at all.  I ran a full reset and flash, using exactly the same config as before.  Going to flash back and see if the double header issue remains and edit reply.  Attempted to insert %N%, %LF%, and for kicks, tried using /n and the output has remained the same as before otherwise.  

`
POST /write?precision=s&db=sensor_data HTTP/1.1
Host: 192.168.0.163:8086
Authorization: Basic redacted 
Accept: */*;q=0.1
User-Agent: ESP Easy/20110/Sep  8 2020 18:02:49
Connection: close
Content-Type: application/x-www-form-urlencoded Content-Length: 94

POST /write?precision=s&db=sensor_data HTTP/1.1
Host: 192.168.0.163:8086
Authorization: Basic redacted 
Accept: */*;q=0.1
User-Agent: ESP Easy/20110/Sep  8 2020 18:02:49
Connection: close
Content-Type: application/x-www-form-urlencoded Content-Length: 94

Temperature,device=DHT22 value=77.36 1599605568%0aHumidity,device=DHT22 value=51.30 1599605568
`

TD-er, 2020-09-09:
OK, I will take a further look at it, so my assumption about the URL encode part was incorrect.
Also good to know about the strange issue with the header.

TD-er, 2020-09-09:
@DrakenGray 

> Fun fact, if you do not hit enter and make a new line at the end of the HTTP header field, the output will produce a identical double header! If you end the line square at the end, the content length field will append to the end of the header without space or return. Took a bit to figure that one out...

Was that already the behavior on older builds?

DrakenGray, 2020-09-09:
The header issue was present before you made the change.  Apparently I had put a return in unintentionally before, and didn't when I re-entered the information on a blank ESP.  Drove me up a wall for 2 hours trying to figure out why I had double headers suddenly lol.

TD-er, 2020-09-09:
I will now make a new test build, which does use the ESP8266HTTPClient library instead of our own code.
So instead of re-inventing the wheel and adding a lot of our own code, we can also use a library which has -hopefully- had more testing.

TD-er, 2020-09-09:
I created a new test build for it.
Sadly the file name of the ZIP has remained the same, but it should be a new build.
So make sure you have bin files of today.

- [ESPEasy_ESP32_mega-20200829-33-PR_3250.zip](https://www.dropbox.com/s/vcdaaqfp3w0eghv/ESPEasy_ESP32_mega-20200829-33-PR_3250.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200829-33-PR_3250.zip](https://www.dropbox.com/s/gdkjkba50dwhszp/ESPEasy_ESP82xx_mega-20200829-33-PR_3250.zip?dl=0)

DrakenGray, 2020-09-10:
Please dont shoot the messanger, but it seems nothings changed.  The return at the end of the header is still required, and the processing of the variables remains the same.  
`Temperature,device=DHT22 value=77.54 1599697643%5cnHumidity,device=DHT22 value=51.80 1599697643`
`Build Time:⋄ | Sep  9 2020 15:32:37`
`Binary Filename:⋄ | ESP_Easy_mega_20200909_test_ESP8266_4M1M_VCC`

A full dump of series of operations from log dev
180426: POST /write?precision=s&db=sensor_data HTTP/1.1^^Host: 192.168.0.xxx:8086^^Authorization: Basic xxxredact ^^Accept:
180426: HTTP before parsing:
180426: %vname1%,device=%tskname% value=%val1% %unixtime%%N%%vname2%,device=%tskname% value=%val2% %unixtime%%N%%vname3%,device=%tsknam
180427: HTTP after parsing:
180427: %vname1%,device=%tskname% value=%val1% %unixtime%%N%%vname2%,device=%tskname% value=%val2% %unixtime%%N%%vname3%,device=%tsknam
180433: No sensor value for TaskIndex: 0 varnumber: 2
180434: HTTP after replacements:
180434: Temperature,device=DHT22 value=77.72 1599698095%5cnHumidity,device=DHT22 value=51.60 1599698095%5cn,device=DHT22 value= 1599698

TD-er, 2020-09-10:
OK, that's good to know, it at least is still working as it was and not more is broken 👍 
So feel safe... for now ;)

The `\n` is still not present?

DrakenGray, 2020-09-11:
Still no go on the newline.  Tried all the varieties without a useful result.   Accepts a single value string fine still, just no go on multivalue without the \n style newline.

TD-er, 2020-09-11:
OK, I will continue working on that today.
What I find quite odd, is that you still need to have a line end for the header, as I now use a library for the http traffic for this controller which has a function to add a header.
This function needs the header to be split into 2 strings (key and value), so if this still requires a newline after the value, then it definitely is something others will also run into when using the library.

So I will also have a look at that part of the library code to see if that one also needs some PR.

TD-er, 2020-09-11:
OK, added a few more changes and did some tests using the online Postman service to see what it does receive.

Tried to send the body with newlines and also %LF% and it looks exactly the same on the Postman service.
Also added support for multiline headers, so each will be parsed and added as separate header.

URI will now also be parsed for variables to replace, so you can now also use system variables or event values in the URL (useful for GET URLs)

Test string used (taken from the influxdata page)
```
cpu_load_short,host=server02 value=0.67%LF%cpu_load_short,host=server02,region=us-west value=0.55 1422568543702900257%LF%cpu_load_short,direction=in,host=server01,region=us-west value=2.0 1422568543702900257
```
and as multi-line header:
```
Content-Type: application/x-www-form-urlencoded
X-Powered-By: ESPEasy
```

(part) of the output on the Postman page:
[image]

I left out the used hostname, or else it could be spammed and using all my free calls.

TD-er, 2020-09-11:
New test builds:

- [ESPEasy_ESP32_mega-20200829-34-PR_3250.zip](https://www.dropbox.com/s/lv75n5zh6pobu7u/ESPEasy_ESP32_mega-20200829-34-PR_3250.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200829-34-PR_3250.zip](https://www.dropbox.com/s/0h0az6k6wsp6z0z/ESPEasy_ESP82xx_mega-20200829-34-PR_3250.zip?dl=0)

DrakenGray, 2020-09-11:
Reflashed, and literally copied your example into espeasy and this is the results.  Color me confused.  
Flashed with 0911 copy, rebooted the device, didnt work with my string so I copied yours, still was not working correctly.  I am capturing the packets incoming to the influx server before it gets processed so that rules out Influx.  The double header, and run-on with content length feild issue is still present when not putting a return at end of header string as well.  For all intents, it appears your changes are not making it into the bin I am flashing?  

```
POST /write?precision=s&db=sensor_data HTTP/1.1
Host: 192.168.0.xxx:8086
Authorization: Basic xxxredact 
Accept: */*;q=0.1
User-Agent: ESP Easy/20110/Sep 11 2020 12:32:16
Connection: close
Content-Type: application/x-www-form-urlencoded
X-Powered-By: ESPEasyContent-Length: 205

POST /write?precision=s&db=sensor_data HTTP/1.1
Host: 192.168.0.163:8086
Authorization: Basic ZXNwZWFzeTplc3BlYXN5 
Accept: */*;q=0.1
User-Agent: ESP Easy/20110/Sep 11 2020 12:32:16
Connection: close
Content-Type: application/x-www-form-urlencoded
X-Powered-By: ESPEasyContent-Length: 205

cpu_load_short,host=server02 value=0.67%0acpu_load_short,host=server02,region=us-west value=0.55 1422568543702900257%0acpu_load_short,direction=in,host=server01,region=us-west value=2.0 1422568543702900257
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: 5cfbbd0e-f422-11ea-b771-00155d009e01
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.8.0
X-Request-Id: 5cfbbd0e-f422-11ea-b771-00155d009e01
Date: Fri, 11 Sep 2020 11:31:41 GMT
Connection: close

```

TD-er, 2020-09-11:
What kind of browser do you use?

TD-er, 2020-09-11:
Just an idea...
Can you write something recognizable in both fields, to make it easier to find and then have a look with a hex viewer to that settings file?

This part I find very very strange:
```
X-Powered-By: ESPEasyContent-Length: 205
```

DrakenGray, 2020-09-11:
I am updating with the flash utility, not through the web UI.  I have been verifying the flash has been incrimenting to match the date of the new bin. I've been using Firefox but I did attempt to use edge one time with same results.  The nodeMCU is wifi connected to the same network as the influx server, and I am intercepting the packet with Wireshark on the influx server to read the output.

The header run-on happens if you don't hit enter at the end of the header and make a new line.  Once you do that it runs as normal.   I'll see about making a couple of scenarios on that and grabbing the hex off it. I'm at work right now and left the ESP unplugged so I can't VM in

TD-er, 2020-09-11:
What build do you need?
I will make one from the source I am running here, just to make sure you are running the same code as I am.

TD-er, 2020-09-11:
OK, it indeed looks like the build script I am using was not fetching the PR.
I will look into it as this may lead to serious issues when I can't trust on test builds.

DrakenGray, 2020-09-11:
I am using a nodeMCU esp-12E esp8266.  I can poke at using the web UI to flash, and I also have a couple spare that are blank.  For laughs I also have a esp-12F series I could load to.  (Went and bought a dozen of these to setup a sensor network lol)

TD-er, 2020-09-11:
So you can test with [ESP_Easy_mega_20200911_custom_ESP8266_4M1M.bin](https://www.dropbox.com/s/nyjnr7j063ukhq0/ESP_Easy_mega_20200911_custom_ESP8266_4M1M.bin?dl=0) I just made :)

TD-er, 2020-09-11:
This build for sure does contain the latest changes as they do contain a function name I created today.

- [ESPEasy_ESP32_mega-20200829-49-PR_3250.zip](https://www.dropbox.com/s/vjmojl95vobncn3/ESPEasy_ESP32_mega-20200829-49-PR_3250.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200829-49-PR_3250.zip](https://www.dropbox.com/s/cjzji9iwngcolqq/ESPEasy_ESP82xx_mega-20200829-49-PR_3250.zip?dl=0)

DrakenGray, 2020-09-12:
Its been a fun ride, but I think I'm gonna have to give it up.  Because it works!  Suddenly your requests for send binary make a lot more sense with new check box.  There were a few odd details to get it working just right once %LR% started working, feel like I need to write up an example configuration to help out the next guy now.  Now... Off to figure out the PMS particulate sensor and how to make it work better, and how to get this running on some ESP01s

TD-er, 2020-09-12:
Good to hear it wasn't the programming, but apparently the test builds were not based on the pull requests (still an error on my side)
I was really starting to doubt myself as I really couldn't explain why you still saw those strange issues.
I will close this one. If you have something to add for the documentation, please open an issue (or pull request :) ) for it.

PR #3250 (2020-09-12): [HTTP Advanced] Do not percent-encode body (#1306)
The replaced values are percent-encoded, which causes issues like described here: #1306

Fixes: #1306
closed 2020-09-12 by TD-er