letscontrolit/ESPEasy #1312: '(200) Beacon timeout' & '(201) No AP found'

Oxyandy, 2018-04-24:
### Summarize of the problem/feature request
'(200) Beacon timeout' & '(201) No AP found'
Make using the recent firmwares useless, it happens so quickly I can never load a web page.
I happens straight away.
Tried virgin devices, Erase_Flash, nothing changes the outcome.

### System configuration
Sonoff Basic units with 1M flash

I can back track this if it helps ?
Not going to back track without being asked to.
Last firmware tried mega-20180424 (ESP82xx Core 2_4_0)

TD-er, 2018-04-24:
I just (minutes ago) added some quick fix to do a bit more active reconnects.

However, if you're getting them Beacon timeout so quickly and so often, there may be some other issue here.
The beacon timeout is a common reason to disconnect. But common as in a few times a week maybe a few times per day. Not as frequent as you describe them.
The "No AP found" is something else I just found out yesterday.
ESP32 has an event to signal "STAstart", which will fire when the wifi module is online and stable.
On the ESP32 this will take between 30 and 450 msec (on my test node).
If you're trying to so some wifi related stuff inbetween starting wifi mode and doing something with the wifi, it will result in strange issues.
For example:

- connected but no traffic possible
- infinite timeouts on connections started before the wifi STA mode was stable.
- No AP found errors.

The ESP8266 does not have such a STAstart event. But you could add some delay in the function that actually sets the mode to `WIFI_STA` 
A delay of 250 msec is almost always fine.

There are other new insights into wifi that I should use and the wifi code has become too complex in the last few weeks.
So there should be a simple redesign (or redesign to simplicity) to the wifi code.
I will work on that for the next couple of days and then release it.

So please test the last version of the code (or wait 3 hours and you'll have the fresh 025 build) 

And sometimes it also helps to restart the accesspoint, since beacon timeout should not happen that often, even though there is some logic error in the current code.

Oxyandy, 2018-04-24:
I have read a few bit about ESP32 issues & wonder if fixing some of those breaks other things...
Does ESP32 need a different branch ?

Oxyandy, 2018-04-25:
ESP_Easy_mega-20180425_normal_ESP8266_1024.bin
Full erase before flash
Opened Serial Monitor, cold booted..
AP is not showing anywhere, tried my PC & Mobile Phone..
Cold boot & tried again, no AP "ESP_Easy_0" found to configure Wifi
Tried again, waited.... Confirmed enough times, no access point visible
Time for step 2, see log below this one - Serial Setup
```
INIT : Booting version: mega-20180425 (ESP82xx Core 2_4_0)
73 : INIT : Warm boot #2
75 : FS   : Mounting...
81 : FS   : Mount successful, used 75802 bytes of 113201
379 : CRC  : program checksum       ...OK
386 : CRC  : SecuritySettings CRC   ...OK
404 : INIT : Free RAM:20400
405 : INIT : I2C

405 : INIT : SPI not enabled
419 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_0)
419 : WIFI : Switch on WiFi
420 : WIFI : Set WiFi to STA
422 : WIFI : No valid wifi settings
423 : WIFI : Connection Failed
525 : WIFI : Set WiFi to AP
1415 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1422 : WD   : Uptime 0 ConnectFailures 0 FreeMem 18144
31421 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18376
61421 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18376
```
# Serial WiFi setup
```
>wifissid MAD_MOB

Ok
361421 : WD   : Uptime 6 ConnectFailures 0 FreeMem 18376

>wifikey 12345676

Ok
>save
383973 : FILE : Saved config.dat
384274 : FILE : Saved security.dat

Ok
384275 : WIFI : Set WiFi to AP+STA
384402 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
391421 : WD   : Uptime 7 ConnectFailures 0 FreeMem 17240
```
Wifidisconnect
Wificonnect
does nothing, just serial response OK
Time for Cold Boot
# Cold Boot
```
INIT : Booting version: mega-20180425 (ESP82xx Core 2_4_0)
76 : RTC  : Checksum error on reading RTC user var
76 : INIT : Warm boot #2147483657
78 : FS   : Mounting...
85 : FS   : Mount successful, used 75802 bytes of 113201
380 : CRC  : program checksum       ...OK
387 : CRC  : SecuritySettings CRC   ...OK
404 : INIT : Free RAM:20416

404 : INIT : I2C
405 : INIT : SPI not enabled
418 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_0)
419 : WIFI : Switch on WiFi
419 : WIFI : Set WiFi to STA
426 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19744
427 : WIFI : Connecting MAD_MOB attempt #0
2568 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2140 ms
2568 : WIFI : Connection Failed
```
2669 : WIFI : Set WiFi to AP
3558 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1

TD-er, 2018-04-25:
Ok, may patch may have broken things regarding the AP mode.
About the errors reported. I just added the reason of disconnect to get more insight on what is happening.
Before there may have been WiFi issues, but never logged

Oxyandy, 2018-04-25:
Took another known working perfectly module with existing firmware, set as static IP
Flashed 0425 over the top
Connected differently, not better, could post whole log but quickly I spotted:
```
Wifi  : Detected strange behavior, reset wifi.
WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 1131 ms
WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 10 s
WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2146 ms
```

svde, 2018-04-25:
Trying ESP_Easy_mega-20180425_normal_ESP8266_4096, I setup the wifi via serial, and am seeing this:

520 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19728
521 : WIFI : Connecting nl5658az49 attempt #0
2658 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2136 ms
2658 : WIFI : Connection Failed
2759 : WIFI : Set WiFi to AP
3648 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1

The ESP_Easy_0 AP is not visible to my laptops or phones.

Grovkillen, 2018-04-25:
Still, that recent? That's after event based wifi, right @TD-er ?

Please report back here more suggestions on releases to pick for the async wifi.

melwinek, 2018-04-25:
The web interface works very fast so surely the event based wifi.
But this version is on core 2.3.0.
Maybe it's really one problem 2.4.0 and not event based wifi?

sassod, 2018-04-25:
I hade the same problem all working perfectly on 04102018 after upgrade to
any version after this one I get a bootloop that I can't even downgrade to
previous versions... Need to check how to get out of bootloop on wemos D1.

As far as I understand the 0410 was the last on 2.3 core and after that all
versions use 2.4

Oxyandy, 2018-04-25:
If you back-stepped and tried merging 'important' fixes over the top why not
ESP_Easy_mega-20180403_normal_ESP8266_1024.bin <- CRC : Settings CRC ...OK #1272 
As far as answering the question 
"OK, what was the last version of ESPeasy that simply worked with wifi?"
I would need to test carefully to supply an answer.. Can do..

TD-er, 2018-04-25:
@sassod You may want to hold the "flash" button during flashing.
So press and hold flash, press reset shortly, start writing the firmware while still holding the flash button.
After writing firmware, release flash button and press reset.

At least, that's the method I use for NodeMCU. I assume the Wemos D1 also has a reset and a flash button?

TD-er, 2018-04-25:
Could you also check if it is possible to connect to the setup page and continue the whole process of setup wifi access, when using core 2.3.0 ?  (you probably need clean settings for that)

If that's working right now, I will switch back to 2.3.0

clumsy-stefan, 2018-04-25:
as I'm in the office I don't have physical access to the devices right now, but as far as I could see the last days, that worked. switchign to 2.4.0 or git will not bring up the page anymore.

connecting to the setup-page of a running unit (2.3.0 ESPEasy from tonight) work also fine and shows me the actual connected IP.

Clearing and flashing a unit I could try tonight thoug...

TD-er, 2018-04-25:
ESPeasy master.... I hope you mean "mega" branch?
Master has been renamed to 'legacy' a few days ago.

clumsy-stefan, 2018-04-25:
yes, sure, sorry... told you I'm a beginner ;)

Oxyandy, 2018-04-25:
ESP_Easy_mega-20180410_normal_ESP8266_1024 the last 2.3 core
Access Point shows & connects but is at http://192.168.244.1/setup
not the usual http://192.168.4.1
haven't had much time to test it yet...

Oxyandy, 2018-04-25:
So what is the plan ? Take mega-20180425 and change compile settings ?

TD-er, 2018-04-25:
I'm not sure there is already a plan :)
But you could try that to see if it helps. Sounds like a good start.

Oxyandy, 2018-04-25:
Had problems with AP, Page Title loaded.. not much else, then seemed to lock-up even via serial
Cold Boot Manually set wifi via serial - connected
Web server responds & wifi stays connected

Oxyandy, 2018-04-25:
### 0425 compiled in platformIO with 2.3.0 Core
Even though I have tortured the device, trying to crash it, it has stayed UP
Not one wifi error as I have previously reported
UP time = Wifi connect time..
So the change from 2.4 down to core 2.3 = a huge difference

clumsy-stefan, 2018-04-25:
Just for information:
I have flashed 6 units about four hours ago. Except one, all of them still up and running (with 2.4.0 or GIT it was less than 5min.). One stopped responding after 2h, but this one I use for HW-Tests (sensors, relaiys, etc), so It's likely an issue of the device and not the SW.

The others are running a build from this morning with core 2.3.0 built wirh Arduino on a Mac with a custom plugin set enabled (32 plugins).

hope this helps...

TD-er, 2018-04-25:
It helps a lot.
At least it somewhat proves my code isn't that horrible ;)
And tomorrows build will be 2.3.0 based for sure.

melwinek, 2018-04-25:
Does the compilation in PlatformIO require any additional settings in current versions?
In the past, everything worked without a problem.
Now after uploading and resetting, it does not save wifi settings.
I give the data, then the countdown, and the button return to the settings.
In logs it looks like this:
23059 : WIFI : Set WiFi to AP
31518 : WIFI : Set WiFi to AP+STA
31520 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
31520 : WIFI : Credentials Changed, retry connection. SSID: lipowo
31606 : WD   : Uptime 1 ConnectFailures 0 FreeMem 11640
61606 : WD   : Uptime 1 ConnectFailures 0 FreeMem 12560

TD-er, 2018-04-25:
Hmm that was the reason I switched to 2.4.0, since there is a bug somewhere in 2.3.0 that was fixed in 2.4.0

I have to look into that.

melwinek, 2018-04-25:
It is interesting that the original compilations work ok, it is only I can not build my own.

Oxyandy, 2018-04-25:
I compiled the latest dev tasmota to investigate the 'new' rules, as is
of course, Wifi issues.. haha :)
Now I have a way of 'testing' the builds since 0403, I am going to build them all since then and compare because 0403 seems blazing fast compared to 0425

TD-er, 2018-04-25:
So 0403 was actually "the best" ?

giig1967g, 2018-04-25:
Hi all,
for me yesterday's version works very well.
Absolutely no problem with Wifi
Wemos D1 - 8266 and 8285.

giig1967g, 2018-04-25:
Just for information:
I have uploaded the 1MB blank file firmware.
Then loaded latest firmware.
Attached to internal AP on 192.168.4.1
Then had to manually type http://192.168.4.1/setup
Then I inserted my Wifi cedentials
Then recreated all the devices and uploaded the rules.
I also made several firmware updates since then both from my own compiler and from pre-build version. Now is solid and stable since yesterady 23:00, so almost 24 hours.
I also find it much much faster in Wifi connection then the previous version.

The only problem was that I had to manually type http://192.168.4.1/setup while before this page was appearing automatically.
Other than this, is perfect.

giig1967g, 2018-04-25:
This is my log:
```
107 : INIT : I2C
107 : INIT : SPI not enabled
1216 : INFO : Plugins: 72 [Normal] [Testing] [Development] (ESP82xx Core 2_4_0)
1217 : EVENT: System#Wake
1292 : WIFI : Switch on WiFi
1292 : WIFI : Set WiFi to STA
mode : sta(60:01:94:8e:ba:c9)
                             add if0
                                    1405 : EVENT: System#Boot
2227 : SYS  : 0.00
2229 : EVENT: SysInfoUptime#UptimeDays=0.00
3157 : WIFI : Connecting KeepOut attempt #0
3158 : IP   : Static IP : 192.168.1.206 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 8.8.8.8
3173 : IMPT : MQTT 037 Intentional reconnect
3526 : WIFI : Static IP: 192.168.1.206 (ESPT6-16) GW: 192.168.1.1 SN: 255.255.255.0   duration: 3158 ms
3537 : EVENT: WiFi#Connected
3606 : Webserver: start
scandone
        state: 0 -> 2 (b0)
                          state: 2 -> 3 (0)
                                           state: 3 -> 5 (10)
                                                             add 0
                                                                  aid 5
                                                                       cnt

                                                                           connected with KeepOut, channel 9
                            ip:192.168.1.206,mask:255.255.255.0,gw:192.168.1.1
                                                                              7261 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
7262 : EVENT: Time#Initialized
7350 : EVENT: Clock#Time=Wed,20:33
7432 : IMPT : MQTT 037 Intentional reconnect
7794 : WIFI : Connected! AP: KeepOut (BC:EE:7B:EF:A3:38) Ch: 9 Duration: 3059 ms
7795 : EVENT: WiFi#ChangedAccesspoint
7877 : WIFI : Static IP: 192.168.1.206 (ESPT6-16) GW: 192.168.1.1 SN: 255.255.255.0
7888 : EVENT: WiFi#Connected

```

giig1967g, 2018-04-25:
There are some strange log messages, but it works

clumsy-stefan, 2018-04-25:
quick update: running now on 11 units between 5 and 10 hours with mega commit 2fd226d from this morning and core 2.3.0. The only issue I found until now is, that the webpage does not load automatically the first time (eg. login-page) I have to call a page directly (eg. /tools or /login) after this, everything runs smoothly in the web interface.

svde, 2018-04-25:
My wemos D1 works fine with ESPEasy_mega-20180410.

Oxyandy, 2018-04-25:
@TD-er said 

> So 0403 was actually "the best" ?
You maybe still awake, I have just woken...
Yes, so far 0403 (pre crc fail), is the best
however before I slept I compiled every release which never loaded with 2.3.0
Now I can better evaluate the difference & get back to you.

TD-er, 2018-04-26:
Well, 'awake' is a bit too optimistic I guess.

I will go to sleep now.
Current progress can be seen here:
https://github.com/TD-er/ESPEasy/tree/bugfix/wifi_stability

It does seem to work quite OK with ESP32 and ESP8266, but only when using DHCP.
For static IP the connection gets lost quite often.

Maybe when built against 2.4.1 it will work. But that's for another day.
Now I'll go to bed and I am not merging this into ESPeasy repository.
But at least others can play with it if they like.

Oxyandy, 2018-04-26:
For 3+ hours I have been testing firmwares compiled on 2.3.0
I have made very specific tests & notes for each, awful results really (overall) 
until I loaded 20180421, then 20180422 - with these there are noticeable differences
0422 fixes a bug in 0421 which makes it useless.. Not yet finished with 0422

TD-er, 2018-04-26:
Acceptable freemen is about 10, but with 2.4.0 we do not get that.
The minimum should be > 5k

Oxyandy, 2018-04-26:
As is downloaded from Github
ESP_Easy_mega-20180426_normal_ESP8266_1024.bin
Not really tested, just at this point after flashing
While testing other April firmwares compiled with 2.3.0, 
still had plenty of Wifi errors, so changing from 2.4.0 to 2.3.0 has not solved 'all' those problems, 
some firmwares the Wifi errors are so bad, I can not load the web server...
Not tested past 0422 yet, but this log is 0426 just flashed over 0422's config for a quick look,
some minor Wifi errors in log, do they look acceptable to you ?
will resume more testing later, just got back home.. hungry
```
INIT : Booting version: mega-20180426 (ESP82xx Core 2_3_0)
79 : INIT : Cold Boot
80 : FS   : Mounting...
86 : FS   : Mount successful, used 75802 bytes of 113201
367 : CRC  : program checksum       ...OK
378 : CRC  : SecuritySettings CRC   ...OK
379 : CRC  : binary has changed since last save of Settings
396 : INIT : Free RAM:19456
396 : INIT : I2C
396 : INIT : SPI not enabled
411 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_3_0)
412 : EVENT: System#Wake
416 : WIFI : Switch on WiFi
416 : WIFI : Set WiFi to STA
428 : EVENT: System#Boot
430 : SW   : Switch state 0 Output value 0
436 : EVENT: Float_SW#Switch=0.00
1022 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19616
1033 : WIFI : Connecting MAD_IOT attempt #0
1034 : IP   : Static IP : 192.168.0.225 GW: 192.168.0.254 SN: 255.255.255.0 DNS: 8.8.8.8
1048 : WIFI : Static IP: 192.168.0.225 (ESP-Easy-0) GW: 192.168.0.254 SN: 255.255.255.0   duration: 1034 ms
1059 : EVENT: WiFi#Connected
1065 : Webserver: start
1159 : MQTT : Intentional reconnect
1169 : MQTT : Failed to connect to broker
1510 : MQTT : Intentional reconnect
1517 : MQTT : Failed to connect to broker
11207 : Host unreachable: 0.0.0.0
11208 : Wifi  : Detected strange behavior, reset wifi.
11208 : Webserver: stop
11209 : WIFI : Switch off WiFi
11310 : WIFI : Switch on WiFi
11310 : WIFI : Set WiFi to STA
11311 : NTP  : NTP host au.pool.ntp.org (0.0.0.0) unreachable
11323 : EVENT: WiFi#Disconnected
11328 : WIFI : Disconnected! Reason: '(4) Assoc expire' Connected for 4063 ms
11336 : WIFI : Connecting MAD_IOT attempt #1
11337 : IP   : Static IP : 192.168.0.225 GW: 192.168.0.254 SN: 255.255.255.0 DNS: 8.8.8.8
11344 : WIFI : Static IP: 192.168.0.225 (ESP-Easy-0) GW: 192.168.0.254 SN: 255.255.255.0   duration: 11338 ms
11358 : EVENT: WiFi#Connected
11360 : Webserver: start
11457 : MQTT : Intentional reconnect
11469 : MQTT : Failed to connect to broker
11807 : MQTT : Intentional reconnect
11814 : MQTT : Failed to connect to broker
15639 : WIFI : Connected! AP: MAD_IOT (F4:F2:6D:25:84:C6) Ch: 11 Duration: 2511 ms
15640 : EVENT: WiFi#ChangedAccesspoint
15647 : WIFI : Static IP: 192.168.0.225 (ESP-Easy-0) GW: 192.168.0.254 SN: 255.255.255.0
15657 : EVENT: WiFi#Connected
15818 : Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 600 min
15819 : EVENT: Time#Initialized
15824 : EVENT: Clock#Time=Thu,18:25
15826 : MQTT : Intentional reconnect
19777 : MQTT : Connected to broker with client ID: ESPClient_5C:CF:7F:72:96:EC
19840 : Subscribed to: domoticz/out
20558 : EVENT: MQTT#Connected
25571 : EVENT: Clock#Time=Thu,18:26
31022 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16688
```

v-a-d-e-r, 2018-04-26:
Compiled with your changes from tonight and it seems to run stable with DHCP and static IP. No errors in the log after 1h of run. :-)

clumsy-stefan, 2018-04-26:
@TD-er : just flashed a couple of D1 units with your latest commit from https://github.com/TD-er/ESPEasy/tree/bugfix/wifi_stability compiled with latest GIT core version.... I'll let you know how they behave later today.

TD-er, 2018-04-26:
@Oxyandy That's exactly what I saw last night, right before my eyes told me to get some downtime myself.

Static IP + core 2.3.0 and my branch was absolutely not stable.
It seems like it does not register itself as "connected + got IP" and thus the MQTT connection does not do its loop() calls and the MQTT broker will disconnect the client.
That's how far I got last night.
But at least it is a lot more clean than the code of the last few weeks and that's always nice to have.
I threw away a lot of code and tried to keep it simple.

clumsy-stefan, 2018-04-26:
quick catchup: after the first units ran fine with your simplified code from tonight I flushed now all test devices (~10). We'll see how it goes.

but already now I can say, that my devices seem to like your code quite good, I see certain reconects from time to time (200&201) but at least they reconnected everytime successfully.

one thing that seems to be important when using lwIP2 make sure you take the high bandwith one, otherwise it seems to loose some data from time to time when there is a lot to send (eg. multiple values in a sensor) and the packets get big therefore. With the high bandwith lwIP this seems to be gone..

webinterface looks speedy also cpu is a bit lower in average and free memory higher.

melwinek, 2018-04-26:
Version from 2604 Git compiled with core 2.4.1 not reconnect after router restart, after node restart its ok:
INIT : Booting version:  (ESP82xx Core 2_4_1)
67 : INIT : Warm boot #4
68 : FS   : Mounting...
93 : FS   : Mount successful, used 75802 bytes of 957314
99 : CRC  : No program memory checksum found. Check output of crc2.py
104 : CRC  : SecuritySettings CRC   ...OK
105 : CRC  : binary has changed since last save of Settings
212 : INIT : Free RAM:22576
212 : INIT : I2C
212 : INIT : SPI not enabled
226 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1)
226 : WIFI : Switch on WiFi
226 : WIFI : Set WiFi to STA
228 : WIFI : No valid wifi settings
229 : WIFI : Connection Failed
331 : WIFI : Set WiFi to AP
1221 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1228 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20328
>wifissid lipowo
Ok
>wifikey xxxxxxxxxxxxxxxxxxxxxx
Ok
>save
94313 : FILE : Saved config.dat
94344 : FILE : Saved security.dat
Ok
>wificonnect
Ok
>reboot
INIT : Booting version:  (ESP82xx Core 2_4_1)
61 : INIT : Warm boot #2
62 : FS   : Mounting...
86 : FS   : Mount successful, used 75802 bytes of 957314
93 : CRC  : No program memory checksum found. Check output of crc2.py
98 : CRC  : SecuritySettings CRC   ...OK
204 : INIT : Free RAM:22576
205 : INIT : I2C
205 : INIT : SPI not enabled
218 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1)
219 : WIFI : Switch on WiFi
219 : WIFI : Set WiFi to STA
225 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21920
226 : WIFI : Connecting lipowo attempt #0
3998 : WIFI : Connected! AP: lipowo (11:11:11:11:11:11) Ch: 6 Duration: 3771 ms
8197 : WIFI : DHCP IP: 192.168.3.212 (ESP-Easy-0) GW: 192.168.3.49 SN: 255.255.255.0   duration: 4199 ms
8200 : Webserver: start
30225 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18624
60225 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18072
87339 :      : Rebooting...

INIT : Booting version:  (ESP82xx Core 2_4_1)
63 : INIT : Warm boot #5
64 : FS   : Mounting...
89 : FS   : Mount successful, used 75802 bytes of 957314
96 : CRC  : No program memory checksum found. Check output of crc2.py
100 : CRC  : SecuritySettings CRC   ...OK
207 : INIT : Free RAM:22576
207 : INIT : I2C
207 : INIT : SPI not enabled
221 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1)
221 : WIFI : Switch on WiFi
221 : WIFI : Set WiFi to STA
228 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21920
228 : WIFI : Connecting lipowo attempt #0
5010 : WIFI : Connected! AP: lipowo (11:11:11:11:11:11) Ch: 6 Duration: 4780 ms
5698 : WIFI : DHCP IP: 192.168.3.212 (ESP-Easy-0) GW: 192.168.3.49 SN: 255.255.255.0   duration: 688 ms
5702 : Webserver: start
30228 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18224
2943939 : SYS  : 16928.00
2949938 : SYS  : -72.00
2957938 : SYS  : 49.00
2970232 : WD   : Uptime 50 ConnectFailures 0 FreeMem 17000
3000232 : WD   : Uptime 50 ConnectFailures 0 FreeMem 17000
3003938 : SYS  : 16928.00
3009938 : SYS  : -72.00
3017938 : SYS  : 50.00
3030232 : WD   : Uptime 51 ConnectFailures 0 FreeMem 17000
3039019 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 50 m 33 s
3039936 : Webserver: stop
3039936 : WIFI : Connecting lipowo attempt #0
3042798 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
3042799 : WIFI : Connection Failed
3042899 : WIFI : Set WiFi to AP
3043789 : WIFI : AP Mode ssid will be espt_99 with address 192.168.4.1
3060232 : WD   : Uptime 51 ConnectFailures 0 FreeMem 16632
3064800 : SYS  : 16560.00
3070799 : SYS  : 31.00
3078799 : SYS  : 51.00
3090232 : WD   : Uptime 52 ConnectFailures 0 FreeMem 16632
3120232 : WD   : Uptime 52 ConnectFailures 0 FreeMem 16632
3124800 : SYS  : 16560.00
3130800 : SYS  : 31.00
3138800 : SYS  : 52.00
3150232 : WD   : Uptime 53 ConnectFailures 0 FreeMem 16632
3180234 : WD   : Uptime 53 ConnectFailures 0 FreeMem 16632
3184801 : SYS  : 16560.00
3190813 : SYS  : 31.00
3198812 : SYS  : 53.00
3210234 : WD   : Uptime 54 ConnectFailures 0 FreeMem 16632
3240234 : WD   : Uptime 54 ConnectFailures 0 FreeMem 16632
3244813 : SYS  : 16560.00
3250813 : SYS  : 31.00
3258813 : SYS  : 54.00
3270234 : WD   : Uptime 55 ConnectFailures 0 FreeMem 16632
3300234 : WD   : Uptime 55 ConnectFailures 0 FreeMem 16632
3304813 : SYS  : 16560.00
3310813 : SYS  : 31.00
3318813 : SYS  : 55.00
3330234 : WD   : Uptime 56 ConnectFailures 0 FreeMem 16632
3360234 : WD   : Uptime 56 ConnectFailures 0 FreeMem 16632
3364814 : SYS  : 16560.00
3370814 : SYS  : 31.00
3378814 : SYS  : 56.00
3390234 : WD   : Uptime 57 ConnectFailures 0 FreeMem 16632
3420234 : WD   : Uptime 57 ConnectFailures 0 FreeMem 16632
3424814 : SYS  : 16560.00
3430814 : SYS  : 31.00

v-a-d-e-r, 2018-04-26:
Update:
WD   : Uptime: 369 min, ConnectFailures: 0, FreeMem: 24784 bytes
EVENT: Clock#Time=Thu,22:10

No issues after over 6h running. No error in log! :-)

Oxyandy, 2018-04-27:
I have seen people have success with 2.40 & 2.41 core, I have tried building 0422, 0425 & 0426 
connectivity has become worse for me than using 2.3.0
Still 0403 seems to be last stable wifi for me so I tried 0403 compiled using 2.4.1 core, works perfect but looks like less free mem

giig1967g, 2018-04-27:
@Oxyandy : could it be that it's a hardware problem? Either your unit, or your unit model or your power supply or your router? Or maybe your configuration?

Oxyandy, 2018-04-27:
Tried a few units the results are identical, absolute minimal config... 
any units I've had problem with after downgrading to 0403 have been perfect..

Oxyandy, 2018-04-27:
0403, compiled using 2.4.1 Core, connects first try, updates time straight away, no wifi errors & stays connected, web server responds every time.. 
Point is, with 0403 it doesn't seem to matter which core I use, it just works... 
later versions using a higher core makes existing issues with wifi worse..
Note: firmware compiled with 2.4.1 is 41K larger.
```
INIT : Cold Boot
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 113201
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK
INIT : Free RAM:20176
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy-0 with address 192.168.4.1
IP   : Static IP :3774916800
WIFI : Connecting MAD_IOT attempt #0
EVENT: System#Boot
SW   : Switch state 1 Output value 1
EVENT: Float_SW#Switch=1.00
WD   : Uptime 0 ConnectFailures 0 FreeMem 18608
IP   : Static IP :3774916800
IP   : Static IP :3774916800
IP   : Static IP :3774916800
IP   : Static IP :3774916800
IP   : Static IP :3774916800
WIFI : Connected! AP: MAD_IOT (F4:F2:6D:25:84:C6) Ch: 11 Duration: 3999 ms
EVENT: WiFi#ChangedAccesspoint
WIFI : Static IP: 192.168.0.225 (ESP-Easy-0) GW: 192.168.0.254 SN: 255.255.255.0
EVENT: WiFi#Connected
MQTT : Intentional reconnect
MQTT : Connected to broker with client ID: ESPClient_5C:CF:7F:72:96:EC
Subscribed to: domoticz/out
EVENT: MQTT#Connected
Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 600 min
EVENT: Time#Initialized
EVENT: Clock#Time=Fri,21:57
EVENT: Clock#Time=Fri,21:58
WD   : Uptime 1 ConnectFailures 0 FreeMem 17504
WD   : Uptime 1 ConnectFailures 0 FreeMem 17504
EVENT: Clock#Time=Fri,21:59
WD   : Uptime 2 ConnectFailures 0 FreeMem 17504
WD   : Uptime 2 ConnectFailures 0 FreeMem 17504
EVENT: Clock#Time=Fri,22:00
```

melwinek, 2018-04-27:
Build from git https://github.com/TD-er/ESPEasy/tree/bugfix/wifi_stability with core 2.4.1 from git https://github.com/esp8266/Arduino is not sensitive to router reset or reset node .

Perhaps the problem was:
https://github.com/esp8266/Arduino/commit/b08d282673055b4758cd73d3cd99573f619112a5
This has been repaired after the release of 2.4.1.

Check at yourself. For me, this is the first build above 20180410, which gets up after resetting the router.

An attachment this firmware for those who do not want or can not compile.
[firmware.zip](https://github.com/letscontrolit/ESPEasy/files/1955550/firmware.zip)

giig1967g, 2018-04-27:
@melwinek : how do I change the core or update it to the latest patch?
I am using 8285 now, so cannot test your firmware.

TD-er, 2018-04-27:
For those who need a little help in building, I just built a version of the patch I wrote 2 days ago, but now with core 2.4.1:
[TD-er_wifi_stability_core-2.4.1](https://www.dropbox.com/s/ju61xukhclidppu/TD-er_wifi_stability_core-2.4.1.rar?dl=0)

giig1967g, 2018-04-27:
@melwinek : thanks for the detailed instructions. I compiled.

Result: 
With STATIC IP it doesn't reconnect to Wifi after a router reboot (but the serial log is misleading as it says Wifi#Connected).
With DHCP it reconnects and it is very stable.

@TD-er : any patch to apply in order to fix the static IP problem?

giig1967g, 2018-04-27:
@TD-er : thanks!
For your info, with static IP, in version mega-20180330 the unit reconnects after a router reboot.

melwinek, 2018-04-27:
@TD-er @giig1967g  20180410 also

melwinek, 2018-04-27:
@TD-er your build from dropbox is with core 2.4.1 or 2.4.1 GIT ?

TD-er, 2018-04-27:
I used these settings in PlatformIO.ini

```
[core_2_4_1]
platform         = espressif8266@1.7.0

[common]
build_flags      = -D BUILD_GIT='"${env.TRAVIS_TAG}"'  ; ${compiler_warnings.build_flags}
                   -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
```

melwinek, 2018-04-27:
I think you're using https://github.com/esp8266/Arduino/releases/tag/2.4.1 instead https://github.com/esp8266/Arduino

On a stable 2.4.1 do not reconnect. At least for me.
But I do not know how to use 2.4.1 GIT professionally, I went on the easy way. Platformio and git is a novelty for me.

TD-er, 2018-04-27:
With the build I just shared, I am now running with static IP.
It looks like the problems with static IP are gone when using core 2.4.1
Nothing changed on the sources, just other platformIO settings.

I will test a bit more and have to fix a number of warnings (or else Travis will fail), but I guess I could push and merge it to the main branch.

It would be really great if the wifi issues were really over, then we could continue on the rest of the development.

giig1967g, 2018-04-27:
@TD-er : did you try rebooting the wifi router?

TD-er, 2018-04-27:
Yep, including switching over to another AP.

[94 more replies]

closed 2018-05-02 by Oxyandy