letscontrolit/ESPEasy #1313: v2.0-20180322/mega-20180424 and TP-Link TL-WA901ND Wireless N450 3TER Access Point

rwlove, 2018-04-25:
### Summarize of the problem/feature request

v2.0-20180322 and mega-20180424 will not connect to my TP-Link access point, R120 will.

I am running a Linux server (DHCP/DNS) behind the AP and I never see a DHCP request. The ESPEasy FW returns to the 192.168.4.1/setup page after the 20 second wait/timeout.

### Expected behavior

ESPEasy should connect to the AP and send a DHCP request. This request should be visible in the dhcp server's logs.

### Actual behavior

ESPEasy does not connect to the AP.

### Steps to reproduce

1. Flash ESPEasy image (v2.0-20180322 or mega-20180424) 
2. disconnect from computer and connect to USB power source
3. Access ESPEasy Access Point and select local network SSID. Enter password.
4. ESPEasy counts down for 20 seconds and returns to the initial SSID selection page.

### System configuration

Hardware:
*  [TP-Link TL-WA901ND Wireless N450 3TER Access Point](https://www.amazon.com/gp/product/B002YETVXC/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1)
* [ESP8266 ESP-12 ESP12 WeMos D1 Mini Module Wemos D1 Mini](https://www.aliexpress.com/item/ESP8266-ESP12-ESP-12-WeMos-D1-Mini-WIFI-Dev-Kit-Development-Board-NodeMCU-Lua/32653918483.html)

FAILED ESP Easy versions:
* [mega-20180424](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180424)
* [v2.0-20180322](https://github.com/letscontrolit/ESPEasy/releases/tag/v2.0-20180322)

WORKING ESP Easy version:
* [R120](http://www.letscontrolit.com/downloads/ESPEasy_R120.zip)

ESP Easy settings/screenshots: 

### Rules or log data

```

```

Oxyandy, 2018-04-25:
With the very latest releases there are Wifi issues
Try ESP_Easy_mega-20180403_normal_ESP8266_1024.bin
Which exact images have you tried ?
Cheers

rwlove, 2018-04-25:
ESP_Easy_mega-20180424_normal_ESP8266_4096.bin
ESP_Easy_v2.0-20180319_normal_ESP8266_4096.bin
both fail.

Oxyandy, 2018-04-25:
Try 
ESP_Easy_mega-20180403_normal_ESP8266_4096.bin
Was last release version, I had reliably load..

rwlove, 2018-04-25:
I just got ESP_Easy_v2.0-20180322_normal_ESP8266_4096.bin to connect to my AP.

I'm not sure why I thought the v2.0-20180319 was the most recent, when v2.0-20180322 was published.

I may or may not try the mega-20180403 build, depending on how this v2.0-20180322 treats me.

Thanks, @Oxyandy !

closed 2018-04-25 by rwlove

Oxyandy, 2018-04-25:
@rwlove 
Check back in a month or so, recently there has been lots of changes and although these changes have introduced bugs, the work is constant & the improvements (when they work) are great.