letscontrolit/ESPEasy #1316: ESP32 Rules - download to file

flexiti, 2018-04-25:
just ..."This page not work"

TD-er, 2018-04-25:
What is the link you are using?
Please add a bit more information.

flexiti, 2018-04-25:
Rules, download to file button on battom

TD-er, 2018-04-25:
OK, I have to check when I am at home. I guess I am not too bright right now, to remember how that page looks like.

Grovkillen, 2018-04-25:
Does the link point to `<ip-of-unit>/rules1.txt` ?

flexiti, 2018-04-25:
http://192.168.0.68/rules1.txt  , yes

Grovkillen, 2018-04-25:
And you have rules stored?

TD-er, 2018-04-25:
One thing that comes to mind is that the path to the files on SPIFF in the ESP32 have a different path.
All have a "/" prefix.

Grovkillen, 2018-04-25:
Ah, so that path would be present if looked in the FLASH file list `http://<ip>/filelist`?

flexiti, 2018-04-25:
1. at ESP8266 works
2. stored ? means submited ? ( if, yes)

flexiti, 2018-04-25:
on esp 8266 looks:  rules1.txt,
on esp32  :  /rules1.txt

Grovkillen, 2018-04-25:
Aha, and if you direct your browser to `<ip-of-unit>//rules1.txt` it will download?

flexiti, 2018-04-25:
looks like yes

Grovkillen, 2018-04-25:
I will fix, thanks!

TD-er, 2018-04-25:
@Grovkillen I am really curious about what fix you will come up with ;)

Grovkillen, 2018-04-25:
Okay, not a fast fix. Not to me... ;)

liklev, 2019-05-22:
hi, esp-wroom-esp32 modul flashed with flashserial.cmd ESPEasy32_R20100.bin
web gui seems to work but
rules randomly truncated on submit (both chrome and firefox browser)
flash upload does not upload the rules1.txt saved from a working esp mega from an esp12 modul)
i cannot place a sendtohttp long command to my sonoff. (cut off from sonoff_ip,port,/cm?)
looks like short timerset and gpio,x,x in on-do-endon saved and works, but longer lines cut off
regards, lev

TD-er, 2019-05-23:
That's a binary file from more than a year ago.
Can you try one of the latest nightly builds?
Please note that the partitioning has changed, so if you flash the ESP32 build of the latest nightly builds, it will remove the SPIFFS and thus also the existing settings.

liklev, 2019-05-23:
hi TD-er, should i use the mega nightly builds or have some designated stuff for esp32 chips?
cause i found only the megas.

liklev, 2019-05-23:
well, last i downloaded 20190508 there was not esp32 version yet but today 20190523 version i see the mega package nicely beefed up. sooo much that i got lost what i should put finally on my nodemcu board. the shield print says its a wroom 32.
the serialflasher.cmd script was putting 3 or 4 bin files in a row.

TD-er, 2019-05-23:
i don't know that flasher.
there should be a single bin file.
something with 1m8 partition in the name.

liklev, 2019-05-23:
i used the latest available ESPEasy_mega-20190523 package and the flasher
what is provided with it esptool.exe with a gui.
the FW is ESP_Easy_mega-20190523_esp32test_1M8_partition.bin
the flasher can not start flashing unless i press the boot/flash switch on
the board right on time just after i click the "start FW hlashing" on the
gui screen.
sometimes it succeeds sometimes it does not.
even it succeeds  i dont have the usual web interface to make /setup/config.
maybe the board is much exotic design.

liklev, 2019-05-24:
i can not get any of the newer versions (other than 20100) to work (some loop going on at boot complains some c file error) 
but i found the reason of rules page submit truncation on the 20100 FW.
it was the ampersand character in the sendtohttp line. the submit button saved the rules up to the first occurrence of "&"  character (user/password etc. separated by "&") and cut away the rest.
after i replaced all "&" with "%26"  the whole rules was saved uncutted and the command went through without problem.

TD-er, 2019-05-24:
The ESPeasy flasher is currently not working well with the ESP32, so that's indeed not really a good solution at this moment. Sorry about that.
We have to invest more time into the ESP32 than we do now.

closed 2023-09-03 by TD-er