letscontrolit/ESPEasy #1317: Simple wall switch trigger

radebebek, 2018-04-25:
### Summarize of the problem/feature request

I use Sonoff basic for shutting on/off the lights. I want that my simple wall switch turn on relay which is connected to the light. I use this configuration for button:

**Device task:**
 TASK1
```
Device: Switch input
Name: Switch
1st GPIO: Gpio 12
Switch Type: Switch
Switch Button Type: Normal Switch
Value: Stanje
```
 TASK2
```
Device: Switch input
Name: Button
1st GPIO: Gpio 14
Switch Type: Switch
Switch Button Type: PB Active High
Value: Stanje
```
Rules:
```
on Button#Stanje=1 do 
         if [Switch#Stanje]=0 
              gpio,12,1
        else 
              gpio,12,0
        endif 
endon
on Button#Stanje=0 do 
         if [Switch#Stanje]=0 
              gpio,12,1
        else 
              gpio,12,0
         endif 
endon
```
Every time when I use simple wall switch, I must press two times that turn on/off Light.
But when I use Push wall switch I must press only one time that turn on/off Light.
Difference in wall switch: [Picture](https://user-images.githubusercontent.com/29672138/39242631-fbcd881e-488a-11e8-86aa-d3effa590a48.png).

Button#Stanje change only when connect and disconnect. I want that change when I connect and change when I disconnect GPIO14-GND with simple wall switch.
### Expected behavior

I want to use simple wall switch with one press to change state of Lights ON/OFF.

### Actual behavior

Now I need two press. 

### System configuration

Hardware: Sonoff Basic

ESP Easy version: Release v2.0-20180322

### Rules or log data

```

```

TD-er, 2018-04-25:
At first glance, you may want to switch the Button type to a normal switch when using the normal wall switch.

A normal switch does just report the state of the GPIO pin. The other 2 options (PB active high / active low) toggle the internal value.

radebebek, 2018-04-26:
@TD-er 
I set normal switch for button and works without any problems. Thanks a lot.
I have one more question related to this.
I use same configuration but with SONOFF DUAL and I use GPIO 0 and GND for Button1.
I dont have any problem when sonoff dual turn on.
But when the current is gone and after few minutes come. Sonoff go in falsh mode if Button1 (wall switch stay in that position.)
How to avoid it?

closed 2018-04-26 by radebebek

TD-er, 2018-04-26:
Gpio0 is normally used to force flash mode at boot.
I would only use that pin for a push button, not a switch

TD-er, 2018-04-26:
GPIO 0 is the one used to select flash mode.
So as long as that one is connected to GND, then the ESP will boot into flash mode.
So only use that one as a push button, that's what Sonoff does.