letscontrolit/ESPEasy #1318: HWL8012 still missing despite it is listed in release notes

fly74, 2018-04-26:
Hi Developers,

HWL8012 is in release annouced, but not found in devices, what do i wrong?:

 Changes in mega-20180421 (since mega-20180420):

ESPEasy release bot (1):
...
TD-er (7):
...
      [Sonoff Pow] Add support for HLW8012
      [Sonoff Pow] Add support for HLW8012
      [Sonoff POW] Change Plugin ID to 76
      [Sonoff POW] Set some proper defaults for the Sonoff POW

TD-er, 2018-04-26:
It is currently only available for the Sonoff POW.
It is using too much RAM, so it cannot be combined with a 'normal' or other build.
If you're building using PlatformIO, you can find it in the build predefines.

N.B. it has not yet been tested, and as such is not included in the nightly builds.

There will eventually be a separate build for the Sonoff POW and later perhaps also one for the POW R2, which is quite different from the original POW

fly74, 2018-04-26:
At first thanks for quick reply!

I want to use it with a Sonff POW. The last I've flashed was ESP_Easy_mega-20180426_normal_ESP8285_1024.bin. Should I use an other image for the Sonoff POW?

TD-er, 2018-04-26:
The original POW does seem to have the ESP8266. The POW R2 has another CPU.
You should build the "Sonoff POW" version using PlatformIO.

M-e-i-k-e-l, 2018-04-30:
Dear TD-ler, dear Grovkillen,
Thanks a lot for your contributions!
Are binaries available for the Sonoff platforms in the mean time? I have no Arduino IDE installed so far and I want to test the Sonoff Pow current measurement with the 2 devices I have here on my workbench.
Regards; Michael

TD-er, 2018-04-30:
I can build one later this evening.
Please note that they are not yet tested and only support the original POW. Not the POW R2.

fly74, 2018-04-30:
Why not for the POW R2, is the 8285 not compatible with 8266?

TD-er, 2018-04-30:
@M-e-i-k-e-l [click](https://www.dropbox.com/s/68n22svgobbmcvq/ESPeasy_20180430_Sonoff_POW.rar?dl=0)

M-e-i-k-e-l, 2018-05-01:
Thanks a lot, Gijs!
I flashed it on a device, but no AP or Wifi client.
So I took a look again on the PCB: The printed version number on top is 2.0 (but not R2), The ESP is a 8266, but the voltage/current-to-frequency converter is a CSE7759 (https://datasheet.lcsc.com/szlcsc/CSE7759_C126724.pdf).
So I guess, the wifi issue is more related to the 'old' settings of the previous firmware (ESP_Easy_mega-20180422_dev_ESP8266_4096.bin), but it will not work anyway because of the different sensor IC.

So I stuck on the normal mega firmware and use it as a switch only for a while. Maybe it is just an adaptation to the different registers and maybe different squarewave frequency and range. I cannot provide programming support the next weeks as I'm working on another project on a STM32F1 using eclipse.

Regards,
  Michael

TD-er, 2018-05-01:
Hmm, that was the chip used on the POW R2 if I'm not mistaken.
I have to open mine to see what's in there.

fly74, 2018-05-01:
[image]
[image]

here some pics from mine V2.0 POW. It looks like it is not R2.

M-e-i-k-e-l, 2018-05-01:
Here is mine:
To side with version print:
https://www.dropbox.com/s/u2hemfv41c17wli/SonoffPow_top.jpg?dl=0
Botton side:
https://www.dropbox.com/s/9cmc224saqlybuy/SonoffPow_bot.jpg?dl=0
Detail view:
https://www.dropbox.com/s/7dtkrjqqd6wyhzg/SonoffPow_bot_detail.jpg?dl=0

So, it seems to be an update of "R1". An intermediate step towards R2?

Regards,

   Michael

M-e-i-k-e-l, 2018-05-01:
I found some discussions about the versions of the IC, HLW8012, CSE7759 and the BL0937 doing the same job:
https://github.com/xoseperez/espurna/issues/696
another smart plug: https://github.com/xoseperez/espurna/issues/737

TD-er, 2018-05-01:
@Oxyandi was also reporting issues with build 0430 (and 0501), while 0429 worked just fine on his nodes.
I will try to reproduce it this evening.

fly74, 2018-05-01:
I would give it a try if anyone can compile it.

fly74, 2018-05-03:
I compiled mega-20180502 with platformIO and it work's fine. Is it possible to compile it with regular bin-files you provide each time? It seems to be the right one for POW V1,2 not R2.

cflurin, 2018-05-15:
Hi,
I've just received a Sonoff pow R2. Here some infos:
```
Sonoff Pow R2
V1.0
2017-11-30
```
Chips:
```
ESP8266EX
CHIPSEA 7766-SO
```
I tried with the latest version `ESP_Easy_mega-20180515_hard_SONOFF_POW.bin`
but obviously I don't get any power data.

Any news about a version for Sonoff Pow R2?

TD-er, 2018-05-15:
Not yet implemented. I did receive mine 2 weeks ago, but didn't have got time to implement it.

cflurin, 2018-05-15:
Ok thanks, coding ESP8266 isn't my domain but I'd test it. Generally I use ESP8266 (ESPEasy) devices connected with node-red. I'd like to use a `Sonoff Pow R2` to detect whether a water pump is working properly.

TD-er, 2018-11-27:
It was already present for a while.
Now we have a (new) special purpose build for the Sonoff POW/POWr2. This build is called `hard_SONOFF_POW_4M`
Both setups are now included and all button/relay/rules to make some proper default are available in the new Factory Reset page as preset.

So I will close this issue.

closed 2018-11-27 by TD-er