letscontrolit/ESPEasy #1320: ESP32 Tools-Firmware-Load - no functionality

flexiti, 2018-04-26:
ESP32 load firmware is blocked because it need to change a lot of new software, or the function not available by accident?

TD-er, 2018-04-26:
I guess -based on an issue reported yesterday- that there is some issue with path name on SPIFF of the esp32

Budman1758, 2018-04-26:
I have not seen the load firmware function in any of the ESP32 firmwares. If you enable the Arduino OTA checkbox on the Advanced Settings page you can update the firmware that way.  That functionality has always worked quite well.

flexiti, 2018-04-26:
[image]

Budman1758, 2018-04-26:
Someone may know why that is missing in ESP32 but as I said the same functionality is available with the OTA upload.

waspie, 2018-04-27:
use flashota as stated, works a treat

mvdbro, 2018-04-27:
This feature is not implemented in Arduino ESP32 core, so we can't offer it

flexiti, 2018-04-30:
@mvdbro 
I still do not know why you can not get this functionality for ESP32 how does it work for ES8266 ?? and why did you close the topic? will it never be done because you thought so?
Please read what was the first opening question. (I certainly did not ask how it can workarounded  :)

Budman1758, 2018-04-30:
@flexiti I thought @mvdbro answered your question pretty clearly. OTA flash has exact same function. Perhaps you should try it?

mvdbro, 2018-05-01:
Current ESPEasy on ESP32 only supports "Arduino OTA" because the ESP8266HTTPUpdateServer library has not been provided by the ESP32 core team.

I consider the current OTA implementation on ESP32 as the preferred way on ESP32 and certainly not a workaround. It's even a lot faster.

But i don't think it will never be done and it looks like Gijs wants to have a go at it anyway, so i'll reopen the issue.

flexiti, 2018-05-06:
ESP32 - Arduino OTA to make it work from platformio , I had to change the port to 3232 in the code,
(I set in Platformio.ini upload_port = 192.168.0.69 - this is my ESP)
Anything else should be set or should I change in the code as below?

Misc.ino
void ArduinoOTAInit()
{
  checkRAM(F("ArduinoOTAInit"));
  **#if defined(ESP8266)
  // Default port is 8266
  ArduinoOTA.setPort(8266);
  #endif
  **#if defined(ESP32)
  ArduinoOTA.setPort(3232);
  #endif****

flexiti, 2018-05-06:
Btw. ESP32 looks like it works, but ESP8266 does not work

```
Building .pioenvs\esp32dev\firmware.bin

esptool.py v2.1
Configuring upload protocol...
Looking for upload port...
Use manually specified: 192.168.0.68
Uploading .pioenvs\esp32dev\firmware.bin
20:28:47 [DEBUG]: Options: {'esp_ip': '192.168.0.68', 'host_port': 58995, 'image': '.pioenvs\\esp32dev\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 3232, 'spiffs': False, 'debug': True, 'progr
ess': True}
20:28:47 [INFO]: Starting on 0.0.0.0:58995
20:28:47 [INFO]: Upload size: 898224
Sending invitation to 192.168.0.68
20:28:47 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...
 
20:29:04 [INFO]: Waiting for result...
20:29:04 [INFO]: Result: OK
 [SUCCESS] Took 63.12 seconds
```

```
Building .pioenvs\dev_ESP8266_4096\firmware.bin
Configuring upload protocol...
Looking for upload port...
Use manually specified: 192.168.0.30
Uploading .pioenvs\dev_ESP8266_4096\firmware.bin
20:44:34 [DEBUG]: Options: {'esp_ip': '192.168.0.30', 'host_port': 20567, 'image': '.pioenvs\\dev_ESP8266_4096\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True
, 'progress': True}
20:44:34 [INFO]: Starting on 0.0.0.0:20567
20:44:34 [INFO]: Upload size: 757856
Sending invitation to 192.168.0.30 ..........
20:44:44 [ERROR]: No response from the ESP
*** [upload] Error 1
```

flexiti, 2018-05-06:
ok, ESP8266 is blocked in ESPEasy-Globals.h
In my opinion, in this case, option   **Arduino OTA enable** in tools/Advanced
should be unavailable

```
#if defined(ESP8266)
  //enable Arduino OTA updating.
  //Note: This adds around 10kb to the firmware size, and 1kb extra ram.
  // #define FEATURE_ARDUINO_OTA

  //enable mDNS mode (adds about 6kb ram and some bytes IRAM)
  // #define FEATURE_MDNS
#endif
#if defined(ESP32)
 #define FEATURE_ARDUINO_OTA
 //#define FEATURE_MDNS
#endif
```

TD-er, 2018-05-06:
I guess you're right, when that is not set, the checkbox should be set to disabled (grey-out)

flexiti, 2018-05-06:
@TD-er  Could you add this code? :) I have a few fixes at my fork, and I have not yet learned to do Pull for the selected patch, code snippet 
```
#if defined(ESP8266)
// Default port is 8266
ArduinoOTA.setPort(8266);
#endif
#if defined(ESP32)
ArduinoOTA.setPort(3232);
#endif
```

flexiti, 2018-05-07:
done in #1382

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er