letscontrolit/ESPEasy #1321: dnsServer crashes networkstack on esp8266

Feuerreiter, 2018-04-26:
### Summarize of the problem/feature request

dnsServer crashes networkstack on esp8266

### Expected behavior

dnsServer should response ip of captive portal

### Actual behavior

After calling a webside on the browser (e.g. firefox) the esp freezes. No ping reply from esp. After reset ping is okay.

### Steps to reproduce

1. Connect pc with esp ap
2. call side e.g. 192.168.4.1 or 192.168.4.1/setup
3. device freeze

### System configuration

Hardware:
nodemcu amica (esp8266)

mega-20180426
ESP Easy version: 

if i disable the dnsserver in source, everything is fine. I think that the dnsserver don't call the networkstack (yield) or it runs into memory problems.

Feuerreiter, 2018-04-26:
In ESPEasyWifi.ino i do following changes...
`// Start DNS, only used if the ESP has no valid WiFi config
      // It will reply with it's own address on all DNS requests
      // (captive portal concept)
      dnsServerActive = true;
      // dnsServer crashes network stack on esp8266
      //dnsServer.start(DNS_PORT, "*", apIP);
      break;`

Feuerreiter, 2018-04-27:
I try a fresh build with esp core 2.4.1
Core 2.4.1 would solve the problem with the dnsServer. Captive portal is now working too. :-)
With core 2.3.0 it wouldn´t work.

I suggest to switch over to esp core 2.4.1.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er