letscontrolit/ESPEasy #1326: Wifi no reconnect if connection was lost

fly74, 2018-04-28:
For a while I noticed the wifi is not reconnecting if the Wifi was turn off for a while.
Maybe since 2 month. Before I could start my Home-Wifi after beeing away and the ESPs connected automatically. Has somthing changed? The only thing is when Wifi is turned off that ESP offers the local ESP-AP and stays until reset ESP for connection to Home-Wifi.

TD-er, 2018-04-28:
A lot was going on with regards to WiFi.
Last night's build should be a big step towards stability again.
If you browse through the most recently updated issues you will see there were a lot of WiFi related issues, a lot of tests and changes and discussion.

micropet, 2018-04-28:
For me, the reconnect error is still there.
I'm using the version of tonight on a Wemos D1 mini

244263 : ACT  : Publish ESP-201/IP,192.168.0.201
244277 : ACT  : Publish ESP-201/MAC,5C:CF:7F:0B:68:52
244293 : ACT  : Publish ESP-201/Time,12:46:52
244308 : ACT  : Publish ESP-201/Uptime,4
244323 : ACT  : Publish ESP-201/RSSI,-60
244339 : ACT  : Publish ESP-201/SSID,SMC
244353 : ACT  : Publish ESP-201/BSSID,78:8A:20:D1:9B:D9
244383 : ACT  : Publish ESP-201/CH,1
244397 : ACT  : Publish ESP-201/SYSHEAP,13432
244411 : ACT  : timerSet,1,60
252253 : EVENT: Clock#Time=Sat,12:47
271196 : WD   : Uptime 5 ConnectFailures 0 FreeMem 16144
301196 : WD   : Uptime 5 ConnectFailures 0 FreeMem 16144
302274 : DS   : Temperature: 21.06 (28-ff-b8-ea-b4-16-3-ed)
302276 : EVENT: DS18b20#Temperature=21.06
305257 : EVENT: Rules#Timer=1
305268 : ACT  : Publish ESP-201/IP,192.168.0.201
305282 : ACT  : Publish ESP-201/MAC,5C:CF:7F:0B:68:52
305297 : ACT  : Publish ESP-201/Time,12:47:53
305313 : ACT  : Publish ESP-201/Uptime,5
305328 : ACT  : Publish ESP-201/RSSI,-61
305343 : ACT  : Publish ESP-201/SSID,SMC
305358 : ACT  : Publish ESP-201/BSSID,78:8A:20:D1:9B:D9
305387 : ACT  : Publish ESP-201/CH,1
305402 : ACT  : Publish ESP-201/SYSHEAP,13432
305417 : ACT  : timerSet,1,60
312254 : EVENT: Clock#Time=Sat,12:48
331196 : WD   : Uptime 6 ConnectFailures 0 FreeMem 16144
349941 : MQTT : Connection lost
349941 : EVENT: MQTT#Disconnected
349981 : MQTT : Failed to connect to broker
349982 : EVENT: WiFi#Disconnected
350000 : WIFI : Disconnected! Reason: '(1) Unspecified' Connected for 5 m 45 s
350481 : MQTT : Failed to connect to broker
351231 : MQTT : Failed to connect to broker
352231 : MQTT : Failed to connect to broker
353481 : MQTT : Failed to connect to broker
354981 : MQTT : Failed to connect to broker
356731 : MQTT : Failed to connect to broker
358731 : MQTT : Failed to connect to broker
360731 : MQTT : Failed to connect to broker
361196 : WD   : Uptime 6 ConnectFailures 18 FreeMem 17176
361218 : MQTT : Failed to connect to broker
362275 : DS   : Temperature: 21.06 (28-ff-b8-ea-b4-16-3-ed)
362276 : EVENT: DS18b20#Temperature=21.06
362731 : MQTT : Failed to connect to broker
364731 : MQTT : Failed to connect to broker
366255 : EVENT: Rules#Timer=1
366265 : ACT  : Publish ESP-201/IP,0.0.0.0
366278 : ACT  : Publish ESP-201/MAC,5C:CF:7F:0B:68:52
366290 : ACT  : Publish ESP-201/Time,12:48:54
366301 : ACT  : Publish ESP-201/Uptime,6
366312 : ACT  : Publish ESP-201/RSSI,0
366323 : ACT  : Publish ESP-201/SSID,--
366334 : ACT  : Publish ESP-201/BSSID,00:00:00:00:00:00
366357 : ACT  : Publish ESP-201/CH,0
366368 : ACT  : Publish ESP-201/SYSHEAP,16784
366378 : ACT  : timerSet,1,60
366731 : MQTT : Failed to connect to broker
368731 : MQTT : Failed to connect to broker
370731 : MQTT : Failed to connect to broker

TD-er, 2018-04-28:
Is it a good idea to let N connect to broker failures lead to a forced wifi disconnect?
For example N being 5?

This until we find another way to detect whether there is a connection but no data transfer possible.

micropet, 2018-04-28:
5 times is definite enough.

TD-er, 2018-04-28:
OK, will add that.  If it is too long, or causing other issues, we could always change it ofcourse.

mvdbro, 2018-04-28:
refer to #1302

closed 2018-04-28 by mvdbro