letscontrolit/ESPEasy #1327: FreeMem as Systemvariable

micropet, 2018-04-28:
Can we have FreeMem as Systemvariable?

TD-er, 2018-04-28:
Sure, although I am not really sure what value it should reflect.
We have a stack and a heap. Normally the distance between those two addresses is the free memory.

But for our purpose, that value is not very interesting, since we do not variate on stack usage very much.
On the other hand, we use the heap a lot with our Strings.

micropet, 2018-04-28:
Well, I thought freeMem is important.
Because you have ever asked for it.
Then we do not need it.

TD-er, 2018-04-28:
Freemem is important, but a single number doesn't tell the whole story.
Logging freemem and graphing it, is a good indication of how much free mem is needed and if there is a memory leak.
But maybe I'm thinking too deep now and it is just a good start to have it.
It is already present as a parameter in the General System Info plugin, but having it as a systemvariable makes it way more flexible.

micropet, 2018-04-28:
If you mean me - I do not know.
I thought FreeMen is important.

Grovkillen, 2018-05-17:
We should add freemem as variable because we will get the value better in the future. Thoughts on that?

Grovkillen, 2019-10-10:
This is added, closing.

closed 2019-10-10 by Grovkillen