letscontrolit/ESPEasy #1329: NTP doesn't work anymore

susisstrolch, 2018-04-28:
With current build (29f89b687d40b21f032c6c71911b165a63fe318c) NTP doesn't work anymore.
It shows a wrong date (Local Time: 2036-02-07 07:40:26) after reboot.

To find the reason I've added
```
  String log1 = F("getNtpTime()");
  addLog(LOG_LEVEL_DEBUG_MORE, log1);
```
to getNtpTime() function. This log entry never shows up, so it looks like getNtpTime() isn't called at all...
It seems there's no call to getNtpTime() in current code:
```
strolch@Strolchi ~/Development/esp8266/ESPEasy $ grep -r getNtpTime src
src/TimeESPeasy.ino:    unsigned long  t = getNtpTime();
src/TimeESPeasy.ino:unsigned long getNtpTime()
src/TimeESPeasy.ino:  String log1 = F("getNtpTime()");
```

TD-er, 2018-04-28:
Did you also see this with the bin file from the daily build, or only one built yourself?
I am seeing NTP results here on my own builds.

Where in the function did you add this log?
The first line of the log may already return when wifi is not connected.

susisstrolch, 2018-04-28:
Put it as first lines (before the test for setting and Wificonnected).
Same core but ESPEasy from March works properly.

TD-er, 2018-04-28:
Maybe the debuglevel is not set at that level yet when it is called?
For testing I usually set it to INFO

v-a-d-e-r, 2018-04-28:
Yep, I did this too, but permanent. It's only one line during boot process. So what.... ;-)

susisstrolch, 2018-04-28:
Ok, first NTP showed up after ~6min:
```3606428 : getNtpTime()
3606431 : NTP  : NTP host 192.168.254.20 (192.168.254.20) queried
3606442 : NTP  : NTP replied: 10 mSec
3606443 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
```

v-a-d-e-r, 2018-04-28:
Should be faster. Right after the Wifi connection is successfully done.

susisstrolch, 2018-04-28:
Strange... not really sure about the reason - multiple reboots now show the NTP time after 5sec...
Also changed the logleven in src to LOG_LEVEL_INFO...
```
INIT : Booting version: 62e6317 (ESP82xx Core 41a64707)
74 : INIT : Warm boot #1
76 : FS   : Mounting...
100 : FS   : Mount successful, used 76053 bytes of 957314
110 : CRC  : No program memory checksum found. Check output of crc2.py
141 : CRC  : SecuritySettings CRC   ...OK 
247 : INIT : Free RAM:31624
248 : INIT : I2C
248 : INIT : SPI not enabled
253 : INFO : Plugins: 8 (ESP82xx Core 41a64707)
253 : EVENT: System#Wake
261 : WIFI : Set WiFi to STA
mode : sta(5c:cf:7f:f1:bb:e1)
add if0
263 : WIFI : Connecting SusiconStrolch attempt #0
267 : OTA  : Arduino OTA enabled on port 8266
368 : getNtpTime()
378 : NTP  : check connection state failed
379 : EVENT: System#Boot
389 : WD   : Uptime 0 ConnectFailures 0 FreeMem 30112
493 : getNtpTime()
503 : NTP  : check connection state failed
1493 : getNtpTime()
1503 : NTP  : check connection state failed
2493 : getNtpTime()
2503 : NTP  : check connection state failed
scandone
state: 0 -> 2 (b0)
4014 : BMx280 : Detected BME280
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with SusiconStrolch, channel 13
dhcp client start...
ip:192.168.254.71,mask:255.255.255.0,gw:192.168.254.1
5603 : BME280: dew point 7.61C
5603 : BME280 : Address: 0x76
5603 : BME280 : Temperature: 20.02
5603 : BME280 : Humidity: 44.81
5604 : BME280 : Barometric Pressure: 1009.99
5611 : EVENT: BMx280#Temperature=20.02
5620 : EVENT: BMx280#Humidity=44.81
5626 : EVENT: BMx280#Pressure=1009.99
5865 : getNtpTime()
5870 : NTP  : NTP host 192.168.254.20 (192.168.254.20) queried
5881 : NTP  : NTP replied: 11 mSec
5882 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
5885 : EVENT: Time#Initialized
5894 : EVENT: Clock#Time=Sat,17:34
5898 : ACT  : taskvalueset 12,1,0
5903 : ACT  : taskvalueset 12,2,-55
5908 : ACT  : taskvalueset 12,3,28520
5915 : ACT  : taskvalueset 12,4,63283
5920 : WIFI : Connected! AP: SusiconStrolch (38:10:D5:B2:22:1E) Ch: 13 Duration: 3789 ms
5921 : EVENT: WiFi#ChangedAccesspoint
5926 : IP   : Static IP : 192.168.254.71 GW: 192.168.254.1 SN: 255.255.255.0 DNS: 192.168.254.1
STUB: dhcp_stop
5931 : WIFI : Static IP: 192.168.254.71 (D1pro-01-11) GW: 192.168.254.1 SN: 255.255.255.0   duration: 1878 ms
5940 : getNtpTime()
5945 : NTP  : NTP host 192.168.254.20 (192.168.254.20) queried
5958 : NTP  : NTP replied: 10 mSec
5960 : Current Time Zone:  DST time start: 2036-03-30 02:00:00 offset: 120 minSTD time start: 2036-10-26 03:00:00 offset: 60 min
5962 : EVENT: Time#Set
5969 : EVENT: WiFi#Connected
5976 : Webserver: start
5986 : EVENT: Clock#Time=Thu,07:28
5996 : ACT  : taskvalueset 12,1,0
6005 : ACT  : taskvalueset 12,2,-56
6013 : ACT  : taskvalueset 12,3,26216
6020 : ACT  : taskvalueset 12,4,26896
6025 : EVENT: Clock#Time=Thu,07:28 Processing time:39 milliSeconds
6602 : UDP  : 5C:CF:7F:23:C5:5A,192.168.254.96,6
ping 1, timeout 0, total payload 32 bytes, 1042 ms
ping 1, timeout 0, total payload 32 bytes, 1042 ms
12003 : Dummy: value 1: 0.00
12004 : Dummy: value 2: -56.00
12005 : Dummy: value 3: 26216.00
12006 : Dummy: value 4: 26896.00
12008 : EVENT: sysinfo#uptime=0.00
12015 : EVENT: sysinfo#uptime=0.00 Processing time:7 milliSeconds
12016 : EVENT: sysinfo#RSSI=-56.00
12024 : EVENT: sysinfo#RSSI=-56.00 Processing time:8 milliSeconds
12025 : EVENT: sysinfo#sysheap=26216.00
12034 : EVENT: sysinfo#sysheap=26216.00 Processing time:9 milliSeconds
12034 : EVENT: sysinfo#syssec_d=26896.00
12043 : EVENT: sysinfo#syssec_d=26896.00 Processing time:9 milliSeconds
12070 : HTTP : connecting to 192.168.254.27:8383
```
Closing that one...

closed 2018-04-28 by susisstrolch

thymjan, 2018-04-29:
Noticed only now my entry (a WAN-Adress) for NTP was insufficient / not reachable. Nevertheless older versions of ESPEasy got the time right (fallback to router IP?). Now I entered the router-ip and it works flawlessly.

TD-er, 2018-04-29:
You can also leave it empty and then one of 1...4 .pool.ntp.org will be chosen at random.