letscontrolit/ESPEasy #1330: Playground plugin fails, exception29

s0170071, 2018-04-28:
current mega causes my P183_SMA playground plugin to crash whith exception29, which is obviously a faulty pointer. 

@TD-er: you can probably tell me briefly what has changed in the last two weeks that could cause that, maybe the number of the plugin ? Have the jump tables changed ?

There was a message immediately before the exception that was indicating I should increase the number of plugins. But where to ? 255 did not change anything.

TD-er, 2018-04-28:
It just crashes, so it is not a build error?
[Exception 29](https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.rst)
Probably trying to write to something that's no longer a valid pointer.
Increase the number of plugins is not the solution I guess.

Only thing that's really changed is in the page load section of the plugin.
You should not write to the `string` anymore.

Is it the same code as on the playground?

s0170071, 2018-04-28:
Yes. Almost. That string is discovered by the compiler. And it contains a copy of the modus class which needs commenting out. Other than that....

TD-er, 2018-04-28:
I added your Modbus class and maybe I had to change just a few things to get PlatformIO to accept it.

s0170071, 2018-04-28:
Will see tomorrow. I suspect the configuration. It was a pre configured unit with a espeasy version tagged serial wd crash or so from 2 months ago.

s0170071, 2018-04-29:
It is configuration related. Just tried another device - working.

s0170071, 2018-04-29:
Ok, now I have a boot loop which I could only break by manually inserting a factory reset command. 

Maybe we should add a version number to the settings. The binary verifies that version number. If anything goes incompatible we can set the minimum required version to a different number and force factory reset. And thus break the loop without recompilation.

The version number should also have a distinct marking- in case it is not set at all. E.g. 0x11220001 where 0x1122 is the mark and 0001 the number.

TD-er, 2018-04-29:
There is already some mark like this, to do exactly that, fix some issues with settings.
Do you have any idea where in the settings things were messed up?

TD-er, 2018-04-29:
What is SSDP and who's using it?

s0170071, 2018-04-29:
https://en.wikipedia.org/wiki/Simple_Service_Discovery_Protocol
I tried it, however, I cannot see the benefit in my network. So no, I am not seriously using it. It was still activated in my settings- dormant so to speak. Maybe ssdp stands for "Simple Set-up of Dfficult Pootloop" ;)

Budman1758, 2018-04-29:
Take any recent build, Go to advanced settings, enable SSDP in "experimental" settings = instant bootloop.

I had thought that was necessary for the node list and had enabled it on all my units. Turns out it does not seem to be needed.  I have a few units that are still running DEV-12 version that have over 200 days uptime and its enabled on those.

Grovkillen, 2018-04-30:
Can confirm this, we should add a warning or remove it from the advanced page for the v2.1 release since we shouldn't waste time on it until after the release.

dkotte, 2018-07-06:
Please add the Warning !!
Today i kill 2 devices. (ESPEasy_mega-20180524)
Some hint how to recover from boot-loop?
Connecting RX and TX pin don't work.

dkotte, 2018-07-06:
Possible you remove the "Inter-ESPEasy Network" checkbox?
I am unable to enable this. With old version i see this option.
But the SSDP checkbox remain active ... without warning. (mega-20180625)

Grovkillen, 2018-07-06:
You can try flashing a blank bin file.

s0170071, 2018-07-06:
@dkotte Been there. If you use Arduino IDE select a different spiffs size. This way the is no config after boot. 
You may also look for the reset command in espeasy.ino and put it right after spiffs init. Flash. Run once. Undo and flash again.

closed 2022-11-09 by TD-er