letscontrolit/ESPEasy #1332: ESP32 - last changes: no MQTT, no NTP ..

flexiti, 2018-04-29:
29.04.2018  ESP32, static IP
recent changes have meant that the system does not even try to connect to the MQTT server, nor is there any connection with NTP

Should I add something in the configuration? (before yesterday's amendments everything worked)

```
INIT : Booting version:  (ESP32 SDK v3.1-dev-239-g1c3dd23f-dirty)
38 : INIT : Cold Boot
39 : FS   : Mounting...
88 : CRC  : No program memory checksum found. Check output of crc2.py
103 : CRC  : SecuritySettings CRC   ...OK 
214 : INIT : Free RAM:128396
215 : INIT : I2C
215 : INIT : SPI not enabled
1107 : INFO : Plugins: 6 [Normal] (ESP32 SDK v3.1-dev-239-g1c3dd23f-dirty)
1107 : EVENT: System#Wake
1121 : WIFI : Set WiFi to STA
1183 : WIFI : Connecting T100 attempt #0
1184 : IP   : Static IP : 192.168.0.68 GW: 192.168.0.1 SN: 255.255.255.0 DNS: 8.8.8.8
1191 : OTA  : Arduino OTA enabled on port 8266
1202 : EVENT: System#Boot
1207 : ACT  : notify 1
1242 : RTOS : Launching tasks
1245 : WD   : Uptime 0 ConnectFailures 0 FreeMem 89788
1271 : SYS  : 0.00
1271 : EVENT: Rssi#RSSI=0.00
1343 : SW   : Switch state 1 Output value 1
1362 : EVENT: switch1#switch1=1.00
1389 : SW   : State 1.00
1390 : EVENT: switch1#switch1=1.00
2507 : SYS  : 89488.00
2507 : EVENT: #free=89488.00
4632 : DS   : Temperature: 21.19 (28-ff-d2-a9-b1-16-4-9f)
4632 : EVENT: DS18B20#Temp=21.19
5707 : SYS  : 0.00
5708 : EVENT: uptime#Uptime=0.00
6805 : WIFI : Static IP: 192.168.0.68 (main/sonoff/68-1) GW: 192.168.0.1 SN: 255.255.255.0   duration: 4209 ms
6815 : EVENT: WiFi#Connected
6827 : Webserver: start
24549 : Ram usage: Webserver only: 0 including Core: 0
26117 : Ram usage: Webserver only: 0 including Core: 0
31245 : WD   : Uptime 1 ConnectFailures 0 FreeMem 85156
60848 : EVENT: Clock#Time=Thu,00:01
60860 : EVENT: Clock#Time=Thu,00:01 Processing time:12 milliSeconds
61245 : WD   : Uptime 1 ConnectFailures 0 FreeMem 85156
61855 : SYS  : -44.00
61855 : EVENT: Rssi#RSSI=-44.00

```

TD-er, 2018-04-29:
The most recent changes should not change anything to ESP32, so it should be comparable with the code of this morning.

I just flashed my ESP32 and the first time it did not complete the boot, but after a reset it did.
With DHCP it just connects to the MQTT broker and NTP also works just fine over here.
With static IP configuration, it reports this:
`3555 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED`

So that's a problem.
Strange thing is that it still responds to the web requests and it is connected. (UDP messages do arrive)

TD-er, 2018-04-29:
With my latest merge this issue should be fixed for both static IP and DHCP.
At least my ESP32 module does do the reconnects + MQTT + NTP like before.