letscontrolit/ESPEasy #1337: Enable hidden SSID when connecting

Wiki591, 2018-04-30:
Up to version mega-20180426 connecting to a network with hidden SSID was no problem.

Using version mega-20180428 the attempt to connect to a network with hidden SSID ends up with error "No AP found"

Pls reenable this feature, its a question of avoiding unwanted and annoying automated connection tries in our recent environment of everything-is-online.

Oxyandy, 2018-04-30:
Please try [mega-20180430](https://github.com/letscontrolit/ESPEasy/releases) and report back..
Is not there yet, should be soon..

DittelHome, 2018-04-30:
**On my side, hidden SSID is working, absolutely no problem.**
Maybe i does not have problem, because i set the ssid via serial command ...

TD-er, 2018-04-30:
Hidden SSID is a flag for scanning, I am not sure it is also a flag for the WiFi.begin().

DittelHome, 2018-04-30:
Strange, now i have done a test with DHCP. My SSID is hidden in the AP. Connect no problem, see here
Maybe your DHCP range is exhausted. Mac Filter ?
438 : CRC : binary has changed since last save of Settings ==> type save

```
104 : INIT : Warm boot #3
105 : FS   : Mounting...
130 : FS   : Mount successful, used 75802 bytes of 957314
420 : CRC  : program checksum       ...OK
452 : CRC  : SecuritySettings CRC   ...OK 
558 : INIT : Free RAM:23528
559 : INIT : I2C
559 : INIT : SPI not enabled
572 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1)
573 : WIFI : Set WiFi to STA
604 : WIFI : Connecting ***** attempt #0
611 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22688
4366 : WIFI : Connected! AP: ****** (30:B5:C2:EB:DB:7D) Ch: 9 Duration: 3760 ms
4708 : WIFI : DHCP IP: 192.168.1.32 (ESP-Easy-7) GW: 192.168.1.1 SN: 255.255.255.0   duration: 341 ms
4711 : Webserver: start
```

Wiki591, 2018-04-30:
Hi,
it seems to me as if you are booting a different version because your line numbers are different. I tried the "normal" bin of 0428 and 0430 which did not work.

The normal bin 0426 is working pretty stable with hidden SSID as well as all other previous versions I have tried.

DittelHome, 2018-04-30:
I have tested with this version:
ESP_Easy_mega-20180430_normal_ESP8266_4096.bin
Hardware: Wemos D1 mini

HV-NL, 2018-04-30:
Flashed with mega-20180430, and performed a factory reset.
No success: My computer does not see the ESP as accesspoint.

2nd try: manually configured the ESP. 
```
wifissid MyOwnAP
wifikey MyWifiPassword
Save
wificonnect
```

Also without success:
```
583 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22688
3436 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
3584 : WIFI : Connecting MyOwnAP attempt #1
6442 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2858 ms
6584 : WIFI : Connecting MyOwnAP attempt #2
9442 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2858 ms
9584 : WIFI : Connecting MyOwnAP attempt #3
12444 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
12584 : WIFI : Connecting MyOwnAP attempt #4
15443 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
15584 : WIFI : Connecting MyOwnAP attempt #5

```
When going back to a previous version (dev-13) it is working again.

So perhaps now only connecting to a **hidden AP** is possible?

DittelHome, 2018-04-30:
Have you tried a clean install ?
[http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/](url)

TD-er, 2018-04-30:
If you are connected to the ESPeasy node via serial, you can simply reset to factory default.
See [command reference](https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference)
Just enter "Reset" in the console.

And no, it is not limited to hiddenAP, because my own AP's are not hidden and they work just fine.

HV-NL, 2018-04-30:
I *did* perform a 'reset' on my NodeMCU device in the console.
However I don't see ESP_Easy as an accesspoint.
After manually configuring my local accespoint it does not connect to my AP, and although the logging says "Set WiFi to AP+STA" I cannot see it in the air.

```
Hardware: NodeMCU
Software: ESP_Easy_mega-20180430_normal_ESP8266_4096.bin
```
**Brainwave**: I flashed a new Wemos D1 mini with this binary file. 
Bingo! Now I do see ESP_Easy as AP, and i can connect and configure it properly. It then normally connects to my accesspoint.

So my problem seems to be hardware-related. 
As a test I also completely erased my nodeMCU (same result: no Wifi), so it must somehow be related to the combination ESP_Easy_mega-20180430_normal_ESP8266_4096.bin, running on this particular nodeMCU device.

The hardware however does function properly, when I go back to dev-13.

It was the cheapest nodeMCU that I ever bought, and does not have the metal shield over the WiFi compartment.
[image]

My guess is that this particular device does not have the proper hardware for the current Wifi code.

TD-er, 2018-04-30:
It can be that some part of the flash is not configured properly.
See these lines in the code:
https://github.com/letscontrolit/ESPEasy/blob/45afaf8d3f860f10011636e675c4879357f9501e/src/ESPEasy.ino#L97-L101

This persistent line, disables writing to that part of the flash.
If you comment that one line, can you connect to the AP?

HV-NL, 2018-04-30:
(took some time to create a fresh build environment)

I commented line 99 (the WiFi.persistent line). Same problem.

Oxyandy, 2018-05-01:
@DittelHome 
`esptool erase_flash`
**Does the same things as flashing a whole Blank Firmware, only faster
I added a "SendTo" batch file in windows so I can flash from anywhere.**

[image]

**I then give myself the option to erase the whole Node before flash..**

[image]

HV-NL, 2018-05-02:
Flashing todays version (mega-20180502) solves the problem. Now I can configure the nodeMCU by connecting to the Wifi (192.168.4.1), and after that it will properly connect to my accesspoint.

As I don't have a hidden AP available, I do not know if the original problem is solved too.

Wiki591, 2018-05-02:
Hi,
yesterday I have toasted my last free Wemos Device. I am awaiting for tomorrow three new devices, so I will report tomorrow.
Thanks for your efforts up to now.

Wiki591, 2018-05-03:
New devices arrived, flashed 0503, still no connection to hidden SSID. Perhaps needed: my AP is an AVM Fritzbox 7272

Protocol:
INIT : Booting version: mega-20180503 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2)
90 : INIT : Warm boot #2
91 : FS   : Mounting...
117 : FS   : Mount successful, used 75802 bytes of 957314
410 : CRC  : program checksum       ...OK
416 : CRC  : SecuritySettings CRC   ...OK 
522 : INIT : Free RAM:22528
523 : INIT : I2C
523 : INIT : SPI not enabled
536 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2)
537 : WIFI : Set WiFi to STA
569 : WIFI : Connecting NTDataHome attempt #0
575 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21736
3421 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2852 ms
3573 : WIFI : Connecting NTDataHome attempt #1
6426 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2852 ms
6573 : WIFI : Connecting NTDataHome attempt #2
9426 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2852 ms
9573 : WIFI : Connecting NTDataHome attempt #3
12425 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2851 ms
12572 : WIFI : Connecting NTDataHome attempt #4
15425 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2852 ms
15570 : WIFI : Connecting NTDataHome attempt #5

Oxyandy, 2018-05-04:
@Wiki591 you say "toasted my last free Wemos Device"
What do you mean ?
Have you tried esptool erase_flash ?
Should recover them just fine ;)
Cheers

Oxyandy, 2018-05-04:
And @Wiki591 - If you make the AP visible do they then connect fine ?

TD-er, 2018-05-04:
@Oxyandy Some of us let the magic smoke escape from the electronics and as we all know, electronics main component is this magic smoke. If you let it out, electronics do not work anymore.
Or... so I heard from a friend of mine, never happened to me of course ;)

Oxyandy, 2018-05-04:
### Oh, this kind of toasted.. Which you rather, jam or honey ?!

[image]

Wiki591, 2018-05-04:
Hi,

@Oxyandy: as TD-er already stated: producing some tiny little blue clouds exhaled by the electronics is not the best way to keep these things running. They end up in the all-embracive Data Nirvana.

Back to problem: I have no problems to connect to my AP (AVM Fritzbox 7272) if it is visible. As well as with the same config.dat, just flashed mega 0421 - 0426 the devices connect to the AP with hidden SSID instantaneously. Flashing the same device with the same config with 0428 and higher trying to connect to the same WLAN produces the protocol posted before - no connection.

Oxyandy, 2018-05-04:
Ok, you have inspired me, I will make my SSID hidden & play.

TD-er, 2018-05-04:
I was thinking about changing this by adding the parameter to the wifi setup function, but I don't want to change too much at the same time anymore, or else you don't know what's causing what anymore.
If I don't forget it, I will set the "hidden" flag for connecting to wifi later this evening.

Oxyandy, 2018-05-04:
meh, my node does not seem to care if set hidden, connects anyway..
But it seems you can get 0502 to load, for me just loops every 11 seconds
Connected -- Disconnected! Reason: '(200) Beacon timeout' -- Connected
Sonoff Basic units

TD-er, 2018-05-04:
@Oxyandy Do you have this Beacon timeout with 0504 on any node, or just a few?

Wiki591, 2018-05-06:
Hi,
as to be seen in your documentation, there was done a lot of work. But still with version 0505 I have no success connecting to my hidden AP. Protocol:

INIT : Booting version: mega-20180505 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
89 : INIT : Warm boot #3
90 : FS   : Mounting...
115 : FS   : Mount successful, used 75802 bytes of 957314
406 : CRC  : program checksum       ...OK
412 : CRC  : SecuritySettings CRC   ...OK 
519 : INIT : Free RAM:23448
519 : INIT : I2C
519 : INIT : SPI not enabled
533 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
533 : WIFI : Set WiFi to STA
565 : WIFI : Connecting *** attempt #0
572 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22680
3426 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
3570 : WIFI : Connecting *** attempt #1
6430 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
6570 : WIFI : Connecting *** attempt #2
9430 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
9570 : WIFI : Connecting *** attempt #3
12430 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
12570 : WIFI : Connecting *** attempt #4
15430 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
15570 : WIFI : Connecting *** attempt #5
18430 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
18568 : WIFI : Set WiFi to AP+STA
19487 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
19488 : WIFI : Connecting *** attempt #6
22349 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
22568 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
22569 : WIFI : Connecting *** attempt #7
25428 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
25568 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
25569 : WIFI : Connecting *** attempt #8
28428 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2858 ms

Nothing changed, just flashed back to 0426:

INIT : Booting version: mega-20180426 (ESP82xx Core 2_3_0)
256 : INIT : Warm boot #5
257 : FS   : Mounting...
281 : FS   : Mount successful, used 75802 bytes of 957314
556 : CRC  : program checksum       ...OK
561 : CRC  : SecuritySettings CRC   ...OK 
562 : CRC  : binary has changed since last save of Settings
668 : INIT : Free RAM:20416
668 : INIT : I2C
668 : INIT : SPI not enabled
682 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_3_0)
683 : WIFI : Switch on WiFi
683 : WIFI : Set WiFi to STA
689 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19760
690 : WIFI : Connecting *** attempt #0
3738 : WIFI : Connected! AP: *** (08:96:D7:34:09:A0) Ch: 12 Duration: 3047 ms
4315 : WIFI : DHCP IP: 192.168.xxx.218 (ESP-Easy-0) GW: 192.168.xxx.93 SN: 255.255.255.0   duration: 577 ms
4318 : Webserver: start

TD-er, 2018-05-06:
That's really strange, since I have been testing with hidden ssid here at home the last couple of days.
My AP was running in hidden mode (Fritzbox 1750E, not running as repeater, but as accesspoint)  and I was wondering this morning why my Chromebook could not find the AP at the garden side. Then I realized it was still in hidden mode, but all of my ESP boards used it just fine.

Wiki591, 2018-05-06:
I tried 0506, this time using the gui instead of setting SSID and Key per serial connecion. In the list of initial Wifiscan my AP is identified as #Hidden#, correct Mac and correct channel (in this case Ch:12). When I try to connect to the #Hidden# entry using the corresponding key, nothing happens. When I choose "different SSID" entering the correct SSID and Key & submit, no connection is established. Protocol entry as usual 'No AP found'. Thats weird.

[OT]
Btw, different theme: Your recent work on the stability of establishing Wifi connection and MQTT was very successful. My sensor for the waterlevel in my cistern, currently awaking every 15 min and connecting to a non-hidden AP on Raspberry Pi, seems to work with 0506 much more reliable, thanks for that. Now I shall be able to raise the interval to 60min resp. the max of 71min. Thats pretty good for the batteries lifetime.
[/OT]

Oxyandy, 2018-05-06:
Hi @Wiki591 
Can you please supply the full name of the latest firmware file you have used ?

Wiki591, 2018-05-06:
Hi,
of course, sorry for my inclomplete documentation.
Last fw used: ESP_Easy_mega-20180506_normal_ESP8266_4096.bin

Oxyandy, 2018-05-07:
[ESP_Easy_mega-20180506_normal_ESP8266_4096_Self_Compiled.zip](https://github.com/letscontrolit/ESPEasy/files/1978167/ESP_Easy_mega-20180506_normal_ESP8266_4096_Self_Compiled.zip)

Try this please, there have been some quirks, maybe this resolves it for you.
Report back Ta

Wiki591, 2018-05-07:
Flashed over the top, no settings changed.
No, didn't solve the problem. Protocol with hidden SSID set:

_INIT : Booting version:  (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
89 : INIT : Warm boot #3
90 : FS   : Mounting...
115 : FS   : Mount successful, used 75802 bytes of 957314
122 : CRC  : No program memory checksum found. Check output of crc2.py
127 : CRC  : SecuritySettings CRC   ...OK 
232 : INIT : Free RAM:22824
233 : INIT : I2C
233 : INIT : SPI not enabled
247 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
247 : WIFI : Set WiFi to STA
279 : WIFI : Connecting *** attempt #0
286 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22040
3140 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
3283 : WIFI : Connecting *** attempt #1
6145 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
6283 : WIFI : Connecting *** attempt #2
9145 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
9283 : WIFI : Connecting *** attempt #3
12145 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
12280 : WIFI : Connecting *** attempt #4
15141 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
15283 : WIFI : Connecting *** attempt #5
18143 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
18283 : WIFI : Set WiFi to AP+STA
19202 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
19203 : WIFI : Connecting *** attempt #6
22065 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
22282 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
22282 : WIFI : Connecting *** attempt #7
25142 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
25282 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
25282 : WIFI : Connecting *** attempt #8
28142 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
28281 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
28282 : WIFI : Connecting *** attempt #9
30285 : WD   : Uptime 1 ConnectFailures 0 FreeMem 19600
31142 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
32281 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
32282 : WIFI : Connecting *** attempt #10_

Connecting same AP, SSID visible:

_INIT : Booting version:  (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
89 : INIT : Warm boot #5
90 : FS   : Mounting...
115 : FS   : Mount successful, used 75802 bytes of 957314
123 : CRC  : No program memory checksum found. Check output of crc2.py
128 : CRC  : SecuritySettings CRC   ...OK 
233 : INIT : Free RAM:22824
233 : INIT : I2C
233 : INIT : SPI not enabled
247 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
247 : WIFI : Set WiFi to STA
279 : WIFI : Connecting *** attempt #0
286 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22040
4050 : WIFI : Connected! AP: *** (08:96:D7:34:09:A0) Ch: 12 Duration: 3770 ms
5088 : WIFI : DHCP IP: 192.168.xxx.203 (ESP-Easy-0) GW: 192.168.xxx.93 SN: 255.255.255.0   duration: 1037 ms
5091 : Webserver: start_

at least, during this established connection made the AP hidden, follow-up in protocol:

_30286 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18528
60286 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18528
90286 : WD   : Uptime 2 ConnectFailures 0 FreeMem 18528
114051 : WIFI : Disconnected! Reason: '(6) Not authed' Connected for 1 m 49 s
114282 : WIFI : Connecting *** attempt #0
117147 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2864 ms
117282 : WIFI : Connecting *** attempt #1
120145 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2863 ms
120282 : WIFI : Connecting *** attempt #2
120286 : WD   : Uptime 2 ConnectFailures 0 FreeMem 19368
123146 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2864 ms
123282 : WIFI : Connecting *** attempt #3
126145 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2863 ms_

TD-er, 2018-05-07:
Apparently it may be related to the first connect attempt when in AP mode: https://github.com/esp8266/Arduino/issues/4519
Could you try to connect to the same AP when it is not set to be hidden, save that settings and then set it to hidden again and try to connect.

Wiki591, 2018-05-11:
Hi,
sorry for late reply. My health played around with me the last days.
I'll try it today.

Wiki591, 2018-05-13:
OK, here we go.

What I have done first:
- factory reset
- made the AP visible
- connected to the AP
- result seee Protocol 1

Then I have hidden the SSID of the AP again - no reboot of the Wemos
- result see Protocol 2
I have noticed, that the Wemos lost immediately the connection to the AP and logged
"288510 : WIFI : Disconnected! **Reason: '(8) Assoc leave'** Connected for 2 m 52 s" 

The I rebooted the Wemos
- result see Protocol 3

**Protocol 1:**

_60286 : WD   : Uptime 1 ConnectFailures 0 FreeMem 18528
60615 : Command: reset
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 0
RESET: formatting...
RESET: formatting done...
71891 : FILE : Saved config.dat
71921 : FILE : Saved security.dat
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
?U88 : 

INIT : Booting version:  (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
89 : INIT : Warm boot #2
90 : FS   : Mounting...
115 : FS   : Mount successful, used 75802 bytes of 957314
121 : CRC  : No program memory checksum found. Check output of crc2.py
126 : CRC  : SecuritySettings CRC   ...OK 
232 : INIT : Free RAM:22824
233 : INIT : I2C
233 : INIT : SPI not enabled
247 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
247 : WIFI : No valid wifi settings
247 : WIFI : Could not connect to AP!
251 : WIFI : Set WiFi to AP
1173 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1180 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20512
30593 : AP Mode: Client connected: 18:3D:A2:09:D2:64 Connected devices: 1
30596 : Webserver: start
31179 : WD   : Uptime 1 ConnectFailures 0 FreeMem 17128
50086 : WIFI : Set WiFi to AP
64414 : AP Mode: Client connected: 18:3D:A2:09:D2:64 Connected devices: 1
64415 : AP Mode: Client disconnected: 18:3D:A2:09:D2:64 Connected devices: 1
64434 : WD   : Uptime 1 ConnectFailures 0 FreeMem 13544
94434 : WD   : Uptime 2 ConnectFailures 0 FreeMem 15728
112727 : WIFI : Credentials Changed, retry connection. SSID: ***
112773 : WIFI : Set WiFi to AP+STA
112805 : WIFI : Connecting *** attempt #0
115099 : AP Mode: Client disconnected: 18:3D:A2:09:D2:64 Connected devices: 0
115144 : AP Mode: Client connected: 18:3D:A2:09:D2:64 Connected devices: 1
116316 : WIFI : Connected! AP: *** (08:96:D7:yy:yy:yy) Ch: 12 Duration: 3510 ms
116538 : AP Mode: Client disconnected: 18:3D:A2:09:D2:64 Connected devices: 0
116585 : AP Mode: Client connected: 18:3D:A2:09:D2:64 Connected devices: 1
117353 : WIFI : DHCP IP: 192.168xx203 (ESP-Easy-0) GW: 192.168xx93 SN: 255.255.255.0   duration: 1037 ms
[... 100 lines cut ...]
36309 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
36310 : WIFI : Connecting *** attempt #11
39171 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
40309 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
40310 : WIFI : Connecting *** attempt #12
43170 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
44309 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
44310 : WIFI : Connecting *** attempt #13
47170 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
48309 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
48310 : WIFI : Connecting *** attempt #14
51170 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
53309 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
53310 : WIFI : Connecting *** attempt #15
56169 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
58310 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
58310 : WIFI : Connecting *** attempt #16
60312 : WD   : Uptime 1 ConnectFailures 0 FreeMem 19600
61170 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
63310 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
63310 : WIFI : Connecting *** attempt #17
66170 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
68310 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
68310 : WIFI : Connecting *** attempt #18
71170 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms_

TD-er, 2018-05-13:
I read last week you have to set the flag to use a BSSID to connect and also make sure to include that BSSID.
I have no idea why my nodes will keep connecting when SSID is hidden.

Wiki591, 2018-05-17:
öhm, does anyone still take care of this issue (of mine)?
I mean, my nodes are working, yes, but with some of the irregularities of the state 20180426, ie slow answer, instable connections etc.

TD-er, 2018-05-18:
It is still on my to-do list.
The biggest limitation is the lack of time.

Wiki591, 2018-05-18:
Thank you, so I'll be patient.
What I am wondering about is the fact, that this issue popped up after the update from 20180426 to 20180428.

TD-er, 2018-05-18:
That's about when we changed to 2.4.1 I guess

Wiki591, 2018-05-18:
Yes, correct, so tells the documentation - and some changes in the [wifi] section

Wiki591, 2018-05-18:
Reading the docs concerning changes in 20180426 -> 20180428, its says "Just disconnect when DNS lookup is not possible"
In my network, router (AP), dhcp server and dns server are not on the same machine, the router (AP) works only as dhcp relay agent. Maybe the issue deals with the structure of my site, but only if the SSID is hidden? Strange? Again, if the AP is visible, mega works flawlessly.

Oxyandy, 2018-05-19:
@Wiki591 
Have you tried setting static IPs ?
You need to define:
IP: 192.168.xxx.yyy <= xxx needs to match range in router 0 or 1 ?
(yyy above range used by DHCP) say above 192.168.xxx.230 as by looking at your previous log
SN: 255.255.255.0
GW: 192.168.xxx.93
DNS: 8.8.8.8 <= Google's DNS

TD-er, 2018-05-19:
When using a static IP config, you now have to use all fields.
Previous versions would let you get away with incomplete settings, but since 2.4.x all fields need to be set.

Wiki591, 2018-05-19:
@Oxyandy 
"DNS: 8.8.8.8 <= Google's DNS"
Seriously? My nodes are not allowed to leave my internal net - never ever.

Yesterday I have configured my AP standard-like, ie DHCP by himself. No success.

I just tried static IP, no success either. Protocol:

INIT : Booting version: mega-20180513 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
89 : INIT : Warm boot #3
91 : FS   : Mounting...
116 : FS   : Mount successful, used 75802 bytes of 957314
437 : CRC  : program checksum       ...OK
443 : CRC  : SecuritySettings CRC   ...OK 
549 : INIT : Free RAM:21416
549 : INIT : I2C
549 : INIT : SPI not enabled
566 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
566 : WIFI : Set WiFi to STA
598 : WIFI : Connecting *** attempt #0
599 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
607 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20568
3460 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
3605 : WIFI : Connecting *** attempt #1
3606 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
6467 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
6605 : WIFI : Connecting *** attempt #2
6606 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
9468 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
9605 : WIFI : Connecting *** attempt #3
9606 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
12468 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
12605 : WIFI : Connecting *** attempt #4
12606 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
15468 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
15605 : WIFI : Connecting *** attempt #5
15606 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
18467 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
18605 : WIFI : Set WiFi to AP+STA
19525 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
19525 : WIFI : Connecting *** attempt #6
19526 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
22396 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
22604 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
22604 : WIFI : Connecting *** attempt #7
22605 : IP   : Static IP : 192.168.xxx.xxx GW: 192.168.xxx.xxx SN: 255.255.255.0 DNS: 192.168.xxx.xxx
25471 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2858 ms

Wiki591, 2018-06-02:
Hi,
mysterious (at least for me) but I have found a workaround.

Due to an overload of APs in my neighbourhood on channels 1-11, I switched in the past my AP to Channel 12. Result was a seriously better performance and stability. I don'nt know why, but in my neighbourhood several APs n the last 2-3 months where switched to channel 12, too. Today I switched to channel 9. Result: better performance and - i can't believe it - ESPEasy is able to connect to my hidden network.

GOTCHA! For me the issue is solved and there is a workaround for other users in the furture.

I am very grateful for your thoughts and your work/efforts in the past two months about this issue. Hopefully with this message you will be freed a little bit concerning the lack of time in this stage of the project.

THANK YOU!

closed 2018-06-02 by Wiki591

Wiki591, 2018-06-02:
Hi,
double-checked: AP on channl 1-10: Connection to AP either hidden or not works fine, AP on channel 11 and above only works with AP SSID visble.

TD-er, 2018-06-02:
OK, that's strange to say the least.
Maybe the scan for networks is either not complete, or the hidden scan is using some regional setting, which is not set and just using the default (China)
I'll keep it in mind and when I'm browsing through some code related to this, I will have a look at it.

Glad to see this to work.

@Grovkillen Can you make some note about this on the Wiki?

Grovkillen, 2018-06-02:
@TD-er I will

Wiki591, 2018-06-03:
Hi,
the most strange thing is, that, scanning for networks, the hidden AP appeared in the list of found networks with its correct mac and the correct channel (ch 12), marked as hidden, but no connection possible. So for me the scan itself is absolutely correct. Thats why I didn't try the change of the channel, I found this workaround only by accident. After setting the AP to channel 9, I even had not to scan the networks, I only entered the ssid and key over the serial connection and the device connected immediately as it did on channel 12 when setting the the AP to visble.

After updating my productive devices to ESP_Easy_mega-20180522_normal_ESP8266_4096 I can report, all are doing their job stable and are performing good during the past 20 hours or so.

TD-er, 2018-06-03:
Good to hear it is working fine now.

Grovkillen, 2018-06-05:
https://www.letscontrolit.com/wiki/index.php/Wifi

Wiki591, 2018-06-05:
fyi:
today night I stepped down into my cellar and noticed, that the devices of my battery monitoring (mega-20180522) are pretty busy on switching between my hidden AP (RSSI between -55 and -60) and my fallback AP (different hardware, different SSID, different ip-net, RSSI between -82 and -84). They are sending robust and stable their data to my server, so I didn't notice that behaviour before from the far. I had upgraded them from mega-20180426, unchanged configuration, this behaviour is absolutely new. They now have built up their own funny lightshow in the dark ;-).

enzon0, 2022-12-15:
my configuration with hidden ssid:
wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    hidden: true

TD-er, 2022-12-15:
> my configuration with hidden ssid: wifi: networks:
> 
> * ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   hidden: true

Not sure what you intend to tell us with this reply.
Can you elaborate what it is you're trying to explain or ask here, in this >4 year old issue?